eruption-project/eruption #78: Input lag on Manjaro version 0.1.21 for TKL Pro

n1coh, 2021-08-18:
Hello, I have lag in input on a TKL Pro, I am on Manjaro.
I have upgrade to version 0.1.21, I thought that the correction of the ticket #76 would correct the problem but not.
When executing these commands everything goes well (no lag) :
```
 $ sudo RUST_LOG=debug eruption
 $ sudo RUST_LOG=trace eruption
```

journalctl :
```
août 18 11:37:52 nicolas-pc eruption[741299]:  WARN  eruption                                   > More than 82 milliseconds of jitter detected!
août 18 11:37:52 nicolas-pc eruption[741299]:  WARN  eruption                                   > This means that we dropped at least one frame
août 18 11:37:52 nicolas-pc eruption[741299]:  WARN  eruption                                   > Loop took: 1201 milliseconds, goal: 50
août 18 11:37:53 nicolas-pc eruption[741299]:  WARN  eruption                                   > More than 82 milliseconds of jitter detected!
août 18 11:37:53 nicolas-pc eruption[741299]:  WARN  eruption                                   > This means that we dropped at least one frame
août 18 11:37:53 nicolas-pc eruption[741299]:  WARN  eruption                                   > Loop took: 600 milliseconds, goal: 50
août 18 11:39:27 nicolas-pc eruption[741299]:  WARN  eruption                                   > More than 82 milliseconds of jitter detected!
août 18 11:39:27 nicolas-pc eruption[741299]:  WARN  eruption                                   > This means that we dropped at least one frame
août 18 11:39:27 nicolas-pc eruption[741299]:  WARN  eruption                                   > Loop took: 136 milliseconds, goal: 50
```

X3n0m0rph59, 2021-08-18:
Hi @n1coh, thank you for reporting this issue!

Are you using the AUR package `eruption` or are you using the package `eruption-git` of which the current version would be `0.1.22.r0.gf0cb968-1`.

If you are using `eruption` then please try out the `eruption-git` package first!

If switching to `eruption-git` does not fix the problem, then please post the full output of:

```sh
 $ sudo RUST_LOG=debug eruption
```

And the full output of this would be nice to have too:

```sh
 $ sudo RUST_LOG=trace eruption
```

n1coh, 2021-08-18:
Indeed, I had not seen this update on eruption-git ... my bad
I have no more lag with this version.
Thx for your work.
I have a ROCCAT Kone Pure first of the name btw, I can help to add it to the devices list if you want.

closed 2021-08-18 by n1coh

X3n0m0rph59, 2021-08-19:
@n1coh Great that it works now! I will soon do a release of the non `-git` package too.

I believe the `ROCCAT Kone Pure` is not an AIMO device, but I am not completely sure about that. If it is not that would mean that the embedded controller is too slow to handle realtime color updates. It then could just be programmed to a color in a one shot mode.

Is the device compatible with ROCCAT Swarm or does it require another driver Software?

n1coh, 2021-08-19:
You are right, it seems to me that another application is needed

sellerie98, 2023-01-26:
@n1coh you can use the "legacy" roccat-tools package for that, which I think are unfortunately not maintained anymore since there are also no new devices made that would be compatible.