eruption-project/eruption #79: Roccat Vulkan TKL - Volume Knob

LLuth0r, 2021-08-23:
**Introduction**
Change the volume knob control back so that it controls the volume in the system.

**Describe The Problem To Be Solved**
Not sure if this is a a problem, but how can I switch back the volume control knob so it no longer changes the brightness?  Would prefer to control audio that way.  I'm not familiar with Rust language, but would love to know how to change this setting.

**Describe the Solution**
See above.

X3n0m0rph59, 2021-08-24:
Hi @LLuth0r, thank you for reporting this issue!

The intended behavior for the dial knob on the `ROCCAT Vulcan TKL` and `ROCCAT Vulcan TKL PRO` keyboard variants is to control the master volume of the system without requiring extra configuration.

The Eruption hardware driver for the `ROCCAT Vulcan TKL` and `ROCCAT Vulcan TKL PRO` variants contains the source code for generating events that change the global brightness too. The keyboards firmware generates USB HID reports on rotation of the dial knob. On the `ROCCAT Vulcan 1xx` series keyboards the mode of the dial knob can be selected/switched between regulating volume or brightness, by pushing some buttons that sit left of the dial knob. The firmware sends an additional byte sequence when in volume control mode. Said byte sequence is missing when the keyboard is set to brightness control mode. This byte sequence is the same on the  `ROCCAT Vulcan 1xx`, `ROCCAT Vulcan TKL` and the `ROCCAT Vulcan TKL PRO` keyboard variants.

So it seems that the state-machine inside Eruption gets confused about the actual mode it is in, because it does not recognize the byte sequence that switches to volume control mode. This most likely happens because the firmware of the device is outdated. Could you please perform a firmware update of your device via the OEM software `ROCCAT Swarm` on a Windows machine, if possible. If this should not be possible for you, then I could add support for the outdated firmware revision to Eruption by changing the behavior of the Eruption hardware drivers for the `ROCCAT Vulcan TKL` and `ROCCAT Vulcan TKL PRO` variants!

LLuth0r, 2021-08-24:
Hi @X3n0m0rph59 !  Thanks for getting back to me so quickly.  So I'm on the latest firmware for my Vulcan TKL, which I updated through Swarm on Windows.  I can dual boot in, but Linux is my primary operating environment.  The button next to the volume control is the mute button.  What happens when I press it is it mutes the volume, but no subsequent turning of the volume control knob adjusts the volume.  So the mute button is only muting and unmuting and not switching the controls of the volume control from brightness adjustment to volume control. 

Is there a different button on the keyboard I should be pressing?  On the Vulcan TKL, next to the volume control knob there is only one button.

X3n0m0rph59, 2021-08-24:
Hi @LLuth0r, thank you! 

> So I'm on the latest firmware for my Vulcan TKL, which I updated through Swarm on Windows

That means that the current code of the Eruption device driver does not work as intended. I will try to fix this, but I currently do not have access to a `ROCCAT Vulcan TKL` keyboard. I am using a `ROCCAT Vulcan Pro TKL` variant, but this one turned out to be different in the past.

> Is there a different button on the keyboard I should be pressing? On the Vulcan TKL, next to the volume control knob there is only one button.

No, on the `TKL` variants of the  keyboards there is only the mute button present. The brightness may be adjusted by using the `FN` + `Cursor UP`  or `FN` + `Cursor DOWN` keys, at least on the `ROCCAT Vulcan Pro TKL` this is the case. Eruption is currently not supporting this, so this will adjust the brightness of the hardware only. It wont adjust the global brightness of Eruption.

I will try to fix the bug without having access to the hardware. Are you willing to build the `master` branch from source and test it out, as soon as I push the fix?

LLuth0r, 2021-08-24:
Hi @X3n0m0rph59, you're welcome!  Yeah, I can absolutely build from master branch from source and test it out.  Sounds like fun!

X3n0m0rph59, 2021-08-24:
@LLuth0r, thank you!

As it turns out, I already added the functionality to adjust the brightness via the keyboard on the `ROCCAT Vulcan Pro TKL`. The Eruption global brightness may be adjusted by using the `FN` + `Cursor UP` or `FN` + `Cursor DOWN` keys, at least it is working on the `Pro TKL` variant.

So apparently there is a difference in the underlying protocol that leads to the `ROCCAT Vulcan TKL` keyboard reporting the wrong operation when using the dial knob.

The `ROCCAT Vulcan Pro TKL` sends the following byte sequence when using the dial knob:

```log
 # rotate right
 |03000b26 01000000|
 |0300cc01 00000000|

 # rotate left
 |03000b27 01000000| 
 |0300ccff 00000000| 
```

This information can be obtained by running Eruption from the command line:
```sh
 # mask the Eruption daemon first, so that it does not get in our way
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

 # run Eruption in debug mode
 $ sudo RUST_BACKTRACE=1 RUST_LOG=debug eruption

 # to run Eruption in trace mode (only FYI, will produce tons of output, not required to run this)
 $ sudo RUST_BACKTRACE=1 RUST_LOG=trace eruption
```

```log
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |42e90000 00000000|                   B.......         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |03000b26 00000000|                   ...&....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |42000000 00000000|                   B.......         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 3 args: 0
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |03000b26 01000000|                   ...&....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |0300cc01 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 4 args: 0
 DEBUG eruption::scripting::script::callbacks      > UserMacros: Executing: 'on_dial_knob_rotate_right'
 DEBUG eruption                                    > FPS: 19
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |42ea0000 00000000|                   B.......         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |03000b27 00000000|                   ...'....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |42000000 00000000|                   B.......         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 3 args: 0
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |03000b27 01000000|                   ...'....         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >   |0300ccff 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_pro_tkl  >                                                          00000008
 DEBUG eruption::scripting::script::callbacks      > Macros: HID event: 4 args: 1
 DEBUG eruption::scripting::script::callbacks      > UserMacros: Executing: 'on_dial_knob_rotate_left'
```

Could you please paste the `DEBUG` log of Eruption running on your system, while rotating the dial knob left and right. I will then update the driver with the respective byte sequence and push out the changes so that you can test them!

LLuth0r, 2021-08-24:
Here's the debug log.  Not sure if my Pulse Audio error is causing interference.

`DEBUG eruption::scripting::script::callbacks > Macros: HID event: 5 args: 0
 DEBUG eruption::scripting::script::callbacks > UserMacros: Executing: 'on_dial_knob_rotate_right'
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02e90000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02000000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 3 args: 0
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |0300cc01 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 5 args: 0
 DEBUG eruption::scripting::script::callbacks > UserMacros: Executing: 'on_dial_knob_rotate_right'
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02e90000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02000000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 3 args: 0
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
`

X3n0m0rph59, 2021-08-24:
@LLuth0r, thanks for providing the logs! I have updated the respective device driver and pushed it to the `master` branch.

X3n0m0rph59, 2021-08-24:
You may want to install the `eruption-git` package on your system, then please clone the repo and build the `master` branch:

```sh
 $ cargo build --release --all 
```

After the build finished successfully, please copy over the binaries:

```sh
$ sudo cp target/release/eruption{,ctl,-netfx,-debug-tool,-hotplug-helper,-gui,-process-monitor} /usr/bin/ && sudo setcap CAP_NET_ADMIN+ep /usr/bin/eruption-process-monitor
```

Then please run this command to start the Eruption daemon:

```sh
 $ sudo systemctl unmask eruption.service && sudo systemctl start eruption.service
```

LLuth0r, 2021-08-24:
Got this when trying to build:

error: failed to run custom build command for `libudev-sys v0.1.4`

Caused by:
  process didn't exit successfully: `/home/pak/Documents/Personal_Projects/eruption/target/release/build/libudev-sys-abbe14258899febd/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: "`\"pkg-config\" \"--libs\" \"--cflags\" \"libudev\"` did not exit successfully: exit status: 1\n--- stderr\nPackage libudev was not found in the pkg-config search path.\nPerhaps you should add the directory containing `libudev.pc'\nto the PKG_CONFIG_PATH environment variable\nPackage 'libudev', required by 'virtual:world', not found\n"', /home/pak/.cargo/registry/src/github.com-1ecc6299db9ec823/libudev-sys-0.1.4/build.rs:38:41
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed

LLuth0r, 2021-08-24:
Never mind, see that it was a dependency issue.  Fixing that now.

LLuth0r, 2021-08-24:
Ok, now it's controlling both at the same time, but it only allows the brightness setting to come up slightly.  I can adjust it through the drop down back to full brightness though.  Here's the new debug log.

ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |0300ccff 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 5 args: 1
 DEBUG eruption::scripting::script::callbacks > UserMacros: Executing: 'on_dial_knob_rotate_left'
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02ea0000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02000000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 3 args: 0
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |0300ccff 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 5 args: 1
 DEBUG eruption::scripting::script::callbacks > UserMacros: Executing: 'on_dial_knob_rotate_left'
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02ea0000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02000000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 3 args: 0
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied

DEBUG eruption::scripting::script::callbacks > UserMacros: Executing: 'on_dial_knob_rotate_right'
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02e90000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >   |02000000 00000000|                   ........         00000000
 DEBUG eruption::hwdevices::roccat_vulcan_tkl >                                                          00000008
 DEBUG eruption::scripting::script::callbacks > Macros: HID event: 3 args: 0
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Pulse Audio error: Could not open Pulse Audio: Access denied
 ERROR eruption::plugins::audio               > Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Access denied

X3n0m0rph59, 2021-08-25:
@LLuth0r, thanks for trying this out!

Could you please always attach to full log output, starting from the first line until the last?
In the log dump you provided I don't see any volume level changes. It only registers mute button and brightness events.

LLuth0r, 2021-08-25:
Yes absolutely!  Though it’ll be a few days before I can get back to this.
Traveling for work tomorrow through Saturday.  I’ll record the log when I
get back Saturday evening or on Sunday.

X3n0m0rph59, 2021-09-01:
@LLuth0r, when rotating the dial knob Eruption will display a volume indicator on the keyboard, showing the current master volume level. The change in the level of brightness you are seeing could be caused by Eruption being unable to query the master volume. It would then just display an overlay showing a volume level of 0. If that is the case the brightness will return to the previous state after a short while (less than 2 seconds).

LLuth0r, 2021-09-01:
Hey, sorry was gone for last week on work travel.  Here's a full log:

Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.23 (0.1.22.r30.g12b5651) (release build)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found supported device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x2fee (0001:0006:00) - ROCCAT ROCCAT Vulcan TKL
 INFO  eruption::hwdevices::roccat_vulcan_tkl > Bound driver: ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x2fee - ROCCAT ROCCAT Vulcan TKL
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2d60 - ROCCAT ROCCAT Kain 200 Aimo Dongle
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2d60 - ROCCAT ROCCAT Kain 200 Aimo Dongle
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1e7d:0x2d60 - ROCCAT ROCCAT Kain 200 Aimo Dongle
 DEBUG eruption::hwdevices                    > Found unsupported device: 0x1532:0x531 - Razer Inc Razer Seiren Mini
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 1.15
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: /var/lib/eruption/profiles/matrix.profile
 DEBUG eruption::dbus_interface               > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.338", Serial: 2, Args: [":1.338"] }
 DEBUG eruption::dbus_interface               > Message: Message { Type: Signal, Path: "/org/freedesktop/DBus", Interface: "org.freedesktop.DBus", Member: "NameAcquired", Sender: "org.freedesktop.DBus", Destination: ":1.338", Serial: 3, Args: ["org.eruption"] }
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/solid.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/ghost.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/wave.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/afterhue.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/macros.lua
 DEBUG eruption                               > Switch successful
[... cut ...]
 INFO  eruption                               > Shutting down all Lua VMs now...
^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^C^[ WARN  eruption                               > Timed out while waiting for a Lua VM to shut down, terminating now
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 INFO  eruption                               > Closing all devices now...
 DEBUG eruption                               > Saving state...
 INFO  eruption                               > Exiting now

X3n0m0rph59, 2021-09-02:
@LLuth0r, thank you for the full logs. I have pushed out new commits on the `master` branch. It would be nice if you could try out the new changes.

LLuth0r, 2021-09-02:
@X3n0m0rph59 Absolutely, will pull down and build.  Will give you a heads up afterwards.  Give me like an hour, have a meeting.

LLuth0r, 2021-09-03:
Looks like we're back in business!  Full volume control with the knob.  The button next to the knob acts as the mute button again.

X3n0m0rph59, 2021-09-03:
@LLuth0r Excellent! Thanks for your help!
I will close this issue now, please feel free to open up issues anytime if you encounter further problems.

closed 2021-09-03 by X3n0m0rph59

mp3bruh, 2025-12-05:
still got the same problem, updating or resetting did nothing :(