eruption-project/eruption #80: Support for Roccat Kain 200/202 AIMO

nenominal, 2021-08-31:
**Introduction**
Do you plan to add support for Roccat Kain 200/202 wireless mouse?

**Describe The Problem To Be Solved**

I'm more interested in battery detection so I can know when I need to recharge the mouse. 
Usually I disable all the RGB crap so battery last longer.

X3n0m0rph59, 2021-09-01:
Hi @nenominal, thank you for your feature request!

I currently do not have access to a `ROCCAT Kain 200/202` wireless mouse, so I am unable to reverse the protocol at this time. I will consider adding support for querying battery levels as soon as I get hold of such a device.

Do you see the battery levels when using `upower`?
```sh
 $ upower --dump
```

nenominal, 2021-09-01:
Hi,

unfortunately there is no output regarding battery using upower.

```
upower -d

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         no
  updated:              Mi 01 sep 2021 14:18:47 +0300 (-10299 seconds ago)
  has history:          no
  has statistics:       no
  unknown
    warning-level:       none
    icon-name:          'battery-missing-symbolic'

Daemon:
  daemon-version:  0.99.11
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: HybridSleep
```

If you need assistance, I can help, unfortunately I'm a sysadmin, not a developer.

X3n0m0rph59, 2021-09-02:
@nenominal to add support for the device I need a trace of the packets on the USB bus that initialize the device, set RGB values, query battery levels, etc... Please find more documentation on how to capture such a trace file in the [Eruption Wiki](/X3n0m0rph59/eruption/wiki).

nenominal, 2021-09-02:
https://easyupload.io/d5iuic

Hope it helps. 
Just remove the zip extension from the file name, it's not an archive. 

Bellow are the device's details. I captured on all usbports, for some reason wireshark didn't work on only one port but if I selected all of them, it worked.

```
T:  Bus=01 Lev=01 Prnt=01 Port=07 Cnt=03 Dev#=  8 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=32 #Cfgs=  1
P:  Vendor=1e7d ProdID=2d60 Rev=02.66
S:  Manufacturer=ROCCAT
S:  Product=ROCCAT Kain 200 Aimo Dongle
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
I:  If#=0x1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#=0x2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```

X3n0m0rph59, 2021-09-03:
@nenominal Thank you for providing the capture file!

It seems that this capture file does not contain the full device initialization sequence. By glancing over the data, I can only see RGB values (?) being set. Please be sure to capture the USB traffic while hotplugging the device, so that ROCCAT Swarm performs the full device initialization.

nenominal, 2021-09-03:
I captured as much as I can, see bellow new file. Just remove .zip extension 
Hope it helps.

https://easyupload.io/8sbtjh

X3n0m0rph59, 2021-09-03:
@nenominal Thanks again for providing the capture file!

I am still unable to locate the device init sequence. Would it be possible to capture the packets that ROCCAT Swarm generates while performing a reset to factory settings? I suspect that this will reset the embedded controller and trigger a full device initialization.

nenominal, 2021-09-03:
https://easyupload.io/71ezd5

It includes 2 factory resets and different settings applied in swarm.

X3n0m0rph59, 2021-09-06:
@nenominal Thanks for providing the files! Seems like the device uses a totally redesigned protocol. Will take me a while to reverse this. I am working on it...

X3n0m0rph59, 2021-09-14:
Hi @nenominal, I have implemented a driver with a basic set of features for the `ROCCAT Kain 200/202` series wireless mice.

The `master` branch contains all necessary commits to enable support for RGB lighting and device status queries.
Eruption `0.1.23` will ship with a newly introduced companion utility called `eruption-hwutil` that can be used to get or set device configuration parameters and query the device status without the Eruption daemon required to be running.

To show the device status please run the following commands:

```sh
$ sudo eruption-hwutil list
$ sudo eruption-hwutil -r status <index of Kain 200 mouse>
```

It currently supports reading the signal strength, and maybe battery levels.

It would be nice if you could give this a try!

nenominal, 2021-09-14:
Wow, thanks man :) I'll give it a try tomorrow and you'll have my feed-back.

nenominal, 2021-09-15:
I failed to build the master build, bellow the errors I got:
```

cargo build --all --release
   Compiling evdev-sys v0.2.3
   Compiling mlua v0.6.2
   Compiling num_cpus v1.13.0
   Compiling atty v0.2.14
   Compiling thread-id v4.0.0
   Compiling signal-hook-registry v1.4.0
   Compiling getrandom v0.2.3
   Compiling getrandom v0.1.16
   Compiling iovec v0.1.4
   Compiling net2 v0.2.37
   Compiling inotify-sys v0.1.5
   Compiling filetime v0.2.15
   Compiling time v0.1.43
   Compiling gethostname v0.2.1
   Compiling memoffset v0.6.4
   Compiling miniz_oxide v0.4.4
   Compiling indexmap v1.7.0
   Compiling libudev-sys v0.1.4
   Compiling libdbus-sys v0.2.1
   Compiling aho-corasick v0.7.18
   Compiling object v0.26.2
   Compiling nix v0.16.1
   Compiling libpulse-simple-sys v1.19.1
   Compiling bstr v0.2.16
   Compiling x11 v2.18.2
   Compiling futures-channel v0.3.17
   Compiling tracing-log v0.1.2
   Compiling mio v0.7.13
   Compiling paste-impl v0.1.18
   Compiling hidapi v1.2.6 (https://github.com/ruabmbua/hidapi-rs.git?branch=master#7faa5744)
   Compiling lockfile v0.3.0
   Compiling crossbeam-channel v0.5.1
   Compiling crossbeam-queue v0.3.2
   Compiling error-chain v0.12.4
   Compiling procmon-sys v0.0.2 (/home/zulu/eruption/eruption-process-monitor/src/procmon-sys)
   Compiling lexical-core v0.7.6
   Compiling semver v0.11.0
   Compiling syn v1.0.76
   Compiling proc-macro-error-attr v1.0.4
   Compiling png v0.16.8
   Compiling bitvec v0.22.3
   Compiling find-crate v0.6.3
   Compiling procinfo v0.4.2
   Compiling num-integer v0.1.44
   Compiling num-traits v0.1.43
   Compiling approx v0.5.0
   Compiling num-complex v0.4.0
   Compiling colored v2.0.0
   Compiling inotify v0.7.1
   Compiling eruption v0.1.23 (/home/zulu/eruption/eruption)
   Compiling rand_core v0.6.3
   Compiling rand_core v0.5.1
   Compiling crossbeam-epoch v0.9.5
   Compiling nix v0.22.1
   Compiling mio v0.6.23
   Compiling libudev v0.2.0
   Compiling udev v0.6.2
   Compiling petgraph v0.5.1
   Compiling regex v1.5.4
   Compiling paste v0.1.18
   Compiling syslog v5.0.0
   Compiling rustc_version v0.3.3
   Compiling backtrace v0.3.61
   Compiling chrono v0.4.19
   Compiling num-rational v0.3.2
   Compiling num-iter v0.1.42
[... 667 lines cut ...]
    = note: see issue #78835 <https://github.com/rust-lang/rust/issues/78835> for more information
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to 32 previous errors

Some errors have detailed explanations: E0658, E0710.
For more information about an error, try `rustc --explain E0658`.
error: could not compile `sysinfo`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error[E0658]: use of unstable library feature 'osstring_ascii'
   --> /home/zulu/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-3.0.0-beta.4/src/parse/matches/matched_arg.rs:130:19
    |
130 |                 v.eq_ignore_ascii_case(val)
    |                   ^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #70516 <https://github.com/rust-lang/rust/issues/70516> for more information

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0658`.
error: build failed

```

X3n0m0rph59, 2021-09-15:
@nenominal Looks like an outdated version of Rust.

To build the Eruption `master` branch you need at least the stable version of `rustc`. Currently this would be version `1.55.0`.
Please refer to: https://rustup.rs/

nenominal, 2021-09-16:
Ok, I've updated my rustc version and I was able to build from master.

```
sudo eruption-hwutil list

Index: 01: ID: 1e7d:2d60 ROCCAT/ROCCAT Kain 200 Aimo Dongle

```
```
sudo eruption-hwutil -r status 01

Index: 01: ID: 1e7d:2d60 ROCCAT/ROCCAT Kain 200 Aimo Dongle
Bound driver: ROCCAT Kain 2xx AIMO
Sending device init sequence...
Polling device status...

Battery Level:   100% (513)
Signal Strength: 69% (13824)

Battery Level:   100% (513)
Signal Strength: 69% (13824)
```

I'm not sure if the battery level is correct as swarm shows it as 60%.

My bet is that battery level should be switched with signal strength

X3n0m0rph59, 2021-09-17:
@nenominal 

>I'm not sure if the battery level is correct as swarm shows it as 60%.
My bet is that battery level should be switched with signal strength

You are right, definitely seems to be incorrect. Seems that I picked the wrong field for the battery level read out.
Hope to have something ready soon...

X3n0m0rph59, 2021-09-17:
@nenominal, I have pushed out another round of commits on the `master` branch!

It would be nice if you could give this a try again!

nenominal, 2021-09-17:
Still needs fixing, now it show like this

sudo eruption-hwutil -r status 01

Index: 01: ID: 1e7d:2d60 ROCCAT/ROCCAT Kain 200 Aimo Dongle

Transceiver enabled:  true
Signal Strength:       66% (13314)
Battery Level:        **unknown** (65)

The signal strength is still read from wrong field and the battery level shows unknown but the level is correct, I think.

X3n0m0rph59, 2021-09-17:
@nenominal Thanks again for testing!

> Still needs fixing, now it show like this

Yes, I see... I now have full access to a `ROCCAT Kain 200 AIMO` mouse device so testing will be much easier for me. I have updated the code to use the following mapping table:

| Value | Percent |
| -----------| -----------|
| 71 | fully charged |
|65..70 | 100% |
|64 | 80% |
|63 | 60% |
|62 | 40% |
|61 | 20% |
|60 | 0% |
| others |  unknown |

Will take a few days to see if this table is correct. I have to complete a full charge/discharge cycle of the device.

>The signal strength is still read from wrong field and the battery level shows unknown but the level is correct, I think.

I believe the signal strength readings to be correct. I determined the scaling factor by placing the transceiver directly beneath the mouse. When positioned this way the indicator should reach 100% signal strength. I could apply a correction factor and show a descriptive text to more closely mimic the behavior of ROCCAT Swarm.
The signal strength should be around 60% - 80% when the transceiver is placed about 40cm away from the mouse.

>Also, If I want to change some profiles from the Gnome extension, the daemon crash. Do you want me to open another ticket for this error?

Yes, please open up another issue for this. Please post full logs! A full backtrace of the crash would be very helpful:
```sh
$ sudo RUST_BACKTRACE=full eruption
```

X3n0m0rph59, 2021-09-18:
Okay, I have gathered some more data and updated the mapping table respectively:

I have seen the following raw values from the `ROCCAT Kain 200` until now: 71, 64, 65.

| Raw Value | ROCCAT Swarm shows |
| -- | -- |
| 71 | 100% |
| 64 | 80% |
| 65 | 60% |
| 66 | 40% |
| 67 | 20% |
| 68 | 0% |
| others | unknown |

nenominal, 2021-09-18:
Ok, I've tested with new build and now it seems the battery level is correct.

On the other hand I still have some doubts about signal strength as it varies a lot, never goes up 62-63%.
Swarm shows 100% and I think it's right as the dongle is at 10 cm from mouse itself.

Also, is there a way to set the DPI?

X3n0m0rph59, 2021-09-18:
> Ok, I've tested with new build and now it seems the battery level is correct.

Excellent!

> On the other hand I still have some doubts about signal strength as it varies a lot, never goes up 62-63%.
Swarm shows 100% and I think it's right as the dongle is at 10 cm from mouse itself.

It seems that ROCCAT Swarm applies a correction factor. I will look into this...

> Also, is there a way to set the DPI?

No, not currently. but I am planning to implement this as well.

nenominal, 2021-09-18:
Maybe add the battery level in the gnome's extension dropdown or show it in some place to be easy noticeable?

X3n0m0rph59, 2021-09-22:
@nenominal, I have released a new version of the [`Eruption Profile Switcher`](https://github.com/X3n0m0rph59/eruption-profile-switcher) GNOME Shell Extension that features battery level and signal strength indicators (where applicable).

[image]

[image]

nenominal, 2021-09-22:
Yey, nice, I'll give it a try. Did the extension got to extentions.gnome.org ?

If it's not very complicated I'd suggest to have the battery value in the taskbar, if it's possible of-course (and maybe the wireless signal on a later build if it will be implemented).

X3n0m0rph59, 2021-09-25:
@nenominal 

> Did the extension got to extentions.gnome.org

Yes I published it on extensions.gnome.org (v18)

> If it's not very complicated I'd suggest to have the battery value in the taskbar, if it's possible of-course (and maybe the wireless signal on a later build if it will be implemented).

Sounds good! I will implement this next

nenominal, 2021-09-25:
I think you have publish it only for Gnome 40 and up, I'm using Ubuntu 20.04 witch is running Gnome 3.36 and I only see v.15

[image]

Nopraz, 2021-12-04:
Hello, is it possible to have the battery indicator on something like KDE ? I think plasma uses upower to fetch this data and I don't know if it's possible for you to make the battery data show up on upower ?

X3n0m0rph59, 2021-12-05:
Hi @Nopraz, last time I looked into supporting Plasma / writing a Plasmoid for KDE I got stuck with implementing the D-Bus based protocol to Eruption. I will look into this again as soon as I find the time...

As you mentioned in your post, the best solution clearly would be to implement this stuff in upower.