eruption-project/eruption #81: Support for Kain 100

CypherDragon, 2021-09-02:
**Introduction**
I'd like to request support for the Kain 100 AIMO wired mouse to be added. I own this mouse and can assist if info is needed.

**Describe The Problem To Be Solved**
Support of additional hardware

**Describe the Solution**
Add ability for eruption to control LEDs (on the mouse wheel on the Kain 100), DPI, switching, etc

CypherDragon, 2021-09-02:
Just read the wiki on collecting the usb pcap - I'll work on that tomorrow and upload it as soon as I can.

X3n0m0rph59, 2021-09-02:
Hi @CypherDragon, thanks for offering your help!

I am interested in adding support for the `ROCCAT Kain 100 AIMO` mouse to Eruption. As you already noticed, I need a capture of the USB traffic between `ROCCAT Swarm` and the device, mainly the init sequence and a few samples where the RGB colors are being set.

X3n0m0rph59, 2022-03-25:
@ndren Thank you very much! This is looking good!

Will try to implement a driver for the `ROCCAT Kain 100 AIMO` using your capture files.

I believe that the angle snapping feature can be disabled by using the OEM software (`ROCCAT Swarm`). If there is no specific control for it, then maybe it can be disabled by performing a factory reset of the device (using `ROCCAT Swarm`).

X3n0m0rph59, 2022-03-26:
Hi @ndren, @CypherDragon!

I have implemented a driver for the `ROCCAT Kain 100` and pushed it to the `develop` branch. If you would like to try it out, please compile from source using this [documentation](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md). Please remember to switch to the `develop` branch before running `make`.

After a successful build, please run the `eruption-debug-tool`:

```shell
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
sudo target/release/eruption-debug-tool list
```

Please find the index of the `ROCCAT Kain 100` and then run:

```shell
sudo target/release/eruption-debug-tool -v run-tests <your-dev-index>
```

You should see the device being initialized, and light up in red and then blue.

After that to re-enable `eruption.service`, please run:
```shell
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

ndren, 2022-03-26:
Tested `eruption-debug-tool`, and it works as intended! Thank you.

X3n0m0rph59, 2022-03-27:
@ndren Great! Thank you very much for testing this!

The same build of Eruption should then be able to initialize your mouse and set colors to it. Could you please run eruption in `debug` mode, and tell me what happens? Please post the full output!

* Does setting the LED colors work too with eruption (change profiles)?
* Do all the mouse keys work?

Please run this command for testing the eruption daemon:
```shell
sudo RUST_LOG=debug RUST_BACKTRACE=full target/release/eruption -c support/config/eruption-dev.conf
```

I am a little uncertain whether the 'wait after sending a USB feature report' source code is correct and working. It probably might fail there... let's see.

ndren, 2022-03-31:
@X3n0m0rph59 hey! Sorry for the delay, I didn't get any notification.
I've gone through and everything works OK as far as I can tell (tested mouse buttons and then changed profiles and brightness in eruption-gui). The only thing that did not work was the DPI button, but this is not something that I would use.

Button tests: [debug-log.txt](https://github.com/X3n0m0rph59/eruption/files/8393156/debug-log.txt)
Profiles and brightness: [eruption-gui-log.txt](https://github.com/X3n0m0rph59/eruption/files/8393157/eruption-gui-log.txt)

Dworv, 2022-09-06:
I have set the driver maturity level to `experimental`, but eruption still can't find my mouse. Is this the right place to discuss this?

X3n0m0rph59, 2022-09-13:
Hi @Dworv, sorry for not answering you earlier...

> I have set the driver maturity level to experimental, but eruption still can't find my mouse. Is this the right place to discuss this?

Yes this is the right place to ask! Apparently Eruption fails to recognize your mouse because the matching of device IDs fails. Do you own a supported mouse device?

Could you please provide USB IDs using the following command:

```sh
sudo lsusb
```

Dworv, 2022-09-14:
Here's the Roccat one:
`Bus 005 Device 003: ID 1e7d:2d00 ROCCAT ROCCAT Kain 100`

I definitely own a supported mouse, at least according to your list of mice. I'm on the most recent version too.

Dworv, 2022-09-24:
Hey it looks like its working now! Awesome software, my mouse colors are back to being on theme.
[image]