eruption-project/eruption #82: Eruption GUI for MATE desktop

CypherDragon, 2021-09-02:
**Introduction**
I would like to request creation of a GUI utility for eruption similar to the GNOME desktop extension, but for the MATE desktop environment. I unfortunately don't know enough to be able to try and convert the existing GNOME extension to MATE, but more than willing to learn with a bit of guidance.

**Describe The Problem To Be Solved**
Add support for GUI configuration in desktop environments other than GNOME

**Describe the Solution**
GUI to enable control of eruption settings in MATE

X3n0m0rph59, 2021-09-02:
Hi @CypherDragon, thank you for your feature request!

I will look into writing an extension for the MATE desktop environment. In the meantime you could try out the `eruption-git` package, that provides a technology preview version of the `Eruption GUI`.

CypherDragon, 2021-09-02:
I didn't even notice that...give it a try and looks awesome! It may only be a tech preview, but the `Eruption GUI` looks like exactly what I'm asking for :)