eruption-project/eruption #83: Create installation script

CypherDragon, 2021-09-02:
**Introduction**
I'd like to request creation of an install script to make installation of eruption easier. I am willing to assist with this request, with a little guidance. I am familiar with shell scripting, and I know a little bit of python, but haven't ever worked on an install script that ties into compiling an app before.

**Describe The Problem To Be Solved**
Make the install process easier for users that may not be devs.

**Describe the Solution**
Install script that can be run after compiling the drivers

X3n0m0rph59, 2021-09-02:
@CypherDragon there is already documentation available that could be converted to some kind of `install.sh` script relatively easily. Please find it here: [INSTALL.md](/X3n0m0rph59/eruption/blob/master/INSTALL.md)

CypherDragon, 2021-09-02:
Yup, I was able to get eruption installed and working by following the [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/INSTALL.md). I did copy the mkdir and cp commands into a quick shell.sh so I didn't have to do it manually. I'm relatively new to this kind of project though, so I wasn't sure if there would be more to the install process that could be scripted. If not, I'll work up a better looking script to submit, if you'd like. Honestly not sure how this kind of thing works, since I'm used to just creating simple scripts for myself, and not submitting them to existing/active projects

X3n0m0rph59, 2021-09-02:
@CypherDragon, the procedure would be to fork the Eruption repository, add the commits to your fork that make up the installation script, updated documentation, etc...  and then please submit a pull request.

X3n0m0rph59, 2022-02-22:
@CypherDragon 

Commit a661ced95a93b5125e81362dfe0edfaf5b7850ea introduces a Makefile based installation script that also supports uninstallation too.

To install Eruption please run:
```sh
make
sudo make install
```

To remove Eruption from your system:
```sh
sudo make uninstall
```

closed 2022-02-22 by X3n0m0rph59

CypherDragon, 2022-02-22:
My apologies @X3n0m0rph59 , I switched over to Arch shortly after our discussion, and the AUR already had a makefile. Thank you for the update though! I'll take a look at the commit to see how a makefile should be created for future knowledge.