eruption-project/eruption #86: Unable to build project

DirkSchueler, 2021-09-13:
**Introduction**
Building is documented as install.md begins with "cargo build --all --release", but leads to several 
errors on my device, although it went fine a few weeks ago.

**Please describe the bug**
cargo build --all --release
    Updating crates.io index
  Downloaded wyz v0.4.0
  Downloaded bitvec v0.22.3
  Downloaded lockfile v0.3.0
  Downloaded crc8 v0.1.1
  Downloaded 4 crates (229.8 KB) in 0.48s
   Compiling crc8 v0.1.1
   Compiling wyz v0.4.0
   Compiling lockfile v0.3.0
   Compiling eruption v0.1.23 (/home/dirk/data/eruption/eruption)
   Compiling chrono v0.4.19
   Compiling clap v3.0.0-beta.4
   Compiling rust-pulsectl v0.2.6 (https://github.com/X3n0m0rph59/pulsectl.git?branch=master#09c36e6c)
   Compiling libpulse-simple-binding v2.24.1
   Compiling dbus-tokio v0.7.4
error[E0658]: arbitrary expressions in key-value attributes are unstable
 --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-3.0.0-beta.4/src/lib.rs:8:10
  |
8 | #![doc = include_str!("../README.md")]
  |          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: see issue #78835 <https://github.com/rust-lang/rust/issues/78835> for more information

   Compiling sysinfo v0.20.3
   Compiling glib-sys v0.14.0
   Compiling tiff v0.6.1
   Compiling bitvec v0.22.3
   Compiling phf v0.9.0
error[E0710]: an unknown tool name found in scoped lint: `rustdoc::broken_intra_doc_links`
  --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/sysinfo-0.20.3/src/lib.rs:13:9
   |
13 | #![deny(rustdoc::broken_intra_doc_links)]
   |         ^^^^^^^

error[E0658]: arbitrary expressions in key-value attributes are unstable
  --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/sysinfo-0.20.3/src/lib.rs:10:10
   |
10 | #![doc = include_str!("../README.md")]
.
.
.
.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error[E0658]: use of unstable library feature 'osstring_ascii'
   --> /home/dirk/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-3.0.0-beta.4/src/parse/matches/matched_arg.rs:130:19
    |
130 |                 v.eq_ignore_ascii_case(val)
    |                   ^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #70516 <https://github.com/rust-lang/rust/issues/70516> for more information

error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0658`.
error: build failed

**To Reproduce**
Steps to reproduce the behavior:
Call cargo build --all --release ....

**Expected behavior**
Succesfull compilation

**Logs**
Please provide logs: `sudo journalctl -u eruption.service -b`

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * Eruption version [e.g. 0.1.20]
 * Your device(s) (`sudo lsusb`)

**Additional context**
Add any other context about the problem here.

X3n0m0rph59, 2021-09-13:
Hi @DirkSchueler, could you please try to update your version of Rust? Eruption requires the current stable version of `rustc` to successfully compile. Currently this would be version `1.55.0`

https://rustup.rs/

DirkSchueler, 2021-09-13:
Quiet easy solution! Thank you!

DirkSchueler, 2021-11-23:
After Pulling the current Repo 
cargo build --all --release
led to 
"error: failed to load manifest for workspace member `/home/dirk/data/eruption/eruption`

Caused by:
  failed to parse manifest at `/home/dirk/data/eruption/eruption/Cargo.toml`

Caused by:
  feature `edition2021` is required

Updating Rust to version 1.56.1 solved that, but 
cargo build --all --release 
now leads to
...
   Compiling i18n-embed v0.13.0
The following warnings were emitted during compilation:

warning: `"pkg-config" "--libs" "--cflags" "gtksourceview-4" "gtksourceview-4 >= 4"` did not exit successfully: exit status: 1

error: failed to run custom build command for `sourceview4-sys v0.2.0 (https://gitlab.gnome.org/World/Rust/sourceview4-rs.git?branch=master#dd6189dc)`

X3n0m0rph59, 2021-11-23:
@DirkSchueler, thanks for reporting this issue!

The problem can be fixed by installing the packages `gtksourceview4` and/or `gtksourceview-4.0-dev`.
I have updated the [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md) respectively.

closed 2021-11-23 by X3n0m0rph59

DirkSchueler, 2021-11-23:
Sorry, but 

sudo apt install libusb-1.0-0-dev libhidapi-dev libevdev-dev libudev-dev libdbus-1-dev \
  libpulse-dev luajit libluajit-5.1-dev libx11-dev libxrandr-dev libgtk-3-dev libgdk-pixbuf2.0-dev \
  libatk1.0-dev libpango1.0-dev libcairo2-dev libgtksourceview-4.0-dev
on my Ubuntu System
gives 
E: Unable to locate package libgtksourceview-4.0-dev
E: Couldn't find any package by glob 'libgtksourceview-4.0-dev'

and

sudo apt install libusb-1.0-0-dev libhidapi-dev libevdev-dev libudev-dev libdbus-1-dev   libpulse-dev luajit libluajit-5.1-dev libx11-dev libxrandr-dev libgtk-3-dev libgdk-pixbuf2.0-dev   libatk1.0-dev libpango1.0-dev libcairo2-dev gtksourceview4
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package gtksourceview4

X3n0m0rph59, 2021-11-23:
@DirkSchueler 

Please try to build using the following command:
```shell
cargo build --release --all --no-default-features
```

This will disable the syntax highlighting in the Eruption GUI since your version of Ubuntu does not provide the package required for it.

alkaris2, 2022-02-18:
build fails because it requires resolver which is nightly feature, but I can't get it to switch to nightly even though it's installed as default with rustup as a required dependency.

```
error: failed to parse manifest at `/var/tmp/pamac-build-alkaris/eruption/src/eruption/Cargo.toml`

Caused by:
  feature `resolver` is required

  this Cargo does not support nightly features, but if you
  switch to nightly channel you can add
  `cargo-features = ["resolver"]` to enable this feature
```

X3n0m0rph59, 2022-02-18:
Hi @alkaris2,

could you please try to update your installation of Rust (and the cargo binary) using:
```sh
$ rustup self update
$ rustup update
$ rustup default nightly
```

If the build still fails, please post the output of
```sh
$ cargo -V
```

commit cbdd482a (2021-11-23): Update documentation

Closes #86