eruption-project/eruption #87: Polluted logs with pulse audio connection

nenominal, 2021-09-24:
Do I need to do extra something to get rid of this errors?

I'm using Ubuntu 20.04 x64.

```
Sep 24 20:49:33 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:33 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:34 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:34 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:34 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:34 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:35 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:35 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:35 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:35 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:36 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:38 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:38 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:38 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:38 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:39 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:39 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:40 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:40 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:41 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:41 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:42 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:42 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:43 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:43 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:44 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:44 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:44 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:44 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:45 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:45 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:46 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:46 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:47 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:47 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:48 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:48 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:50 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:50 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:51 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:51 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:53 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:53 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:53 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:53 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:55 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:55 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:55 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:55 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:56 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:56 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:58 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:49:58 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:50:02 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:50:02 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:50:02 Nautilus eruption[999]: Could not initialize the audio backend: Pulse Audio error: Could not open Pulse Audio: Connection refused
Sep 24 20:50:02 Nautilus eruption[999]: Pulse Audio error: Could not open Pulse Audio: Connection refused

```

X3n0m0rph59, 2021-09-25:
Hi @nenominal, currently it is required for audio to work with Eruption that the steps in [this document](../blob/master/docs/SETUP_AUDIO.md) are performed.

I am working on architectural changes to Eruption that will result in not having to perform these changes to the audio configuration.

X3n0m0rph59, 2021-11-13:
I will close this issue now. Please reopen this issue if the problem persists.

closed 2021-11-13 by X3n0m0rph59