eruption-project/eruption #92: Mint 20.2 / Roccat Burst Pro - "Error opening the mouse device"

aznlir88, 2021-11-07:
Hello, sorry if that's something on my side but I do not know what may it be as I tried to run eruption as root and it runs into similiar problem.

I installed eruption via your PPA, version 0.1.22. When running it from console I get into errors telling me that somehow Eruption cannot open my mouse device:

Device information (lsusb):
`Bus 003 Device 002: ID 1e7d:2de1 ROCCAT ROCCAT Burst Pro`

```
ERROR eruption                              > Error opening the mouse device: Could not open the device file
ERROR eruption                              > This could be a permission problem, or maybe the device is locked by another process?
```
and later on:
`ERROR eruption                              > Could not send the LED map to the device: Device not opened`

I attach log file from `sudo journalctl -u eruption.service -b`
[eruption.log](https://github.com/X3n0m0rph59/eruption/files/7492214/eruption.log)

aznlir88, 2021-11-07:
Tried to solve it a bit and here is what I tried to do:

I discovered that my mouse device is:
`/dev/input/mouse0`
Also added some info to `/etc/eruption/eruption.conf`:
```
[[devices]]
 entry_type = "device"
 device_class = "serial"
 device_name = "ROCCAT Burst Pro"
 device_file = "/dev/input/mouse0"
```
Yet it changed nothing, and as the device `/dev/input/mouse0` belonged to group `input` I added my user to this group - now I get the same error as I got when run as root:
```
 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 () (release build)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 INFO  eruption::hwdevices            > Binding non-pnp serial LEDs device: ROCCAT Burst Pro (/dev/input/mouse0)
 INFO  eruption::hwdevices::custom_serial_leds > Bound driver: Adalight Custom Serial LEDs
 INFO  eruption::hwdevices                     > Found supported mouse device: 0x1e7d:0x2de1 (0003:0002:00) - <unknown> <unknown>
 INFO  eruption::hwdevices::roccat_burst_pro   > Bound driver: ROCCAT Burst Pro
 INFO  eruption                                > Opening mouse device...
 ERROR eruption                                > Error opening the mouse device: Could not open the device file
 ERROR eruption                                > This could be a permission problem, or maybe the device is locked by another process?
 INFO  eruption                                > Initializing mouse device...
 ERROR eruption                                > Could not initialize the device: Device not opened
 INFO  eruption                                > Configuring mouse LEDs...
 ERROR eruption                                > Could not initialize LEDs: Device not opened
 INFO  eruption                                > Firmware revision: <unknown>
 INFO  eruption                                > Spawning mouse input thread...
 INFO  eruption                                > Opening misc device...
 INFO  eruption                                > Now listening on mouse: /dev/input/event2
 INFO  eruption                                > Input device name: "ROCCAT ROCCAT Burst Pro"
 INFO  eruption                                > Input device ID: bus 0x3 vendor 0x1e7d product 0x2de1
 INFO  eruption                                > Physical location: usb-0000:00:12.0-1/input0
 INFO  eruption                                > Grabbing the mouse device exclusively
The application panicked (crashed).
Message:  Could not grab the device, terminating now.: Os { code: 16, kind: Other, message: "Device or resource busy" }
Location: eruption/src/main.rs:494

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
Aborted (core dumped)
```

Problem still not solved.

X3n0m0rph59, 2021-11-08:
Hi @aznlir88, it seems that the run of Eruption from the provided `eruption.log` was successful, except for the yet to be configured audio stuff. Did you see the LEDs being set to a different color on your mouse device at all? Please try out multiple profiles since not all of them support setting colors of the LEDs on mouse devices.

The Eruption daemon needs to be run as the `root` user. The `[[devices]]` section in the configuration file is only needed to add non-plug and play devices. Please undo the respective changes made to `eruption.conf`.

The Eruption daemon will be started via a `udev` rule that ultimately triggers the `eruption.service` systemd unit (via `eruption-hotplug-helper`). Maybe the Eruption daemon has already been running, locking the mouse device exclusively.

Please run these commands to disable the `eruption.service` and then run Eruption as root:
```sh
 $ sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
 $ sudo RUST_LOG=debug eruption
```

Please post the full debug log that Eruption prints to the console.

To enable the `eruption.service` again, please run:
```sh
$ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

aznlir88, 2021-11-08:
Hello, about LEDs changing color, yes - it goes to all(mouse wheel and the shell itself) into purple after running daemon. Also it starts automatically in my case so yes, it is being started by the "root" account.

Ok, I commented these lines from `eruption.conf` as they were before.

I attach the `eruption2.log` with the output you asked for - I just stopped it some time after as it can carry on like that forever only 
[eruption2.log](https://github.com/X3n0m0rph59/eruption/files/7500605/eruption2.log)
FPS value change.

X3n0m0rph59, 2021-11-09:
@aznlir88 Great! So is it working for you now? From the logs I can see that the only problem left is the audio related stuff.

In case you own a supported keyboard and want to use the spectrum analyzer profile with it, please find the documentation on how to set up audio here: [SETUP_AUDIO.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/SETUP_AUDIO.md)

The FPS (frames per second) value changes dynamically and may even go down to zero if there is no update required. E.g. if only a solid color is being displayed it may reach zero. This is done to reduce CPU load and reduce power consumption.

aznlir88, 2021-11-09:
To be honest, I dunno. I mean, I use fluxbox and I do not have these fancy cinnamon dialog boxes etc. is there any guy for eruption so I can set everything or it is more complex to start with? About keyboard - ASUS ROG Strix Flare - not sure if supported.

aznlir88, 2021-11-09:
I checked, it does not work, the problem is the same as it was, that device is busy and after starting service it crashes:
```
INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 () (release build)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2de1 (0003:0002:00) - <unknown> <unknown>
 INFO  eruption::hwdevices::roccat_burst_pro > Bound driver: ROCCAT Burst Pro
 INFO  eruption                              > Opening mouse device...
 ERROR eruption                              > Error opening the mouse device: Could not open the device file
 ERROR eruption                              > This could be a permission problem, or maybe the device is locked by another process?
 INFO  eruption                              > Initializing mouse device...
 ERROR eruption                              > Could not initialize the device: Device not opened
 INFO  eruption                              > Configuring mouse LEDs...
 ERROR eruption                              > Could not initialize LEDs: Device not opened
 INFO  eruption                              > Firmware revision: <unknown>
 INFO  eruption                              > Spawning mouse input thread...
 INFO  eruption                              > Device enumeration completed
 INFO  eruption                              > Performing late initializations...
 INFO  eruption                              > Initializing D-Bus API...
 INFO  eruption                              > Late initializations completed
 INFO  eruption                              > Startup completed
 INFO  eruption                              > Switching to profile: /var/lib/eruption/profiles/blue-fx-swirl-perlin.profile
 INFO  eruption                              > Loading Lua script: /usr/share/eruption/scripts/swirl-perlin.lua
 INFO  eruption                              > Loading Lua script: /usr/share/eruption/scripts/solid.lua
 INFO  eruption                              > Loading Lua script: /usr/share/eruption/scripts/halo.lua
 INFO  eruption                              > Loading Lua script: /usr/share/eruption/scripts/impact.lua
 INFO  eruption                              > Loading Lua script: /usr/share/eruption/scripts/macros.lua
 ERROR eruption                              > Could not process a D-Bus event: receiving on an empty and disconnected channel
 INFO  eruption                              > Now listening on mouse: /dev/input/event2
 INFO  eruption                              > Input device name: "ROCCAT ROCCAT Burst Pro"
 INFO  eruption                              > Input device ID: bus 0x3 vendor 0x1e7d product 0x2de1
 INFO  eruption                              > Physical location: usb-0000:00:12.0-1/input0
 INFO  eruption                              > Grabbing the mouse device exclusively
The application panicked (crashed).
Message:  Could not grab the device, terminating now.: Os { code: 16, kind: Other, message: "Device or resource busy" }
Location: eruption/src/main.rs:494

Backtrace omitted.
Run with RUST_BACKTRACE=1 environment variable to display it.
Run with RUST_BACKTRACE=full to include source snippets.
Aborted (core dumped)
```

X3n0m0rph59, 2021-11-12:
@aznlir88 I have set up a VM running Linux Mint 20 and tested the packages from the PPA. Device hotplug does not work as intended. Otherwise it seems to be working though.

Could you please run the following commands:
```sh
$ sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service

$ eruptionctl status profile
$ eruptionctl switch profile solid.profile
```

This should display the currently active profile and make the mouse show a solid red color.

If it still does not work then please post these logs:
```sh
$ sudo journalctl -u eruption -b
```

Yes, there is a GUI for Eruption but it's still in the technology preview phase (it is not ready for production). It is only shipped in the `eruption-git` package.

aznlir88, 2021-11-13:
Hmm, guess you are right and it works as I ended with solid red color, here is the output:
```
azn@zux-pc:~$ eruptionctl status profile
Current profile: /var/lib/eruption/profiles/blue-fx-swirl-perlin.profile
azn@zux-pc:~$ eruptionctl switch profile solid.profile
Switching to profile: /var/lib/eruption/profiles/solid.profile
```
Guess the issue can be closed, thank you very much for your help :). I will surely look into the GUI.

X3n0m0rph59, 2021-11-13:
Great! I will close this issue now.

closed 2021-11-13 by X3n0m0rph59