eruption-project/eruption #93: Support for Roccat Ryos MK Pro

C0rn3rphr34k, 2021-11-23:
**Introduction**
Hi and thank you for contributing such a great piece of software to the community.
I have a bunch, of mostly older, Roccat devices flying around my home (Ryos MK Pro, Suora, Isku, Kova 2016), but my daily driver for converting my thoughts to text is my Ryos MK Pro keyboard. Would you be willing to add support for older devices to eruption?
Let me know what you need and i can grab device infos, test and probably contribute a little to the code (although I'm not writing in Rust myself, I'm a gopher ;) )

**Describe The Problem To Be Solved**
Ryos MK Pro is not detected by eruption service/daemon

**Describe the Solution**

X3n0m0rph59, 2021-11-23:
Hi @C0rn3rphr34k, thank you for your request!

Yes, I am interested in adding support for older hardware to Eruption too. Some/most of the older hardware does not support realtime lighting effects like ROCCAT AIMO though, which kind of diminishes the purpose of adding support for them to Eruption. Such devices only support enabling hardware generated lighting effects where only some of the parameters can be controlled by software. Eruption is based on the assumption though, that a device allows full control of the RGB(A) or at least monochromatic color values in realtime.

But even devices without any LEDs can be relevant to Eruption too... I would consider hardware like the `ROCCAT Nyth`, for which some (currently dysfunctional) code exists such a device, because of the extra buttons that can only be used with extra software support.

As long as these conditions are met, I believe that it does make sense to add support for even an older device to Eruption.

What kind of features do you have in mind specifically when you request support for the hardware from Eruption? Support for macro keys, or mostly lighting/LED control?

X3n0m0rph59, 2021-11-23:
Regarding how to help to add support for a device to Eruption, please refer to the [Eruption Wiki](https://github.com/X3n0m0rph59/eruption/wiki).

Basically, it boils down to capturing the USB traffic between the device and the OEM provided driver software. With this information I can implement a driver module for that device in Eruption.

X3n0m0rph59, 2021-11-23:
@C0rn3rphr34k 

Does the `ROCCAT Ryos MK Pro` (non FX) variant support `Talk FX` and/or the `ROCCAT Swarm` software? If this is the case then it allows for realtime control of the monochromatic LED values and it would make sense to add support for it.

The `ROCCAT Elo Air 7.1` is working in the current `master` branch of Eruption. I believe that I got the battery levels and LED lighting working, if I recall correctly.

The `ROCCAT Kova 2016` seems to be a `ROCCAT AIMO` compatible device... Is that correct? If so, then it definitely makes sense to add it to Eruption!

> Thx for the Link to the Wiki. I'll try to set up the necessary things for capturing the USB traffic tonight.

Great, you're welcome! And thank you too, for offering your help!

C0rn3rphr34k, 2021-11-23:
You are right. I do not believe that the Ryos MK Pro supports TalkFX/Swarm. At least the keyboard is not recognized in Swarm when i boot into Win 10.

The Kova and Elo do show up though. I'll try the other two (Isku and Suora) later tonight.

C0rn3rphr34k, 2021-11-23:
Alrighty, I've managed to capture all 4 input devices using your excellent instructions from the wiki. :)

For the Ryos MK Pro and the Isku I've used the old non-Swarm driver Packages that are still available via the roccat support site. As they are not Swarm compatible I'd totally understand it if you're not going to try and reverse engineer the drivers.

The Suora and Hova are both Swarm devices and I've tried to use click every option available in the Swarm UI for those devices, so that the capture is mostly complete.

Let me know if you need any more (specific) clicks or tests. :+1

Have a great day/night ;)

[c0rn3rphr34k_roccat_isku_driver-1.22_fw-1.25.zip](https://github.com/X3n0m0rph59/eruption/files/7591453/c0rn3rphr34k_roccat_isku_driver-1.22_fw-1.25.zip)
[c0rn3rphr34k_roccat_kova-2016_driver-1.18.zip](https://github.com/X3n0m0rph59/eruption/files/7591454/c0rn3rphr34k_roccat_kova-2016_driver-1.18.zip)
[c0rn3rphr34k_roccat_ryos-mk-pro_driver-1.41_fw-1.26.zip](https://github.com/X3n0m0rph59/eruption/files/7591455/c0rn3rphr34k_roccat_ryos-mk-pro_driver-1.41_fw-1.26.zip)
[c0rn3rphr34k_roccat_suora_driver-1.10.zip](https://github.com/X3n0m0rph59/eruption/files/7591456/c0rn3rphr34k_roccat_suora_driver-1.10.zip)
: