eruption-project/eruption #95: Add Support for ROCCAT Kova 2016

X3n0m0rph59, 2021-11-24:
This issue will track the progress made on supporting the `ROCCAT Kova 2016` mouse.

Please refer to #93 for previous posts (pcap files)

X3n0m0rph59, 2021-11-24:
@C0rn3rphr34k thank you very much! Seems like this device does not support realtime updates to the LED colors... The capture file looks very good though! It seems to contain nearly everything I need to get this up and running.

I am currently unable to tell exactly what packets set the colors of the LEDs, so could you please capture setting the static colors red, green and blue to the LEDs in its own pcapng file? (The init sequence is not required anymore)
Please use the mode `fully lit` and no hardware effects: `effects off`.

 * Please set front and back LEDs to different colors, so that I can differentiate the LED indices
 * Please tell me in which order you set the front and back LEDs to what color.
 
You may want to use the comment packet feature of Wireshark to annotate packets!

And in a different pcapng file please capture just the clicks of each button (most importantly, the DPI switch buttons) and please tell me in which order you pressed them. With this information I will be able to tell if I can just keep the current HID event processing code or if there are some changes required to it.

C0rn3rphr34k, 2021-11-24:
Hey, thank you for your quick reply. I'll capture the two additional pcapng files from the Kova tonight.
fyi: the dpi switch is a single button that cycles through the 5 variable value DPI slots. I'll try to comment the right packets when setting the DPI values through Swarm and when cycling through those slots.

C0rn3rphr34k, 2021-11-24:
I've created 3 pcapng files one for each:
- Set front and back LED colors
- Press every button once (after applying a factory reset)
- Cycle through the 5 dpi slots (using Button number 10 from the included screenshot)

As you can see from the included screenshots from Swarm, the buttons have two layers, the secondary function is only accessible when a button has EasyShift assigned as it's primary function. Then you hold the EasyShift button and click the button with the desired secondary function.

Just in case the comments in the pcapng file didn't work:

LED color assignement order:

1. Front led: Red
2. Back led: Red
3. Front led: Green
4. Back led: Green
5. Front led: Blue
6. Back led: Blue

Button clicks:
I've clicked the 12 buttons in ascending order

DPI switcher:
I've used Button 10 for this (primary function DPI cycle) starting at dpi slot 2 the first click switched to slot 3, then onward to slot 5, the wraparound to slot 1 and ended at slot 3.

Hope that helps. Let me know if you need anything else.

[c0rn3rphr34k_roccat_kova-2016_leds_buttons_dpi-cycle.zip](https://github.com/X3n0m0rph59/eruption/files/7598815/c0rn3rphr34k_roccat_kova-2016_leds_buttons_dpi-cycle.zip)

X3n0m0rph59, 2021-11-24:
@C0rn3rphr34k thanks for your excellent support, it is much appreciated!

To test the new code, please build the current `master` from source and then run the `eruption-debug-tool`:
```shell
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
sudo eruption-debug-tool list
```

Please pick the index of `Subdev 0` of the `ROCCAT Kova 2016` and then run:

```shell
sudo eruption-debug-tool run-tests <your-dev-index>
```

You should see the device being initialized, and light up in red and then blue.

After that to re-enable `eruption.service`, please run:
```shell
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

C0rn3rphr34k, 2021-11-24:
Awesome :+1:

That works great. I'm attaching a video of the test run.

`sudo eruption-debug-tool list

Please find the device you want to debug below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

Index: 00: ID: 1e7d:3232 ROCCAT/ROCCAT Ryos MK Pro Subdev: 0
Index: 01: ID: 1e7d:3232 ROCCAT/ROCCAT Ryos MK Pro Subdev: 1
Index: 02: ID: 1e7d:2cee ROCCAT/ROCCAT Kova Subdev: 0
Index: 03: ID: 1e7d:2cee ROCCAT/ROCCAT Kova Subdev: 1

Enumeration of HID devices completed

Special devices

Index: 255: Serial Port 1 (/dev/ttyACM0)
Index: 254: Serial Port 2 (/dev/ttyACM1)
Index: 253: Serial Port 3 (/dev/ttyACM2)
Index: 252: Serial Port 4 (/dev/ttyACM3)

`

`sudo eruption-debug-tool -v run-tests 02
Index: 02: ID: 1e7d:2cee ROCCAT/ROCCAT Kova Subdev: 0
Bound driver: ROCCAT Kova 2016
Sending device init sequence...
Step 1
Sending control device feature report
Waiting for control device to respond...
Step 2
Sending control device feature report
Waiting for control device to respond...
Step 3
Sending control device feature report
Waiting for control device to respond...
Setting LEDs from supplied map...
Setting LEDs from supplied map...

`

https://user-images.githubusercontent.com/16436816/143323714-4aace6b3-ea9f-4a1a-aec4-acb7f3683e31.mp4

X3n0m0rph59, 2022-03-05:
Hi @C0rn3rphr34k, I have published new `eruption-git` packages, that ship with support for the `ROCCAT Kova 2016`.

You will need to set the configuration parameter `driver_maturity_level = "experimental"` in `/etc/eruption/eruption.conf`, if you want to try out the new build.

C0rn3rphr34k, 2022-03-07:
Hi @X3n0m0rph59 !

Thanks for reaching out about the new update. I already installed it, I believe, on Saturday. It's working fine so far.
Would you like me to test out anything specific on the new driver package?