eruption-project/eruption #96: evdev input by path disappears on ROCCAT Vulcan 120 Aimo

RolledR, 2021-11-24:
**Description**
Before installing eruption, `/dev/input/by-id` had several inputs for the keyboard. After installing eruption, most of them disappeared, and no new ones were added to replace them.

**To Reproduce**
1. Uninstall eruption
2. `ls /dev/input/by-id` and see the `usb-ROCCAT*` ids
3. Install eruption
4. `ls /dev/input/by-id` and see most `us-ROCCAT*` ids have disappeared, and none of the remaining yield key presses.

**Expected behavior**
New links are created for the new virtual input(s) (or old ones remain accessible).

**Logs**
_Uninstalled:_
```
ls -l /dev/input/by-id
lrwxrwxrwx  10 root 24 Nov 14:28 usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-if01 -> ../event11
lrwxrwxrwx  10 root 24 Nov 11:29 usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-if02 -> ../event10
lrwxrwxrwx   9 root 24 Nov 11:29 usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-kbd -> ../event6
lrwxrwxrwx   9 root 24 Nov 14:28 usb-ROCCAT_ROCCAT_Vulcan_AIMO-if01-event-mouse -> ../event7
lrwxrwxrwx   9 root 24 Nov 14:28 usb-ROCCAT_ROCCAT_Vulcan_AIMO-if01-mouse -> ../mouse0
lrwxrwxrwx  10 root 24 Nov 14:28 usb-ROCCAT_ROCCAT_Vulcan_AIMO-if03-event-kbd -> ../event28
```
`cat /dev/input/by-id/usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-kbd` will spam gibberish when keys are pressed

_Installed:_
```
ls -l /dev/input/by-id
lrwxrwxrwx  10 root 24 Nov 11:29 usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-if02 -> ../event10
lrwxrwxrwx   9 root 24 Nov 11:29 usb-ROCCAT_ROCCAT_Vulcan_AIMO-event-kbd -> ../event6
```
cat-ing those does nothing.
Querying the cat-able input that eruption creates gives this, which I guess isn't enough for evdev to create a link by itself:
```
udevadm info --query=all --name=/dev/input/event27
P: /devices/virtual/input/input81/event27
N: input/event27
L: 0
E: DEVPATH=/devices/virtual/input/input81/event27
E: DEVNAME=/dev/input/event27
E: MAJOR=13
E: MINOR=91
E: SUBSYSTEM=input
E: USEC_INITIALIZED=11330540161
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_SERIAL=noserial
```

**Info:**
 * Version: 0.1.22
 * Device: `Bus 007 Device 006: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO`

**Additional context**
I'm using the evdev inputs to pass a mouse + keyboard through to a windows VM for reduced latency, and I can't do that (easily) with eruption installed. I need to provide the path to the inputs, but the actual /dev/input/eventXX are not meant to be consistent and there's no link in `/dev/input/by-id`. [Example of what I mean..](https://wiki.archlinux.org/title/PCI_passthrough_via_OVMF#Passing_keyboard/mouse_via_Evdev)

X3n0m0rph59, 2021-12-07:
Hi @RolledR, thanks for opening this issue!

A fix for this problem will ship with Eruption `0.1.23`. I have modified the file `/usr/lib/udev/rules.d/99-eruption.rules` and added the following lines:

```shell
# Stable links to Eruption virtual devices

ACTION=="add", ATTRS{name}=="Eruption Virtual Keyboard", SYMLINK+="input/by-id/eruption-virtual-keyboard"
ACTION=="add", ATTRS{name}=="Eruption Virtual Mouse", SYMLINK+="input/by-id/eruption-virtual-mouse"
```

RolledR, 2021-12-07:
Just tested it out. Works perfectly. Thanks for the quick fix!

closed 2021-12-07 by RolledR

JonLit, 2023-10-11:
How would I set up the passthrough to the VM with that?
I get this Error:
```
Error starting domain: internal error: process exited while connecting to monitor: 2023-10-11T23:43:48.024483Z qemu-system-x86_64: /dev/input/by-id/eruption-virtual-mouse: is not an evdev device
```

Also sorry if this doesn't really belong here...