eruption-project/eruption #97: Add support for Roccat Kone Pro Air

hirvoin, 2021-12-01:
**Introduction**
Hi! I recently got Roccat Kone Pro Air wireless mouse. Noticed that there is no way to adjust DPI or see battery level, but then I discovered this great software. 

**Describe The Problem To Be Solved**
Support for Roccat Kone Pro Air is missing

**Describe the Solution**
Add support for Roccat Kone Pro Air

Here's the capture file. I was not able to install Roccat Swarm on VirtualBox so I did the capture on Windows machine. 
Hopefully you can see the required information!

[kone_air_pro_init.zip](https://github.com/X3n0m0rph59/eruption/files/7636942/kone_air_pro_init.zip)

X3n0m0rph59, 2021-12-06:
Hi @hirvoin, thank you very much for creating the capture file!

The protocol seems to be rather different from that of the other supported wireless mice (`ROCCAT Kain 200/202 AIMO`). It will take some time to figure this out completely without me having access to the hardware.

On first glance over the entries in the pcapng file, I can't tell what packets convey the RGB color values. Did you capture the packets while the mouse was set to AIMO mode? If not, then could you please create another capture file that mostly contains RGB packets? I need a few seconds worth of traffic to tell what bytes constitute the RGB data.

Thanks again for your assistance!

n1coh, 2021-12-13:
Hi, thank you @hirvoin for capture file and @X3n0m0rph59 for all this project.
Yesterday the update to the version 0.1.22.r186 crash the service :
`eruption[8862]: Lua error in file /usr/share/eruption/scripts/lib/failsafe.lua: runtime error: /usr/share/eruption/scripts//lib/declarations.lua:71: module 'hwdevices/mice/roccat_kone_pro_air' not found:
                                                     no field package.preload['hwdevices/mice/roccat_kone_pro_air']
                                                     no file './hwdevices/mice/roccat_kone_pro_air.lua'
                                                     no file '/usr/share/luajit-2.0.5/hwdevices/mice/roccat_kone_pro_air.lua'
                                                     no file '/usr/local/share/lua/5.1/hwdevices/mice/roccat_kone_pro_air.lua'
                                                     no file '/usr/local/share/lua/5.1/hwdevices/mice/roccat_kone_pro_air/init.lua'
                                                     no file '/usr/share/lua/5.1/hwdevices/mice/roccat_kone_pro_air.lua'
                                                     no file '/usr/share/lua/5.1/hwdevices/mice/roccat_kone_pro_air/init.lua'
                                                     no file '/usr/share/eruption/scripts//lib/hwdevices/mice/roccat_kone_pro_air'
                                                     no file '/usr/share/eruption/scripts//lib/hwdevices/mice/roccat_kone_pro_air.lua'
                                                     no file './hwdevices/mice/roccat_kone_pro_air.so'
                                                     no file '/usr/local/lib/lua/5.1/hwdevices/mice/roccat_kone_pro_air.so'
                                                     no file '/usr/lib/lua/5.1/hwdevices/mice/roccat_kone_pro_air.so'
                                                     no file '/usr/local/lib/lua/5.1/loadall.so'
                                             stack traceback:
                                                     [C]: in function 'require'
                                                     /usr/share/eruption/scripts//lib/declarations.lua:71: in function 'load_support_scripts'
                                                     /usr/share/eruption/scripts//lib/declarations.lua:75: in main chunk
                                                     [C]: in function 'require'
                                                     [string "eruption/src/scripting/script.rs:846:21"]:16: in main chunk
                                                     UnknownError
`
For now I've blacklisted the mouse and this is fine.
I'll be glad to help you.

X3n0m0rph59, 2022-03-19:
I have just added basic LED support for the `ROCCAT Kone Pro Air` to the `master` branch!

Connection via cable should be working as well as the connection via the included RF-transceiver. Bluetooth connection is entirely untested.

X3n0m0rph59, 2022-03-19:
Since the driver is feature gated as `experimental`, you will have to change `/etc/eruption/eruption.conf`. Please set `driver_maturity_level = "experimental"` if you want to try the new driver. This will allow the eruption daemon to bind `experimental` drivers.

n1coh, 2022-03-23:
I just tested the master branch and I must say that it reacts well in RF and wired. I don't use bluetooth but I tested it and it doesn't work as expected.
Thank you for adding this mouse.

X3n0m0rph59, 2022-03-26:
Hi @n1coh, @hirvoin!

I have added support for signal strength reporting and battery status to the `master` branch. Bluetooth most like is still not working.

esiqveland, 2022-07-16:
> Hi @n1coh, @hirvoin!
> 
> I have added support for signal strength reporting and battery status to the `master` branch. Bluetooth most like is still not working.

I am trying to get this to work for my Kone Pro Air device `1e7d:2c8e`, but it is not recognized:

```
$ eruptionctl devices list
Dumping Eruption managed devices list

Keyboard devices:
<No supported devices detected>

Mouse devices:
<No supported devices detected>

Miscellaneous devices:
<No supported devices detected>
```

```
$ eruption --version
Eruption 0.1.24 (0.1.23.alpha.r171.g3e096b36) (release build)
```

X3n0m0rph59, 2022-07-17:
Hi @esiqveland, the driver for the `ROCCAT Kone Pro Air` is still experimental.

Did you try to enable experimental drivers in `/etc/eruption/eruption.conf`?
```toml
[global]
# ....
driver_maturity_level = "experimental"
```

Please note that the quality of the driver for the `ROCCAT Kone Pro Air` currently is not the best... You can make it crash pretty easily e.g. by cycling trough the hardware profiles using the button on the bottom of the device and then clicking some mouse buttons. Bluetooth mode is not implemented, but RF and cable modes are confirmed to work!

esiqveland, 2022-07-17:
I had actually set `experimental`, but I forgot to restart the service. 

```
$ eruptionctl devices list
Dumping Eruption managed devices list

Keyboard devices:
<No supported devices detected>

Mouse devices:
00: ROCCAT Kone Pro Air Dongle

Miscellaneous devices:
<No supported devices detected>
```

```
$ eruption-hwutil list     

Please find the device you want to address below and use its respective
index number (column 1) as the device index for the other sub-commands of this tool

01: <unknown>/<unknown>
04: <unknown>/<unknown>
05: <unknown>/<unknown>
09: <unknown>/<unknown>
```

Thank you for your hard work, this is an enormous project!