eruption-project/eruption #99: eruption-audio-proxy.service does not exist after fresh install

MrYdse, 2021-12-04:
**Introduction**
After installing eruption via AUR, it seems that eruption-audio-proxy is missing.

**Please describe the bug**
After install, eruption-audio-proxy is missing, causing the error `Failed to enable unit: Unit file eruption-audio-proxy.service does not exist.` after running `systemctl --user enable --now eruption-audio-proxy.service`

**To Reproduce**
Steps to reproduce the behavior:
1. Install eruption via pacman or yay
2. Try the step in AUDIO.md: `systemctl --user enable --now eruption-audio-proxy.service`
3. Get the error: `Failed to enable unit: Unit file eruption-audio-proxy.service does not exist.`

**Expected behavior**
The service is enabled and the audio device is proxied to eruption.

**Logs**
```
-- Journal begins at Sat 2021-12-04 20:48:38 CET, ends at Sat 2021-12-04 21:45:54 CET. --
Dec 04 21:27:50 EY-pc-arch systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Dec 04 21:27:50 EY-pc-arch systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Dec 04 21:27:50 EY-pc-arch eruption[596]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 (0.1.22.r0.gf0cb968) (release build)
Dec 04 21:27:50 EY-pc-arch eruption[596]:  INFO  eruption > Loading saved state...
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption > Registering plugins...
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Sensors - Query system sensor values
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Audio - Audio related functions
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugin_manager > Registering plugin: Animal - Simulation of organic movements (support library)
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption::plugins::persistence > Loading persistent state data from disk...
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption                       > Plugins loaded and initialized successfully
Dec 04 21:27:52 EY-pc-arch eruption[596]:  INFO  eruption                       > Enumerating connected devices...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x3098 (0001:0003:00) - ROCCAT ROCCAT Vulcan AIMO
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x AIMO
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Opening keyboard device...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Initializing keyboard device...
Dec 04 21:27:53 EY-pc-arch systemd[1]: Stopping Linux user-mode input and LED driver for keyboards, mice and other devices...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Configuring keyboard LEDs...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Firmware revision: 136
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Spawning keyboard input thread...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Device enumeration completed
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Performing late initializations...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Initializing D-Bus API...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Late initializations completed
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Startup completed
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Switching to profile: /var/lib/eruption/profiles/blue-fx-swirl-perlin.profile
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Now listening on keyboard: /dev/input/event6
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Input device name: "ROCCAT ROCCAT Vulcan AIMO"
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Input device ID: bus 0x3 vendor 0x1e7d product 0x3098
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Physical location: usb-0000:2a:00.1-2/input0
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Grabbing the keyboard device exclusively
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/swirl-perlin.lua
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/solid.lua
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/halo.lua
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/impact.lua
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/macros.lua
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Shutting down all Lua VMs now...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
Dec 04 21:27:53 EY-pc-arch eruption[596]:  INFO  eruption                               > Closing all devices now...
Dec 04 21:27:54 EY-pc-arch eruption[596]:  INFO  eruption                               > Exiting now
Dec 04 21:27:54 EY-pc-arch systemd[1]: eruption.service: Deactivated successfully.
Dec 04 21:27:54 EY-pc-arch systemd[1]: Stopped Linux user-mode input and LED driver for keyboards, mice and other devices.
Dec 04 21:27:54 EY-pc-arch systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Dec 04 21:27:54 EY-pc-arch systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Dec 04 21:27:54 EY-pc-arch eruption[852]:  INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.22 (0.1.22.r0.gf0cb968) (release build)
Dec 04 21:27:54 EY-pc-arch eruption[852]:  INFO  eruption > Loading saved state...
[... 59 lines cut ...]
 * Eruption version [e.g. 0.1.20]
 ```
2 aur/eruption-git 0.1.22.r28.g1ba264c-1 (+1 0.16) (Installed)
    Linux user-mode input and LED driver for keyboards, mice and other devices
```
 * Your device(s) (`sudo lsusb`)
```
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 046d:c07d Logitech, Inc. G502 Mouse
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0951:16ae Kingston Technology DT microDuo 3C
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID abcd:1234 LogiLink UDisk flash drive
Bus 003 Device 004: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 1462:7c84 Micro Star International MYSTIC LIGHT 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 003: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
Bus 001 Device 002: ID 046d:0836 Logitech, Inc. B525 HD Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
**Additional context**
Add any other context about the problem here.

X3n0m0rph59, 2021-12-05:
Hi @MrYdse, thank you for opening this issue!

It seems that you are running Eruption version `0.1.22` but the audio-proxy feature was only recently introduced with Eruption version `0.1.23` (not yet released). You may want to build `master` from source ([HowTo](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md)) if you don't want to resort to the rather invasive changes that are required for Eruption version `0.1.22`.

[SETUP_AUDIO.md docs from Eruption v0.1.21](https://github.com/X3n0m0rph59/eruption/blob/releases/v0.1.21/SETUP_AUDIO.md)

I will try to find some time to make a new release of Eruption `0.1.23` soon...

Please feel free to contact me if you need further assistance

X3n0m0rph59, 2021-12-05:
I will close this issue now, please open a new issue anytime if you encounter further problems.

closed 2021-12-05 by X3n0m0rph59