eruption-project/eruption #100: Add Support for <make and name of the device goes here>

jimmon89, 2021-12-17:
**Introduction**
I will be adding the required info for my new "Roccat Horde AIMO Keyboard" as well as my new "Roccat Kova AIMO Mouse" once i finish downloading the windows 10 ISO and set up the VM

**Device to add support for**
`
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1e7d:303e ROCCAT ROCCAT HORDE AIMO
Bus 001 Device 003: ID 0d8c:0012 C-Media Electronics, Inc. USB Audio Device
Bus 001 Device 002: ID 28de:1142 Valve Software Wireless Steam Controller
Bus 001 Device 005: ID 1e7d:2cf1 ROCCAT ROCCAT Kova Aimo
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

**Please attach your *.pcapng file, if possible**
coming soon

X3n0m0rph59, 2022-02-25:
Duplicate of #22

X3n0m0rph59, 2022-02-25:
Hi @jimmon89, does this device work for you with a current release of Eruption?

The USB ID matches that of the already supported `Roccat Kova AIMO` mouse. Your capture file seems to match the current device driver code as well. If the device does not work, could you please provide logs of Eruption running in debug mode?

```sh
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
sudo RUST_LOG=debug eruption

sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

jimmon89, 2022-02-26:
it seems to be working, now i just have to wrap my head around re-creating my macros from the roccat swarm software saved to my mouse in Lua

X3n0m0rph59, 2022-02-26:
@jimmon89, great that it is working!

Feel free to open up a new issue or discussion if you should require assistance in creating Lua macros. You can find some documentation [here](https://github.com/X3n0m0rph59/eruption/blob/master/docs/MACROS.md). Please let me know where you find the currently available documentation to be lacking and where it could be improved.

X3n0m0rph59, 2022-02-26:
I will close this issue now since it is a duplicate.
Please follow up in issue #22.