eruption-project/eruption #109: Big latency for eaqch keystroke since updtate to Eruption 0.1.23 (0.1.22.r290.g07bb1ec) (release build)

SHU-red, 2022-02-22:
**Introduction**
Everything was running fine until the update to Eruption 0.1.23 (0.1.22.r290.g07bb1ec) (release build) came yesterday

**Please describe the bug**
If i type something there is a big delay
Holding down a button causes a very long time after releasing where the input is still firing

**To Reproduce**
Steps to reproduce the behavior:
Everything i type

**Expected behavior**
A clear and concise description of what you expected to happen.
No input lag when typing

**Logs**
Please provide logs: `sudo journalctl -u eruption.service -b`
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: timed out waiting on send operation
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: timed out waiting on send operation
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: timed out waiting on send operation
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: More than 82 milliseconds of jitter detected!
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: This means that we dropped at least one frame
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: Loop took: 208 milliseconds, goal: 50
Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: timed out waiting on send operation

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * Eruption version Eruption 0.1.23 (0.1.22.r290.g07bb1ec) (release build)
 * Your device(s) (`sudo lsusb`) Bus 001 Device 005: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO

**Additional context**
Add any other context about the problem here.

System:
  Kernel: 5.15.24-1-MANJARO x86_64 bits: 64 compiler: gcc v: 11.2.0
    Desktop: awesome 4.3-1355-gbd776c980 vt: 1 dm: SDDM Distro: Manjaro Linux
    base: Arch Linux

X3n0m0rph59, 2022-02-22:
Hi @SHU-red, thank you for reporting this issue. I am currently in the process of trying to reproduce this bug inside a Manjaro VM.

Could you please trigger the bug and post the full output of this command:
```sh
sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf
```
This will provide some insights on where the problem lies specifically.

> Feb 22 20:08:20 -manjaro eruption[4308]: -manjaro eruption[4308]: timed out waiting on send operation

Looks like a thread panicked somewhere earlier. Could be something going on with the new audio-proxy code.

Please post the full output of this command too:
```sh
systemctl --user status eruption-audio-proxy.service eruption-process-monitor.service
```
> NOTE: Please don't place sudo in front of the command. It has to act on the user instance of systemd.

SHU-red, 2022-02-22:
No Problem i'll do what i can
Hopefully i dont waste your time to trace to a very different root-cause (as always, several updates came in in parralel)

> sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf

```
 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.23 (0.1.22.r290.g07bb1ec) (release build)
 INFO  eruption > Using only drivers that are marked as stable (default)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugins::macros > Initializing Linux virtual input devices
 INFO  eruption::plugin_manager  > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager  > Registering plugin: Linux ULEDs - Linux Userspace LEDs interface
 INFO  eruption::plugins::uleds  > The ULEDs subsystem is not available on this kernel
 INFO  eruption::plugin_manager  > Registering plugin: SDK Support - Support for the Eruption SDK
 INFO  eruption::plugin_manager  > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager  > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported device: 0x46a:0xb103 - <unknown> USB Device
 DEBUG eruption::hwdevices            > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x3098 (1-13:1.0) - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption                               > Opening keyboard device...
 ERROR eruption                               > Error opening the keyboard device: Could not open the device file
 ERROR eruption                               > This could be a permission problem, or maybe the device is locked by another process?

```

> systemctl --user status eruption-audio-proxy.service eruption-process-monitor.service

```

○ eruption-audio-proxy.service - Audio proxy daemon for the Eruption Linux user-mode dri>
     Loaded: loaded (/usr/lib/systemd/user/eruption-audio-proxy.service; disabled; vendo>
     Active: inactive (dead)
       Docs: man:eruption-audio-proxy(1)
             man:audio-proxy.conf(5)
             man:eruptionctl(1)

○ eruption-process-monitor.service - Process Monitoring and Introspection for Eruption
     Loaded: loaded (/usr/lib/systemd/user/eruption-process-monitor.service; disabled; v>
     Active: inactive (dead)
       Docs: man:eruption-process-monitor(1)
             man:process-monitor.conf(5)
             man:eruptionctl(1)

```

X3n0m0rph59, 2022-02-22:
Ah, please run this command first:
```sh
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service 
```

So that this will actually be able to access the hardware:
```sh
sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf
```

SHU-red, 2022-02-22:
I think it worked.
I just pushed some random buttons during this log.
Seems that eruption restarts after stopping it with ctr+c
Before it automatically restarts, i have full typing speed again ... so problem only occures during eruption active in background

```

❯ sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf
 INFO  eruption > Starting Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices: Version 0.1.23 (0.1.22.r290.g07bb1ec) (release build)
 INFO  eruption > Using only drivers that are marked as stable (default)
 INFO  eruption > Loading saved state...
 INFO  eruption > Registering plugins...
 INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
 INFO  eruption::plugins::macros > Initializing Linux virtual input devices
 INFO  eruption::plugin_manager  > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager  > Registering plugin: Linux ULEDs - Linux Userspace LEDs interface
 INFO  eruption::plugins::uleds  > The ULEDs subsystem is not available on this kernel
 INFO  eruption::plugin_manager  > Registering plugin: SDK Support - Support for the Eruption SDK
 INFO  eruption::plugin_manager  > Registering plugin: Audio - Audio related functions
 INFO  eruption::plugin_manager  > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence > Loading persistent state data from disk...
 INFO  eruption                       > Plugins loaded and initialized successfully
 INFO  eruption                       > Enumerating connected devices...
 DEBUG eruption::hwdevices            > Found unsupported device: 0x46a:0xb103 - <unknown> USB Device
 DEBUG eruption::hwdevices            > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices            > Found supported keyboard device: 0x1e7d:0x3098 (1-13:1.0) - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption::hwdevices::roccat_vulcan_1xx > Bound driver: ROCCAT Vulcan 100/12x AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 DEBUG eruption::hwdevices                    > Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
 INFO  eruption                               > Opening keyboard device...
 INFO  eruption                               > Initializing keyboard device...
 INFO  eruption                               > Configuring keyboard LEDs...
 INFO  eruption                               > Firmware revision: 1.36
 INFO  eruption                               > Spawning keyboard input thread...
 INFO  eruption                               > Device enumeration completed
 INFO  eruption                               > Performing late initializations...
 INFO  eruption                               > Loading saved device state...
 DEBUG eruption::state                        > 0x1e7d:0x3098: Brightness: 100
 INFO  eruption                               > Initializing Linux Userspace LEDs interface...
 INFO  eruption                               > Initializing D-Bus API...
 DEBUG eruption                               > Registering filesystem watcher...
 INFO  eruption                               > Late initializations completed
 INFO  eruption                               > Startup completed
 DEBUG eruption                               > Entering the main loop now...
 INFO  eruption                               > Switching to profile: /var/lib/eruption/profiles/matrix.profile
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/solid.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/ghost.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/wave.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/afterhue.lua
 INFO  eruption                               > Loading Lua script: /usr/share/eruption/scripts/macros.lua
 DEBUG eruption                               > Switch successful
 DEBUG eruption::scripting::script            > Applying parameter Color { name: "color_background", description: "Background color", min: None, max: None, default: 4294905872 }
 DEBUG eruption::profiles                     > Using value from .profile file for config param 'color_background' (value: '4279303952') [5]
 DEBUG eruption::scripting::script            > Applying parameter Float { name: "opacity", description: "Opacity value in the range 0.0 .. 1.0, where 1.0 is fully opaque", min: Some(0.0), max: Some(1.0), default: 1.0 }
[... cut ...]
^C DEBUG eruption                               > Left the main loop
 INFO  eruption                               > Shutting down all Lua VMs now...
 INFO  eruption::plugins::persistence         > Storing persistent state data to disk...
 INFO  eruption                               > Saving global runtime state...
 DEBUG eruption::state                        > 0x1e7d:0x3098: Brightness: 100
 INFO  eruption                               > Closing all devices now...
 INFO  eruption                               > Exiting now

```

X3n0m0rph59, 2022-02-22:
> Hopefully i dont waste your time to trace to a very different root-cause (as always, several updates came in in parralel)

That would not be a problem. This definitely is a bug with Eruption.

This one causes the perceived delay:

> ERROR eruption                               > timed out waiting on send operation

>WARN  eruption                               > More than 82 milliseconds of jitter detected!
 WARN  eruption                               > This means that we dropped at least one frame
 WARN  eruption                               > Loop took: 210 milliseconds, goal: 50

A huge delay...

X3n0m0rph59, 2022-02-23:
I just pushed out some performance improvement fixes. I have tested this on Arch, Fedora 35 and Manjaro Linux. The situation was especially pronounced on Manjaro. The huge input lag should be reduced now.

@SHU-red Could you please test this on your system?

SHU-red, 2022-02-23:
OK i just installed your update and the latency is gone.
But it seems that my "custom" setup isnt there anymore ... do i have to unmask eruption again or something?

Not knowing much about that ...

X3n0m0rph59, 2022-02-23:
> OK i just installed your update and the latency is gone.

OK, great!

> But it seems that my "custom" setup isnt there anymore ... do i have to unmask eruption again or something?

Profiles that are shipped with Eruption are not marked as conffiles and will be overwritten by the package manager.
Customized profiles with filenames distinct from the profiles shipped by `eruption` or `eruption-git` will not be overwritten.

X3n0m0rph59, 2022-02-23:
> do i have to unmask eruption again or something?

```sh
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

SHU-red, 2022-02-23:
Thanks!
That was it!
My Profile immediately went active again BUT! The latency is there again!

Is there something wrong with my custom setup then?
Its a long time ago where i digged into eruption and made my setup.

Not sure if this is possible, because i did nothing too fancy and until now it worked ...

Can i provide you further informations?

X3n0m0rph59, 2022-02-23:
@SHU-red

> My Profile immediately went active again

Great! So no customizations got lost.

> Is there something wrong with my custom setup then?

Nope, I don't think so. It's a bug in Eruption.

In the meantime I narrowed the bug down to be related to the new audio proxy code indeed. The audio backend performs async network i/o on the main input thread. This somehow slipped through testing on my system.

The bug can be reliably triggered by enabling "SoundFX Audio Effects". This will lead to huge latencies if the audio-proxy daemon is busy or disabled.

Does it affect the latencies if you run this command:
```sh
systemctl --user enable --now eruption-audio-proxy.service
```
> Please don't put sudo in front of that command

I am preparing the next round of fixes for this issue...

SHU-red, 2022-02-23:
OK,
i unmasked everything --> Latency back again
then i typed in
`systemctl --user enable --now eruption-audio-proxy.service`
--> Latency is much much lower but (i think) it still feels slower than before ... but not sure!

So i think you found the root cause

Thx!

SHU-red, 2022-02-23:
I did a update because i saw your commit.
Rebootet my PC

I think everything seems fine now

Thank you very very much for your effort!

X3n0m0rph59, 2022-02-23:
@SHU-red thank you for your feedback and for taking the time to help me getting this bug fixed. It is much appreciated!

I will close this issue now, please feel free to open up issues anytime if you encounter further problems.

closed 2022-02-23 by X3n0m0rph59