eruption-project/eruption #113: Unable to install after successful make

DirkSchueler, 2022-02-28:
Using Ubuntu 21.10 I was able to run "make", but "sudo make install" led to
`Shutting down daemons from previous installations...
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
make: [Makefile:32: install] Error 1 (ignored)
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
make: [Makefile:33: install] Error 1 (ignored)
Created symlink /etc/systemd/system/eruption.service → /dev/null.
Unit /etc/systemd/system/eruption.service is masked, ignoring.
Commencing installation of Eruption...
cp: cannot stat 'target/release/eruption{,ctl,-cmd,-hwutil,-netfx,-debug-tool,-hotplug-helper,-gui,-audio-proxy,-process-monitor}': No such file or directory
make: *** [Makefile:118: install] Error 1`

X3n0m0rph59, 2022-02-28:
Hi @DirkSchueler, looks like the directory `target/release/` does not contain the compiled binaries. The other error messages can be safely ignored. Could you please post the output of:

```sh
ls -alh target/release/
```

X3n0m0rph59, 2022-02-28:
> Unit /etc/systemd/system/eruption.service is masked

To unmask `eruption.service` you may want to run this command:

```sh
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

DirkSchueler, 2022-02-28:
> Hi @DirkSchueler, looks like the directory `target/release/` does not contain the compiled binaries. The other error messages can be safely ignored. Could you please post the output of:
> 
> ```shell
> ls -alh target/release/
> ```

`total 42M
drwxrwxr-x    7 dirk dirk 4,0K Feb 28 12:23 .
drwxrwxr-x    4 dirk dirk 4,0K Mär 23  2021 ..
drwxrwxr-x 1750 dirk dirk 100K Feb 27 14:08 build
-rw-rw-r--    1 dirk dirk    0 Mär 23  2021 .cargo-lock
drwxrwxr-x    2 dirk dirk 772K Feb 28 12:23 deps
-rwxrwxr-x    2 dirk dirk 6,3M Feb 25 14:32 eruption
-rwxrwxr-x    2 dirk dirk 2,6M Feb 25 14:30 eruption-audio-proxy
-rw-rw-r--    1 dirk dirk  622 Feb 25 14:33 eruption-audio-proxy.d
-rwxrwxr-x    2 dirk dirk 2,3M Feb 25 14:30 eruption-cmd
-rw-rw-r--    1 dirk dirk  341 Feb  9 11:27 eruption-cmd.d
-rwxrwxr-x    2 dirk dirk 3,6M Feb 25 14:31 eruptionctl
-rw-rw-r--    1 dirk dirk  546 Nov 23 13:49 eruptionctl.d
-rw-rw-r--    1 dirk dirk 3,2K Feb 28 12:14 eruption.d
-rwxrwxr-x    2 dirk dirk 3,0M Feb 25 14:30 eruption-debug-tool
-rw-rw-r--    1 dirk dirk 2,0K Feb  9 11:27 eruption-debug-tool.d
-rwxrwxr-x    2 dirk dirk 5,4M Feb 28 12:23 eruption-gui
-rw-rw-r--    1 dirk dirk 3,3K Feb  9 11:27 eruption-gui.d
-rwxrwxr-x    2 dirk dirk 2,5M Feb 25 14:30 eruption-hotplug-helper
-rw-rw-r--    1 dirk dirk  429 Nov 23 13:49 eruption-hotplug-helper.d
-rwxrwxr-x    2 dirk dirk 3,3M Feb 25 14:30 eruption-hwutil
-rw-rw-r--    1 dirk dirk 1,9K Feb  9 11:27 eruption-hwutil.d
-rwxrwxr-x    2 dirk dirk 4,1M Feb 25 14:31 eruption-netfx
-rw-rw-r--    1 dirk dirk  850 Nov 23 13:49 eruption-netfx.d
-rwxrwxr-x    2 dirk dirk 3,9M Feb 25 14:31 eruption-process-monitor
-rw-rw-r--    1 dirk dirk 1,1K Nov 23 13:49 eruption-process-monitor.d
-rwxrwxr-x    2 dirk dirk 2,6M Feb 25 14:30 eruption-util
-rw-rw-r--    1 dirk dirk  849 Feb  9 11:27 eruption-util.d
drwxrwxr-x    2 dirk dirk 4,0K Mär 23  2021 examples
drwxrwxr-x 6405 dirk dirk 316K Feb 27 14:08 .fingerprint
drwxrwxr-x    2 dirk dirk 4,0K Mär 23  2021 incremental
-rw-rw-r--    1 dirk dirk  857 Feb 25 14:33 liberuption.d
-rw-rw-r--    1 dirk dirk  742 Feb 25 14:33 liberuption_rs.d
-rw-rw-r--    2 dirk dirk 282K Feb 25 14:29 liberuption_rs.rlib
-rwxrwxr-x    2 dirk dirk 251K Feb 25 14:30 liberuption.so
-rw-rw-r--    1 dirk dirk  213 Sep 13 19:04 libprocmon_sys.d
-rw-rw-r--    2 dirk dirk 9,6K Feb 25 14:28 libprocmon_sys.rlib
-rwxrwxr-x    2 dirk dirk 495K Feb 25 14:29 simple
-rw-rw-r--    1 dirk dirk  791 Feb 25 14:33 simple.d`

DirkSchueler, 2022-02-28:
> > Unit /etc/systemd/system/eruption.service is masked
> 
> To unmask `eruption.service` you may want to run this command:
> 
> ```shell
> sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
> ```

Done
`sudo systemctl status eruption.service`
shows
`× eruption.service - Linux user-mode input and LED driver for keyboards, mice and other devices
     Loaded: loaded (/lib/systemd/system/eruption.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-02-28 13:42:48 CET; 29s ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
    Process: 86101 ExecStart=/usr/bin/eruption -c /etc/eruption/eruption.conf (code=exited, status=5)
    Process: 86111 ExecStopPost=/usr/bin/rm -f /run/eruption/eruption.pid (code=exited, status=0/SUCCESS)
   Main PID: 86101 (code=exited, status=5)
        CPU: 321ms

Feb 28 13:42:47 inspiron7506 eruption[86101]: inspiron7506 eruption[86101]: To enable this device driver, please change the 'driver_maturity_level' sett>
Feb 28 13:42:47 inspiron7506 eruption[86101]: inspiron7506 eruption[86101]: Not binding the device driver because it would require a lesser code maturit>
Feb 28 13:42:47 inspiron7506 eruption[86101]: inspiron7506 eruption[86101]: To enable this device driver, please change the 'driver_maturity_level' sett>
Feb 28 13:42:47 inspiron7506 eruption[86101]: inspiron7506 eruption[86101]: Not binding the device driver because it would require a lesser code maturit>
lines 1-16...skipping...`

X3n0m0rph59, 2022-02-28:
@DirkSchueler 

Eruption now filters out unstable drivers in the default configuration. To enable testing or even experimental drivers, please change the respective line in the `[global]` section of `/etc/eruption/eruption.conf`.

The default is: `driver_maturity_level = "stable"`. Valid values are `driver_maturity_level = "testing"` or even `driver_maturity_level = "experimental"`

```toml
# Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices
# Main configuration file

[global]
enable_experimental_features = false
driver_maturity_level = "testing"

profile_dirs = ["/var/lib/eruption/profiles/"]
script_dirs = ["/usr/share/eruption/scripts/"]

# select your keyboard variant
# keyboard_variant = "ANSI"
keyboard_variant = "ISO"

# Mouse handling
enable_mouse = true
grab_mouse = true

# "Away from keyboard" handling
afk_profile = "/var/lib/eruption/profiles/rainbow-wave.profile"
afk_timeout_secs = 0

# [[devices]]
# entry_type = "device"
# device_class = "serial"
# device_name = "Adalight Custom Serial LEDs"
# device_file = "/dev/ttyACM0"

# [[devices]]
# entry_type = "blacklist"
# vendor_id = 0x1e7d
# product_id = 0x311a
```

DirkSchueler, 2022-02-28:
Setting driver_maturity_level = "experimental" , git pull, make, sudo make install (no errors anymore), make start brought me Roccat Volcano Pro back to light. Thank you so far!

But: I cannot start the eruption gui "Could not start Eruption GUI, is the daemon running?" Reason; The name org.eruption was not provided by any .service files.

closed 2022-02-28 by DirkSchueler