eruption-project/eruption #116: lua or script error - /script.rs:718 - service failed

Lila-Kuh, 2022-03-04:
**Introduction**
Runing a long time fine..then i changed:
After uninstall jdk8/jre8 (jdk11,17 & jre11,17 were already installed);
recompiling `eruption-git` from AUR (eruption-git 0.1.22.r299.gfca00a5-1 & eruption-git 0.1.22.r320.g1d97024-1); 
crashed & failed eruption.service

**Please describe the bug**
compile and install works
after that eruption service does not start and 
<pre><code>The application panicked (crashed).
Message:  index out of bounds: the len is 0 but the index is 0
Location: eruption/src/scripting/script.rs:718

Please consider reporting a bug at https://github.com/X3n0m0rph59/eruption

  ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ BACKTRACE ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━
   1: lua_callk<unknown>
      at <unknown source file>:<unknown line>
   2: lua_pcallk<unknown>
      at <unknown source file>:<unknown line>
   3: start_thread<unknown>
      at <unknown source file>:<unknown line>
   4: __clone<unknown>
      at <unknown source file>:<unknown line></code></pre>

**To Reproduce**
Steps to reproduce the behavior:
- see Logs ?
- or of course i am the source of error -
human and systems ;)

**Expected behavior**
- the services are working again xD

**Logs**
Please provide logs: 

<details>
<summary><b>sudo journalctl -u eruption.service -b</b></summary>
<pre><code>
Mär 04 12:40:10 MSIMjO1902 systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Mär 04 12:40:10 MSIMjO1902 systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Mär 04 12:40:11 MSIMjO1902 eruption[767]: The application panicked (crashed).
Mär 04 12:40:11 MSIMjO1902 eruption[767]: Message:  index out of bounds: the len is 0 but the index is 0
Mär 04 12:40:11 MSIMjO1902 eruption[767]: Location: eruption/src/scripting/script.rs:718
Mär 04 12:40:11 MSIMjO1902 eruption[767]: Please consider reporting a bug at https://github.com/X3n0m0rph59/eruption
Mär 04 12:40:11 MSIMjO1902 systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Mär 04 12:40:11 MSIMjO1902 systemd[1]: eruption.service: Failed with result 'signal'.
Mär 04 12:51:49 MSIMjO1902 systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Mär 04 12:51:49 MSIMjO1902 systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
Mär 04 12:51:49 MSIMjO1902 eruption[13712]: The application panicked (crashed).
Mär 04 12:51:49 MSIMjO1902 eruption[13712]: Message:  index out of bounds: the len is 0 but the index is 0
Mär 04 12:51:49 MSIMjO1902 eruption[13712]: Location: eruption/src/scripting/script.rs:718
Mär 04 12:51:49 MSIMjO1902 eruption[13712]: Please consider reporting a bug at https://github.com/X3n0m0rph59/eruption
Mär 04 12:51:50 MSIMjO1902 systemd[1]: eruption.service: Main process exited, code=killed, status=6/ABRT
Mär 04 12:51:50 MSIMjO1902 systemd[1]: eruption.service: Failed with result 'signal'.
</code></pre>
</details>

<details>
<summary><b>systemctl --user status eruption-audio-proxy.service eruption-process-monitor.service</b></summary>
<pre><code>
● eruption-audio-proxy.service - Audio proxy daemon for the Eruption Linux user-mode driver
     Loaded: loaded (/usr/lib/systemd/user/eruption-audio-proxy.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-03-04 12:40:18 CET; 5min ago
       Docs: man:eruption-audio-proxy(1)
             man:audio-proxy.conf(5)
             man:eruptionctl(1)
   Main PID: 1205 (eruption-audio-)
      Tasks: 2 (limit: 19117)
[... 788 lines cut ...]
Failed to connect to bus: No medium found
Failed to connect to bus: No medium found
Failed to connect to bus: No medium found
:: Post-transaction hooks are fired... 
</code></pre>
</details>

############

**Please complete the following information:**

 * Version 0.1.23 (0.1.22.r299.gfca00a5) (release build)
 * Version 0.1.23 (0.1.22.r320.g1d97024) (release build)
 * Bus 003 Device 005: ID 1e7d:2de1 ROCCAT ROCCAT Burst Pro
 * Found supported device: 0x1e7d:0x2de1 - ROCCAT ROCCAT Burst Pro
 * Found supported mouse device: 0x1e7d:0x2de1 (3-5:1.0) - ROCCAT ROCCAT Burst Pro

**Additional context**
- System
<pre><code>Kernel: 5.16.11-2-MANJARO x86_64 bits: 64 compiler: gcc v: 11.2.0
Desktop: Xfce 4.16.0 tk: Gtk 3.24.29 info: xfce4-panel, plank
    wm: xfwm 4.16.1 vt: 7 dm: LightDM 1.30.0 Distro: Manjaro Linux
    base: Arch Linux</code></pre>

- Please excuse my english and copy/past denglish mix ;)

X3n0m0rph59, 2022-03-04:
Hi @Lila-Kuh, thanks for reporting this issue!

I have pushed out a fix for this bug to the `develop` branch. If you want to build from source please have a look at [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md). Before running `make` please switch to the `develop` branch.

I will backport this fix and will include it in the next builds of the `eruption-git` packages too.

Lila-Kuh, 2022-03-04:
yes this is successful.

- Switched branch to `develop`;
- make; sudo make install; make start

and now
<details>
<summary><b>systemctl status eruption*</b></summary>
<pre><code>systemctl status eruption*
Unit eruption-audio-proxy.service could not be found.
Unit eruption-cmd.service could not be found.
Unit eruptionctl.service could not be found.
Unit eruption-debug-tool.service could not be found.
Unit eruption-gui.service could not be found.
Unit eruption-hwutil.service could not be found.
Unit eruption-netfx.service could not be found.
Unit eruption-process-monitor.service could not be found.
Unit eruption-util.service could not be found.
● eruption.service - Linux user-mode input and LED driver for keyboards, mice and other devices
     Loaded: loaded (/usr/lib/systemd/system/eruption.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-03-04 16:35:37 CET; 6min ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
   Main PID: 249256 (eruption)
      Tasks: 15 (limit: 19117)
     Memory: 4.6M
        CPU: 17.102s
     CGroup: /system.slice/eruption.service
             └─249256 /usr/bin/eruption -c /etc/eruption/eruption.conf

Mär 04 16:35:37 MSIMjO1902 systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
Mär 04 16:35:37 MSIMjO1902 systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.

○ eruption-hotplug-helper.service - Eruption device hotplug notification daemon
     Loaded: loaded (/usr/lib/systemd/system/eruption-hotplug-helper.service; disabled; vendor preset: disabled)
     Active: inactive (dead) since Fri 2022-03-04 12:40:10 CET; 4h 1min ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
   Main PID: 982 (code=exited, status=0/SUCCESS)
        CPU: 9ms

Mär 04 12:40:10 MSIMjO1902 systemd[1]: Starting Eruption device hotplug notification daemon...
Mär 04 12:40:10 MSIMjO1902 systemd[1]: Started Eruption device hotplug notification daemon.
Mär 04 12:40:10 MSIMjO1902 systemd[1]: eruption-hotplug-helper.service: Deactivated successfully.</code></pre>
</details>

Thank you for your quick response and good work

X3n0m0rph59, 2022-03-05:
@Lila-Kuh I have update all `eruption-git` packages to build against the latest commit.

If you want to migrate away from the very unstable `develop` branch you can build from source against the more stable `master` branch:

```sh
git switch master
make
make stop
sudo make install
make start
```

Or try to install the latest `eruption-git` package from the AUR. You have to uninstall Eruption first!

> WARNING: This will delete all state and custom profiles in `/var/lib/eruption/profiles`

```sh
make stop
sudo make uninstall

paru -Sy eruption-git
```

Lila-Kuh, 2022-03-06:
Thank you very much for your quick reaction.

Everything has worked so far.

- Uninstall `develop` and all services.
- Then build the package from the AUR (`eruption-git`), install it and it works as expected.

Have a great weekend

closed 2022-03-06 by Lila-Kuh