eruption-project/eruption #117: Issues with Taskbar caused by eruption on elementaryOS

Vite, 2022-03-08:
**Introduction**
Hello. Thanks for the work on eruption. After uninstalling eruption and restarting my PC, the issue was solved. After installing it again, the issue was present again.

**Please describe the bug**
The little "dot" below a programs icon is not shown after installation of eruption. If I click the on the applications icon in the Taskbar it will open the program another time instead of focusing the already opened one.
A picture of the dot I am talking about:
[image]
This dot indicates that the application is opened.

**To Reproduce**
I installed it with the ppa as it is written down for Ubuntu or Pop!_OS installation.
latest elementaryOS 6

**Logs**
Will provide logs soon, issue will then be edited.

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * Eruption version [e.g. 0.1.20]
0.1.22-ubuntu4
 * Your device(s) (`sudo lsusb`)
ROCCAT Kone Pure Ultra
Logitech, Inc. Unifying Receiver
Anua Mic CM 900

X3n0m0rph59, 2022-03-08:
Hi @Vite, thank you for reporting this issue!

This is a bug with the `eruption-process-monitor`. I am currently investigating this issue and will try to implement a fix for it soon.

A workaround is to disable the process monitor for now:

```sh
systemctl --user disable --now eruption-process-monitor.service
```

After executing the aforementioned command in a terminal you will have to re-login your X session to fix the issue.
When disabling the `eruption-process-monitor` automatic switching of profiles will not be available.

X3n0m0rph59, 2022-03-08:
This bug seems to be fixed in the latest code of the `eruption-process-monitor`. You could therefore (temporarily) switch to the `eruption-git` package. Please refrain from switching to it, if you require a stable system since the `-git` snapshot packages are inherently unstable.

X3n0m0rph59, 2022-10-03:
@Vite Has this issue been resolved for you in the latest release?

X3n0m0rph59, 2022-10-12:
@Vite I have just checked with a custom build of Eruption in an elementaryOS virtual machine and the issue seems to have been solved. Closing this now... Please reopen the issue if the problem persists!

closed 2022-10-12 by X3n0m0rph59