eruption-project/eruption #118: Eruption no longer recognises my Roccat Kone Aimo mouse

Marethyu61, 2022-03-22:
**Introduction**
Hello,

Today I noticed that the application is not working properly.

**Please describe the bug**

I installed the application a few months ago for my Roccat Kone Aimo and everything worked fine. I could customize the leds and disable the service easily.

Now I can't change the LEDs and no action I take in the software has any effect. I have unplugged and replugged the mouse, uninstalled and reinstalled the application, but even after a reboot, nothing happens.

**To Reproduce**

Use a Roccat Kone Aimo mouse, download the gnome package and application, and restart the computer.

**Expected behavior**

When I use the application, the colours of the mouse leds change.

**Logs**

-- Logs begin at Wed 2021-11-17 17:19:21 EET, end at Tue 2022-03-22 10:46:49 EE>
mars 22 10:14:58 fixe-louis systemd[1]: Starting Linux user-mode input and LED >
mars 22 10:14:58 fixe-louis systemd[1]: Started Linux user-mode input and LED d>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Brightne>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Not bind>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: To enabl>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Not bind>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: To enabl>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: No suppo>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Lua erro>
                                            stack traceback:
                                                    [C]: in function 'animal_cr>
                                                    [string "eruption/src/scrip>
                                            caused by: runtime error: Could not>
                                                    UnknownError
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Script e>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Entering>
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Skipping>
mars 22 10:15:10 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Invalid >
mars 22 10:30:09 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Lua erro>
                                            stack traceback:
                                                    [C]: in function 'animal_cr>
                                                    [string "eruption/src/scrip>
lines 1-23...skipping...
-- Logs begin at Wed 2021-11-17 17:19:21 EET, end at Tue 2022-03-22 10:46:49 EET. --
mars 22 10:14:58 fixe-louis systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
mars 22 10:14:58 fixe-louis systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Brightness configuration value is set very low, the LEDs will probably stay dark!
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Not binding the device driver because it would require a lesser code maturity level
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: To enable this device driver, please change the 'driver_maturity_level' setting in eruption.conf respectively
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Not binding the device driver because it would require a lesser code maturity level
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: To enable this device driver, please change the 'driver_maturity_level' setting in eruption.conf respectively
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: No supported devices found!
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Lua error in file /usr/share/eruption/scripts/animal.lua: callback error
                                            stack traceback:
                                                    [C]: in function 'animal_create'
                                                    [string "eruption/src/scripting/script.rs:852:21"]:26: in main chunk
                                            caused by: runtime error: Could not instantiate the animal object: Lost connection to device
                                                    UnknownError
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Script execution failed
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Entering failsafe mode now, due to previous irrecoverable errors
mars 22 10:14:58 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Skipping unload of a failed tx
mars 22 10:15:10 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Invalid profile: failsafe.profile, refusing to switch profiles
mars 22 10:30:09 fixe-louis eruption[1175]: fixe-louis eruption[1175]: Lua error in file /usr/share/eruption/scripts/animal.lua: callback error
                                            stack traceback:
                                                    [C]: in function 'animal_create'
                                                    [string "eruption/src/scripting/script.rs:852:21"]:26: in main chunk
                                            caused by: runtime error: Could not instantiate the animal object: Lost connection to device
                                                    UnknownError
[... 35 lines cut ...]

**Please complete the following information:**

 * Eruption version : Profile switcher 15 (I downloaded the shell version few months ago )
 
* Your device(s)  :

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1e7d:2e27 ROCCAT ROCCAT Kone Aimo
Bus 001 Device 004: ID 04f3:0103 Elan Microelectronics Corp. ActiveJet K-2024 Multimedia Keyboard
Bus 001 Device 002: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 005: ID 3285:0607 Nacon GC100
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**Additional context**

The problem occurred today.

English is not my first language. Sorry if there are mistakes.

Thank you in advance !

Marethyu

Marethyu61, 2022-03-25:
Hello, thank you for responding so quickly,

I just changed the line. Strangely, the Eruption module doesn't work at all. Instead, a cross is displayed. Here is a picture (https://zupimages.net/viewer.php?id=22/12/5vau.jpg). I can't interact with the extension anymore. 

I think I made a mistake in the registry. I'm sending you a copy of the .conf file (https://zupimages.net/viewer.php?id=22/12/6pxj.jpg). Should I uninstall eruption (and if so, how to do it properly?) and its extension or is there just some adjustments to be made please? 

Thank you very much 

Marethyu

X3n0m0rph59, 2022-03-25:
> just changed the line. Strangely, the Eruption module doesn't work at all. Instead, a cross is displayed. Here is a picture (https://zupimages.net/viewer.php?id=22/12/5vau.jpg). I can't interact with the extension anymore.

If the Eruption daemon is not running, the GNOME shell extension will just show an error icon.

There is no need to reinstall Eruption just due to an invalid configuration. You have to replace your current configuration in `/etc/eruption/eruption.conf` with the following lines:

```toml
# Eruption - Linux user-mode input and LED driver for keyboards, mice and other devices
# Main configuration file

[global]
enable_experimental_features = false
driver_maturity_level = "experimental"

profile_dirs = ["/var/lib/eruption/profiles/"]
script_dirs = ["/usr/share/eruption/scripts/"]

# select your keyboard variant
# keyboard_variant = "ANSI"
keyboard_variant = "ISO"

# Mouse handling
enable_mouse = true
grab_mouse = true

# "Away from keyboard" handling
afk_profile = "/var/lib/eruption/profiles/rainbow-wave.profile"
afk_timeout_secs = 0

# [[devices]]
# entry_type = "device"
# device_class = "serial"
# device_name = "Adalight Custom Serial LEDs"
# device_file = "/dev/ttyACM0"

# [[devices]]
# entry_type = "blacklist"
# vendor_id = 0x1e7d
# product_id = 0x311a
```

Marethyu61, 2022-03-25:
I have just implemented the change, and it works again! Thank you very much for your responsiveness!

The bug can be considered solved.

Marethyu

X3n0m0rph59, 2022-03-25:
Excellent! I will close this issue now.

closed 2022-03-25 by X3n0m0rph59