eruption-project/eruption #119: Could not start Eruption GUI

alkaris2, 2022-03-25:
**Introduction**
Could not open Eruoption GUI because it says the service org.eruption was not provided by any .service files

**Please describe the bug**

[image]

**Logs**
[eruption_service.txt](https://github.com/X3n0m0rph59/eruption/files/8347515/eruption_service.txt)

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

5.15.28-1-MANJARO

Bus 001 Device 009: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
Bus 001 Device 010: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K

X3n0m0rph59, 2022-03-25:
Hi @alkaris2!

The driver for the device `ROCCAT Kone Aimo 16K` is still flagged as experimental.

> Mar 25 14:59:49 tohru eruption[2014831]: tohru eruption[2014831]: Not binding the device driver because it would require a lesser code maturity level
Mar 25 14:59:49 tohru eruption[2014831]: tohru eruption[2014831]: To enable this device driver, please change the 'driver_maturity_level' setting in eruption.conf respectively

To fix the problem please run:

```sh
sudoedit /etc/eruption/eruption.conf
```

And change the line (In the section `[global]`)

```toml
driver_maturity_level = "testing"
```

to this:

```toml
driver_maturity_level = "experimental"
```

Then please restart the eruption daemon.

```sh
sudo systemctl restart eruption.service
```

The change is required since I introduced the concept of driver code maturity levels. The driver for the `ROCCAT Kone Aimo 16K` is flagged as `experimental`, so eruption refuses to bind the driver for this device in the default configuration.

X3n0m0rph59, 2022-03-25:
Duplicate of #118

X3n0m0rph59, 2022-03-25:
@alkaris2 

Additionally, Eruption fails to open the device file of your keyboard. This could happen when eruption is not running as the `root` user, or when the device is opened exclusively by other software.

alkaris2, 2022-03-25:
More issues arise after trying to change things in the GUI. When updating something it gives me this;

```
tohru eruption[2085169]: Error during notification of observers [keyboard_event]: timed out waiting on send
```

repeats this message over and over with timeout errors when trying to update device. And also this

[eruption_fatalerror.txt](https://github.com/X3n0m0rph59/eruption/files/8349529/eruption_fatalerror.txt)

X3n0m0rph59, 2022-03-25:
Could you please post the full logs of Eruption using this command:

```sh
sudo journalctl -u eruption.service -b
```

And please post the failing `.lua` and `.lua.manifest` files too

alkaris2, 2022-03-25:
I haven't even touched the LUA files and somehow after modifying settings from the GUI with the sliders just breaks when trying to update to the keyboard device.

Here's the journalctrl output.
[eruption_journalctlout.txt](https://github.com/X3n0m0rph59/eruption/files/8349635/eruption_journalctlout.txt)

X3n0m0rph59, 2022-03-25:
@alkaris2 

Thank you for posting the full logs. It seems that the failing Lua scripts is just a symptom of a fatal error that happened earlier in the startup process of Eruption.

You are running an outdated firmware revision on the `ROCCAT Vulcan AIMO` that is known to cause problems with Eruption. Could you please upgrade the firmware to the latest revision, using `ROCCAT Swarm`?

alkaris2, 2022-03-25:
How would I go about updating that through Linux?

X3n0m0rph59, 2022-03-25:
Firmware updates are only supported through the device OEM software `ROCCAT Swarm`, which only runs on Windows.
If you don't have access to a Windows PC you could set up a Virtual Machine and perform the update this way.

alkaris2, 2022-03-25:
It seems I was able to update by installing ROCCAT Swarm through Wine, but when I reloaded Eruption daemon services every light on the keyboard when out except for the Audio buttons and NumLock key light. I've tried cycling through different color profiles but its not doing anything.

X3n0m0rph59, 2022-03-25:
@alkaris2

Could you please post full logs of Eruption running in `debug` mode by running the following commands in a terminal:

```sh
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service 
sudo RUST_LOG=debug RUST_BACKTRACE=full eruption -c /etc/eruption/eruption.conf
```

To re-enable Eruption please run this command:

```sh
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service 
```

alkaris2, 2022-03-26:
Here's the debug output.
[eruption_debug.txt](https://github.com/X3n0m0rph59/eruption/files/8354566/eruption_debug.txt)

it always seems after restarting the system that `eruption.service` always fails to start, resulting said error `org.eruption was not provided by any .service files`

Another thing, when I manually restart eruption daemon service it doesn't detect my mouse in the GUI for `Bus 001 Device 002: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K`

Anytime I edit one of the profiles by changing colors, Eruption just completely gives up, turns off RGB lights and hangs on time out waiting on send operations. Seems pretty broken for "stable" software if you can't edit a profile configuration without it breaking completely and turning off RGB completely and getting hung up. If I try restart the service it stays in it's hung up state in an irrecoverable state;

```
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Entering failsafe mode now, due to previous irrecoverable errors
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Skipping unload of a failed tx
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Skipping unload of a failed tx
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Skipping unload of a failed tx
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Skipping unload of a failed tx
Mar 26 13:47:29 tohru eruption[373692]: tohru eruption[373692]: Skipping unload of a failed tx
```

I've reinstalled this software twice now and it always goes back to this whenever it doesn't like changes in the profile configs.

X3n0m0rph59, 2022-03-26:
@alkaris2 

>  WARN  eruption::hwdevices::roccat_vulcan_1xx > Outdated firmware version: 1.31, should be: >= 1.36

You are still running an unsupported firmware version on your device. If you want to use Eruption it is required to update the firmware of your keyboard to the latest revision, by using the OEM software on a Windows PC.

> Seems pretty broken for "stable" software if you can't edit a profile configuration without it breaking completely and turning off RGB completely and getting hung up. 

As stated in the [README.md](https://github.com/X3n0m0rph59/eruption#important-information), Eruption currently is **not** stable software. Only basic LED lighting support is working on a few selected devices. Please be aware that the GUI is a technology preview version.

> WARN  eruption::hwdevices                    > To enable this device driver, please change the 'driver_maturity_level' setting in eruption.conf respectively

To enable support for your mouse, please update `/etc/eruption/eruption.conf`:
```toml
driver_maturity_level = "experimental"
```

> Also I notice in the AUR package build instructions, that compiling of eruption-gui is commented out, why is that? And why doesn't it compile the GUI along with everything else?

The GUI is a technology preview, and is only available in `-git` snapshot packages.

alkaris2, 2022-03-26:
That makes no sense, I downloaded the ROCCAT Swarm and then it prompted me to download a firmware and it installed it. So I don't get why it says I'm still on outdated firmware version 1.31, from the ROCCAT Swarm it says 1.9402.

X3n0m0rph59, 2022-03-26:
> ROCCAT Swarm it says 1.9402

This is the revision of `ROCCAT Swarm` itself, not that of your keyboard's firmware. In `ROCCAT Swarm`, the firmware version can be checked and flashed in the settings sub-menu. You may want to go to `Settings` > `Update-Center` > `Recovery`.
Please don't use Wine for the update process.

alkaris2, 2022-03-26:
I tried with VirtualBox but nothing shows up there. Device doesn't show in VM and can't update via VirtualBox, and there is no Recovery option.

[image]

X3n0m0rph59, 2022-03-26:
@alkaris2 

I only use KVM/QEMU but Virtual Box _should_ work too. Please find a guide on how to enable USB devices in Virtual Box here:  [https://www.tecmint.com/enable-usb-in-virtualbox/](https://www.tecmint.com/enable-usb-in-virtualbox/)

alkaris2, 2022-03-26:
I eventually got the devices updated with QEMU KVM, but took a bit of fiddling around. I had to manually mount and dismount the devices from the VM just to get Swarm to update the devices properly.

The device reads firmware 1.36 now, but with Eruption GUI it reads as `<unknown>` for Firmware, not sure if that's normal, but when I checked with Debug mode it definitely reports 1.36. The application doesn't seem to simply close or hang-up now when I try to edit a profile. However when I do try to change colors of a profile, nothing changes, it just keeps the same color values, or defaults to another value despite using the color picker.

X3n0m0rph59, 2022-03-26:
> I eventually got the devices updated with QEMU KVM, but took a bit of fiddling around.
> ... The application doesn't seem to simply close or hang-up now when I try to edit a profile.

Excellent!

**Regarding setting colors or other parameters:**

Lua scripts define default parameter values in their associated `.lua.manifest` files. These default values can be selectively overridden by the author of a `.profile` file that references the script, therefore defining the default look and feel of said profile.

To customize, the user of the profile can either copy that profile and specify differing parameter overrides in the new file or preferably, one can use the configuration mechanism of Eruption.

An example:

Switch to a profile that shows a solid color:
```shell
eruptionctl switch profile solid.profile
```

List all available parameters
```shell
eruptionctl -v param
```

Switch from the default color to ARGB `ff00ff00` (bright green)
```shell
eruptionctl param "Solid Color" color_background "#ff00ff00"
```

Switch from the default color to ARGB `ffffff00` (yellow)
```shell
eruptionctl param "Solid Color" color_background "#ffffff00"
```

When a parameter value is overridden this way, Eruption will create a `.profile.state` file (here `solid.profile.state`) in `/var/lib/eruption/profiles/` holding the overridden parameter values. All values specified in the `.profile` file will be overridden by the values from the `.profile.state` file. To restore default settings of the profile, simply delete the `.profile.state` file.

The color picker in the `Eruption GUI` is not fully implemented yet. Simple selection works by double clicking on a predefined color though.

X3n0m0rph59, 2022-05-06:
I will close this issue now, please feel free to open up a new issues any time!

closed 2022-05-06 by X3n0m0rph59