eruption-project/eruption #120: Add Support for ROCCAT Kone Pro

MyChaOS87, 2022-04-13:
**Introduction**
Hi, just found out that the `Kone Pro` is missing.
I did some testing and found it very similar to the `Burst Pro`. burst. 
I have tried the code on a branch with the eruption-debugtool as I implemented it like the `Burst Pro` in `eruption-hwutil/src/hwdevices/`
Just saw that initialization use `j` as `0x80` and `0x90`  instead `0..1` in the `0x04` ctrl_report, so i adapted that unless it seems unused anyways in the `Burst Pro`

I'd be happy to also help on the final implementation, but it does not seem perfectly straight forward, i don't understand why the 0x04 is unused for example. Will give you a PR of what i have done so far

**Device to add support for**
Roccat Kone Pro
Bus 005 Device 004: ID 1e7d:2c88 ROCCAT ROCCAT Kone Pro

**Please attach your *.pcapng file, if possible**
For further information, please see the [Eruption Wiki](https://github.com/X3n0m0rph59/eruption/wiki).
[roccatkonepro-3.7.0.zip](https://github.com/X3n0m0rph59/eruption/files/8479774/roccatkonepro-3.7.0.zip)
Here the USB-address is different as this was on windows...

MyChaOS87, 2022-04-13:
```
$ sudo eruption-debug-tool -v run-tests 6                       
Index: 06: ID: 1e7d:2c88 ROCCAT/ROCCAT Kone Pro Subdev: 0
Bound driver: ROCCAT Kone Pro
Sending device init sequence...
Step 2
Sending control device feature report
Waiting for control device to respond...
Step 3
Sending control device feature report
Waiting for control device to respond...
Setting LEDs from supplied map...
Setting LEDs from supplied map...
Setting LEDs from supplied map...
```

Left Led is 0, right LED is 1

MyChaOS87, 2022-04-13:
added the experimental eruption support based on the Burst, I changed the `0x04` control to use the `0x80` and `0x90` from my pcap

torgefaehrlich, 2022-07-29:
Hi,
I have a kone pro and tried to make it work with eruption.
I just managed to checkout [86801](https://github.com/X3n0m0rph59/eruption/commit/868015733b0f390192fc60f6270bb4524db21808) and compile it. Had to fix the array lengths once again.
hwutil and debug-tool seem to run OK, but still no supported devices detected in the gui.
Any way I can help? What are the next steps?
Regards
Torge

MyChaOS87, 2022-07-29:
Yes you need to activate maturity level 

Which array length you had to fix? 

I am actually running this since I committed that on a daily basis and had not a single crash, perhaps not using the features you tried...

Could you tell how to reproduce the crashes? And how they surface? Then I can try to reproduce and hopefully find a way to fix some of it.

MyChaOS87, 2022-07-29:
@torgefaehrlich btw I believe the experimental code already reached master in the meantime, so no need to check out the old commit

torgefaehrlich, 2022-08-22:
@MyChaOS87 Yes, you're right. Should have gone for master. The array length I was referring to is this hard-coded number: https://github.com/X3n0m0rph59/eruption/commit/868015733b0f390192fc60f6270bb4524db21808#diff-99a195c833a05e62e6f727abb048813161d111d004034c945c6d9b5351a9f035L75 (26 instead of 27) which has to be fixed every time a new item is added to the list. There is one more place in the code somewhere. I just stumbled upon it due to the way I was accessing the experimental version (by following this comment: https://github.com/X3n0m0rph59/eruption/issues/120#ref-commit-8680157)