eruption-project/eruption #122: Add Support for Kone XP

muhgo, 2022-04-14:
**Introduction**
Hi,
I recently got my hands on a Kone XP and wanted to contribute and help supporting it. I just captured the required files and did not dig into the code yet. My rustlang knowledge is sadly quite limited as of now but I may be able to help as I work as an embedded Linux developer and I'm trying to learn rustlang in my free time anyways.
And as a sidenote: thanks so much for this project, I've been using it for quite a while now for my Vulcan 120 and my now broken Kone Remastered :-).

Cheers

**Device to add support for**
1e7d:2c8b ROCCAT ROCCAT Kone XP

[konexp_v1.09_swarm_v1.9403.pcapng.zip]

X3n0m0rph59, 2022-04-30:
Hi @muhgo, thank you very much for creating the capture file! Your support is much appreciated!

I just pushed out the first implementation to the `develop` branch.

JonLit, 2022-06-15:
Hi, I also have a Kone XP, but when I try to start eruption, it instantly crashes:
Log:
`INFO  eruption > Starting Eruption - Realtime RGB LED Driver for Linux: Version 0.1.24 (0.1.23.r77.g1a8569e) (release build)
WARN  eruption > ** EXPERIMENTAL DRIVERS are ENABLED, this may expose serious bugs! **
INFO  eruption > Loading saved state...
INFO  eruption > Registering plugins...
INFO  eruption::plugin_manager > Registering plugin: Keyboard - Keyboard related functions
INFO  eruption::plugin_manager > Registering plugin: Mouse - Mouse related functions
INFO  eruption::plugin_manager > Registering plugin: Macros - Inject programmable keyboard and mouse events
INFO  eruption::plugin_manager > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
INFO  eruption::plugin_manager > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
INFO  eruption::plugin_manager > Registering plugin: Profiles - Switch profiles based on system state
INFO  eruption::plugin_manager > Registering plugin: System - Basic system information and status
INFO  eruption::plugins::macros > Initializing Linux virtual input devices
INFO  eruption::plugin_manager  > Registering plugin: Sensors - Query system sensor values
INFO  eruption::plugin_manager  > Registering plugin: Linux ULEDs - Linux Userspace LEDs interface
INFO  eruption::plugins::uleds  > The ULEDs subsystem is not available on this kernel
INFO  eruption::plugin_manager  > Registering plugin: SDK Support - Support for the Eruption SDK
INFO  eruption::plugin_manager  > Registering plugin: Audio - Audio related functions
INFO  eruption::plugin_manager  > Registering plugin: Animal - Simulation of organic movements (support library)
INFO  eruption::plugins::persistence > Loading persistent state data from disk...
INFO  eruption                       > Plugins loaded and initialized successfully
INFO  eruption                       > Enumerating connected devices...
INFO  eruption::hwdevices            > Found supported mouse device: 0x1e7d:0x2c8b (1-11.2:1.0) - ROCCAT ROCCAT Kone XP
INFO  eruption::hwdevices::roccat_kone_xp > Bound driver: ROCCAT Kone XP
INFO  eruption::hwdevices                 > Found supported keyboard device: 0x1e7d:0x30f7 (1-11.1:1.0) - ROCCAT ROCCAT Vulcan Pro
INFO  eruption::hwdevices::roccat_vulcan_pro > Bound driver: ROCCAT Vulcan Pro
INFO  eruption                               > Opening keyboard device...
INFO  eruption                               > Initializing keyboard device...
INFO  eruption                               > Configuring keyboard LEDs...
INFO  eruption                               > Firmware revision: <unknown>
INFO  eruption                               > Spawning keyboard input thread...
INFO  eruption                               > Opening mouse device...
INFO  eruption                               > Initializing mouse device...
ERROR eruption::hwdevices::roccat_kone_xp    > Wait 1: Invalid result
ERROR eruption::hwdevices::roccat_kone_xp    > Wait 2: Invalid result
ERROR eruption::hwdevices::roccat_kone_xp    > Wait 3: Invalid result
INFO  eruption                               > Configuring mouse LEDs...
The application panicked (crashed).
Message:  index out of bounds: the len is 180 but the index is 180
Location: eruption/src/hwdevices/roccat_kone_xp.rs:1248`

I installed `eruption-git` from the AUR, `eruption --version` says this: `Eruption 0.1.24 (0.1.23.r77.g1a8569e) (release build)`

`lsusb`:
`Bus 002 Device 003: ID 0781:558a SanDisk Corp. Ultra
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0dba:3000 Digidesign Mbox 2
Bus 001 Device 003: ID 046d:c214 Logitech, Inc. ATK3 (Attack III Joystick)
Bus 001 Device 002: ID 11c9:55f0 Nacon GC-100XF
Bus 001 Device 008: ID 048d:5702 Integrated Technology Express, Inc. ITE Device
Bus 001 Device 011: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 001 Device 009: ID 046d:c24f Logitech, Inc. G29 Driving Force Racing Wheel [PS3]
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 013: ID 22b8:2eb7 Motorola PCS Android
Bus 001 Device 015: ID 06a3:0763 Saitek PLC Pro Flight Rudder Pedals
Bus 001 Device 014: ID 056a:0303 Wacom Co., Ltd CTH-680 [Intuos Pen & Touch (M)]
Bus 001 Device 012: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 016: ID 1e7d:2c8b ROCCAT ROCCAT Kone XP
Bus 001 Device 017: ID 1e7d:30f7 ROCCAT ROCCAT Vulcan Pro
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

JonLit, 2022-06-16:
Thanks, now eruption itself works, but the GUI (for me) crashes now with this Error:
`The application panicked (crashed).
Message:  called `Result::unwrap()` on an `Err` value: Error { domain: g-resource-error-quark, code: 0, message: "Die Ressource auf »/org/eruption/eruption-gui/img/roccat-kone-xp.png« existiert nicht" }
Location: eruption-gui/src/ui/mouse/hwdevices/roccat_kone_xp.rs:42` - Transaltion: The Resource at <path> doesn't exist

X3n0m0rph59, 2022-06-16:
Hi @JonLit, thanks for testing the driver!

The driver is currently in development and most of its functionality is still missing. I currently do not have access to a `ROCCAT Kone XP` so I will have to implement this blindfolded.

I have pushed a few fixes to the `develop` branch. Could you please test them and give feedback by posting logs?

Thanks in advance!

X3n0m0rph59, 2022-06-16:
@JonLit 

>eruptionctl devices profile 00 /var/lib/eruption/profiles/swirl-perlin-rainbow.profile error: Invalid value "/var/lib/eruption/profiles/swirl-perlin-rainbow.profile" for '<PROFILE>': invalid digit found in string

The command `eruptionctl devices profile` will query or switch the hardware profile of the selected device. This may change parameters like e.g. DPI settings, where implemented. It is not to be confused with an Eruption `.profile` file, which is an entirely unrelated/different concept!

JonLit, 2022-06-16:
> @JonLit
>
>
> The command `eruptionctl devices profile` will query or switch the hardware profile of the selected device. This may change parameters like e.g. DPI settings, where implemented. It is not [...]

Interesting, because even with the eruption ctl switch command I can't change the rgb, I'll send the logs in a few minutes. Currently it's stuck in the matrix rgb profile

JonLit, 2022-06-16:
The gui now works, but when I start eruption, my cursor can't be moved by my mouse anymore until I stop eruption.
Also, in the gui, the page for the mouse looks a bit "incomplete":
[image]

JonLit, 2022-06-16:
> 

I tried clicking the "Angle Snapping" toggle, but then eruption-gui crashes and I can't get cursor control back with my mouse until I replug it, killing eruption also doesn't work to get mouse control back

JonLit, 2022-06-16:
btw rgb brightness control works on the Kone XP

X3n0m0rph59, 2022-06-16:
Thanks again for testing this!

The only supported features of the `ROCCAT Kone XP` driver are LED lighting control at this time. The hardware configuration functionality is currently not implemented at all. So DPI switching, angle snapping, firmware status, DCU parameters ... will not work at the moment.

> btw rgb brightness control works on the Kone XP

Ah, great to hear that!  This means that the LED support code seems to work... 

I will try to fix the bug that causes the device to hang. Would it be possible for you to post a video of your mouse showing some effects rendered by the Eruption daemon then?

JonLit, 2022-06-16:
> 

well that's the other issue I wrote about, I can't switch profiles, it always snaps back to the matrix profile
But that works on both the Keyboard and Mouse, on my Vulcan Pro it has the green/black colors scrolling down, the mouse is just entirely green

but I'll post a picture of it in a second

JonLit, 2022-06-16:
Please note that the colors are very far off, in reality they're the same as the keyboard, a nice and full green[image][image]

X3n0m0rph59, 2022-06-16:
To debug this issue could you please run Eruption with debug logging enabled? Please mask the `eruption.service` before.

```shell
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service

sudo RUST_LOG=debug target/debug/eruption
```

To unmask the `eruption.service` later on, please run:

```shell
sudo systemctl unmask eruption.service && sudo systemctl restart eruption.service
```

Please post the complete debug log!

JonLit, 2022-06-16:
> To debug this issue could you please run Eruption with debug logging enabled? Please mask the `eruption.service` before.
>
> ```shell
> sudo systemctl mask eruption.service && sudo systemctl stop [...]

oh damn the last few times I recompiled it I reinstalled the whole program :joy: I'm a complete noob with rust

[https://pastebin.com/FBc0AaMB](https://pastebin.com/FBc0AaMB)

X3n0m0rph59, 2022-06-16:
Thanks for posting the full debug logs! Could you please post the debug logs of eruption when you run:

```shell
eruptionctl switch profile solid.profile
```

From a different terminal to switch to some other `.profile` file. 

Do the LEDs of the `Kone XP` still light up with the latest commit from the `debug` branch applied?

JonLit, 2022-06-16:
Here are a Video and the logs for the buttons on the mouse
https://streamable.com/9swnq1
https://pastebin.com/B56NQN01

JonLit, 2022-06-16:
> 

https://pastebin.com/pYYVHjFT

JonLit, 2022-06-16:
ah wait, I didn't rebuild eruption

X3n0m0rph59, 2022-06-16:
Great that really helps a lot, thank you for your excellent support!
I will try to fix the crasher bug related to the mouse buttons tomorrow.

JonLit, 2022-06-16:
> 

sure, thank YOU for this excellent software and being willing to make it work for other people!

JonLit, 2022-06-17:
Great! Now I can switch profiles, and they work on both the keyboard and the mouse (I have no Idea, but now it doesn't work anymore, and the mouse is stuck on red leds)

Also I think the DPI button LED is in the wrong order, it is in the middle of the mouse, but for color effects I think it's on the right of the mouse or on the left of the keyboard

magicalraccoon, 2023-12-28:
Is there anything needed from Kone XP users in order to get this supported?