eruption-project/eruption #126: Could not connect to the Eruption daemon

ahopkins, 2022-05-11:
I cannot seem to connect to the service even though it is running.

[image]

```
▶ systemctl status eruption.service                                                                                                           ☸ prod (core) 
● eruption.service - Linux user-mode input and LED driver for keyboards, mice and other devices
     Loaded: loaded (/usr/lib/systemd/system/eruption.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-05-11 15:15:26 IDT; 3h 38min ago
       Docs: man:eruption(8)
             man:eruption.conf(5)
             man:eruptionctl(1)
             man:eruption-netfx(1)
   Main PID: 653 (eruption)
      Tasks: 8 (limit: 77072)
     Memory: 9.3M
        CPU: 3min 37.376s
     CGroup: /system.slice/eruption.service
             └─653 /usr/bin/eruption -c /etc/eruption/eruption.conf

May 11 15:15:26 hopkins systemd[1]: Starting Linux user-mode input and LED driver for keyboards, mice and other devices...
May 11 15:15:26 hopkins systemd[1]: Started Linux user-mode input and LED driver for keyboards, mice and other devices.
May 11 15:15:26 hopkins eruption[653]: hopkins eruption[653]: Could not parse state file: Could not load global runtime state: configuration file "/var/lib/er>
May 11 15:15:27 hopkins eruption[653]: hopkins eruption[653]: Could not load persisted state: No such file or directory (os error 2)
May 11 15:15:27 hopkins eruption[653]: hopkins eruption[653]: Outdated firmware version: 1.18, should be: >= 1.36
May 11 15:15:54 hopkins eruption[653]: hopkins eruption[653]: Socket error: Audio proxy disconnected
```

Perhaps it is because of the missing file? Not sure what that is or what it should be.

X3n0m0rph59, 2022-05-11:
Hi @ahopkins, thanks for reporting this issue!

I suspect that Eruption crashes due to a firmware version mismatch and therefore does not react to requests over D-Bus anymore:

```
Outdated firmware version: 1.18, should be: >= 1.36
```

So the firmware of the device needs to be updated to at least version `1.36` for Eruption to function properly. This is a known problem. The recommended fix would be to update your device using the OEM software `ROCCAT Swarm` on Windows.

If the problem persists, could you please post the full debug logs of Eruption daemon startup?

Please disable eruption first:
```sh
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
```

Execute eruption with debug logging enabled:
```sh
sudo RUST_LOG=debug RUST_BACKTRACE=full eruption
```

ahopkins, 2022-05-11:
Thanks for the heads up. I'll go try and hunt down/dust off a windows laptop and let you know. Thanks so much for the quick response and helping me out.

ahopkins, 2022-05-11:
That totally did it! :clap: 

I was able to update the firmware on my Windows laptop, and then when I moved it back to my Arch computer it worked like a charm. Awesome awesome. Thanks!!

closed 2022-05-11 by ahopkins