eruption-project/eruption #140: No character key press detected after pausing music with Fn + F11

SHU-red, 2022-06-15:
**Introduction**
Sometimes i have the weird behavior, that i pause the music, while beeing in the browser at one of my other virtual desktops (using window-tiling manager AwesomeWM), afterwards no character-keystroke is recognized anymore

**Please describe the bug**
- Browsing in the web while listening to music
- Pausing the Music (Fn+F11)
- Unpausing the Music (Fn+F11)
- Trying to type something into the address-bar of the browser
--> No keystroke is recognized for normal typing
--> I still can change Windwos with Mod+1 or Mod+2 or Mot+3 ...
--> If i then re-press Fn+F11 to pause the music, the keystrokes start fully working again

**To Reproduce**
- As described above but it only appears sporadically

**Expected behavior**
- Normaly no matter what, a character keystroke shall aways be detected for writing text

**Logs**
```
...............
Jun 15 21:22:33 sebastian-manjaro eruption[673]: sebastian-manjaro eruption[673]: Error during notification of observers [keyboard_event]: timed out sending on a full channel
Jun 15 21:22:33 sebastian-manjaro eruption[673]: sebastian-manjaro eruption[673]: Error during notification of observers [keyboard_event]: timed out sending on a full channel
Jun 15 21:22:33 sebastian-manjaro eruption[673]: sebastian-manjaro eruption[673]: Error during notification of observers [keyboard_event]: timed out sending on a full channel
..................

```
**Screenshots (optional)**
No

**Please complete the following information:**

 * Eruption version Eruption 0.1.24 (0.1.23.r77.g1a8569e) (release build)
 * Your device(s) Bus 001 Device 003: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO

**Additional context**
No

SHU-red, 2022-06-16:
Just if all of this leads to the same problem
Even if i do not recongize any error i get sometimes this one:;

```
Jun 16 13:52:16 sebastian-manjaro eruption[676]: sebastian-manjaro eruption[676]: More than 82 milliseconds of jitter detected!
Jun 16 13:52:16 sebastian-manjaro eruption[676]: sebastian-manjaro eruption[676]: This means that we dropped at least one frame
Jun 16 13:52:16 sebastian-manjaro eruption[676]: sebastian-manjaro eruption[676]: Loop took: 221 milliseconds, goal: 41
```

The `timed out on sending on a full channel` appears at each and every keystroke as far as i can determine

SHU-red, 2022-06-16:
One more additional information if the "This means that we dropped at least one frame" message appears, it also comes to multiplication of keystrokes

What means that if i type hello, this can be the result: "hellllllllllllllllllllllllo"

X3n0m0rph59, 2022-10-01:
@SHU-red I strongly believe that the fix for issue #160 fixes the bug that you are encountering as well. 

Could you please build the `develop` branch, and give it a try?

X3n0m0rph59, 2022-10-12:
Hi @SHU-red, any news on this one?

I believe that this issue may have gotten solved too! If you are running the latest `eruption-git` package it should  be fixed.

SHU-red, 2022-10-12:
I think you can close this topic.
Didnt see it for a long time.

Thanks

X3n0m0rph59, 2022-10-12:
OK, great! I will close this issue now!

closed 2022-10-12 by X3n0m0rph59