eruption-project/eruption #145: Unable to build project from source

DirkSchueler, 2022-06-24:
**Introduction**
Trying to follow the instructions from your install.md.
I'm running an Ubuntu 22.04 machine so it's a debian distro.
cargo 1.63.0-nightly (03a849043 2022-06-19) is in place.

**Please describe the bug**
Installing the dependencies using 
```
sudo apt install libusb-1.0-0-dev libhidapi-dev libevdev-dev libudev-dev libdbus-1-dev \
  libpulse-dev lua liblua-5.4-dev libx11-dev libxrandr-dev libgtk-3-dev libgdk-pixbuf2.0-dev \
  libatk1.0-dev libpango1.0-dev libcairo2-dev libgtksourceview-4.0-dev

```
leads to
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package lua is a virtual package provided by:
  lua5.4:i386 5.4.4-1
  lua5.3:i386 5.3.6-1build1
  lua5.2:i386 5.2.4-2
  lua5.1:i386 5.1.5-8.1build4
  lua5.4 5.4.4-1
  lua5.3 5.3.6-1build1
  lua5.2 5.2.4-2
  lua5.1 5.1.5-8.1build4
You should explicitly select one to install.

E: Package 'lua' has no installation candidate
E: Unable to locate package liblua-5.4-dev
E: Couldn't find any package by glob 'liblua-5.4-dev'
E: Unable to locate package libgtksourceview-4.0-dev
E: Couldn't find any package by glob 'libgtksourceview-4.0-dev'

Maybe this is the source of the problem that arises whenever I try to use 
`make `

that fails with 
```
.
.
.
  Compiling libsystemd-sys v0.9.3
error: failed to run custom build command for `libsystemd-sys v0.9.3`

Caused by:
  process didn't exit successfully: `/home/dirk/data/eruption/target/release/build/libsystemd-sys-fda6896cbbba0f62/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=SYSTEMD_LIBS_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=SYSTEMD_LIBS_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_SYSTEMD_LIBS
  cargo:rerun-if-env-changed=SYSTEMD_LIBS
  cargo:rerun-if-env-changed=SYSTEMD_LIB_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=SYSTEMD_LIB_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_SYSTEMD_LIB_DIR
  cargo:rerun-if-env-changed=SYSTEMD_LIB_DIR
  cargo:rerun-if-env-changed=SYSTEMD_PKG_NAME_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=SYSTEMD_PKG_NAME_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_SYSTEMD_PKG_NAME
  cargo:rerun-if-env-changed=SYSTEMD_PKG_NAME
  cargo:rerun-if-env-changed=LIBSYSTEMD_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBSYSTEMD_STATIC
  cargo:rerun-if-env-changed=LIBSYSTEMD_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  pkg_config could not find "libsystemd": `"pkg-config" "--libs" "--cflags" "libsystemd"` did not exit successfully: exit status: 1
  error: could not find system library 'libsystemd' required by the 'libsystemd-sys' crate

  --- stderr
  Package libsystemd was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libsystemd.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libsystemd' found

warning: build failed, waiting for other jobs to finish...
make: *** [Makefile:30: build] Error 101
```
**Expected behavior**
make should build the project without any errors

 * Eruption version : Most current version pulled from your github repo
 * Your device(s) : Roccat Vulcan Pro

X3n0m0rph59, 2022-06-24:
Hi @DirkSchueler, thanks for reporting this issue!

I have pushed updated installation instructions on the `master` branch

DirkSchueler, 2022-06-25:
Hi @X3n0m0rph59,

That solved the problem! Thank you.

DirkSchueler, 2022-07-13:
Hi, 
Again I have problems building the project:

`sudo make install`
leads to 
```
# Please be sure that all Eruption daemons have been shut down completely!
# Otherwise there will be errors during installation (file busy)

Commencing installation of Eruption...
cp: cannot stat 'target/release/eruption-keymap': No such file or directory
make: *** [Makefile:163: install] Error 1
```

`sudo make stop`
says
```
Notifying system daemons about Eruption...
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
make: [Makefile:55: stop] Error 1 (ignored)
Shutting down daemons...
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
make: [Makefile:59: stop] Error 1 (ignored)
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
make: [Makefile:60: stop] Error 1 (ignored)
Unit /etc/systemd/system/eruption.service is masked, ignoring.
```

What am I doing wrong?

X3n0m0rph59, 2022-07-13:
Hi @DirkSchueler, thanks for reporting this issue!

Which branch are you currently on? The `master` as well as the `develop` branch both contain the newly added utility called `eruption-keymap`. So this has to be a problem with the build system. Did you try to run `make clean`? This might fix the problem.

`make stop` must be used without `sudo` in front of it. It will execute `sudo` internally and prompt for a password if needed. Only `make install` requires `sudo` in front of the command.

If all else fails and to start from a clean state, you could run the follow sequence of commands:

```shell
git clone https://github.com/X3n0m0rph59/eruption.git
cd eruption
make
make stop; sudo make install; make start
```

Then you will be on the `master` branch again using the latest commits.

DirkSchueler, 2022-07-13:
Thank you for your kind support.
That solved this problem. 
I was able to build and install the latest stuff from the master branch.
Now I am not able to switch to any profile neither by using FN-Key nor by using the eruption GUI Tool. 
I would appreciate getting a solution as soons as possbile.
And additionally: After each and every install I have to change /etc/eruption/eruption.conf to force eruption to work with my ROCCAT Vulcan Pro by setting the driver_maturity_level back to "experimental".

closed 2022-07-13 by DirkSchueler

X3n0m0rph59, 2022-07-13:
@DirkSchueler 

> Now I am not able to switch to any profile neither by using FN-Key nor by using the eruption GUI Tool.

Could you please provide log output regarding this issue?

```shell
sudo journalctl -u eruption.service -b
```

Please open up another issue for this!

WolfBlader, 2022-09-11:
To install from source with a fresh Ubuntu 22.04, i also needed to install protoc.
apt update && apt upgrade -y
apt install -y protobuf-compiler libprotobuf-dev