eruption-project/eruption #147: eruption-process-monitor[2884]: netlink recv: Bad file descriptor

Gonzo2O28, 2022-06-30:
**Introduction**
Hello: "eruption-process-monitor[2884]: netlink recv: Bad file descriptor" is filling my journal, which has 100% cpu load now. Any idea?

device: roccat vulcan 120 aimo

its working fine otherwise, except for that it sometimes crashes and i need to restart the service.

X3n0m0rph59, 2022-06-30:
Hi  @Gonzo2O28, sounds like `eruption-process-monitor` is unable to read from the netlink Linux kernel socket that delivers notifications about process events. It just spins in a tight loop trying to read indefinitely.

Could you please check the permissions of `/usr/bin/eruption-process-monitor`?
The binary file needs to have the `cap_net_admin` capability set: `/usr/bin/eruption-process-monitor cap_net_admin=ep`

```shell
sudo setcap CAP_NET_ADMIN+ep /usr/bin/eruption-process-monitor
```

To verify:

```shell
getcap /usr/bin/eruption-process-monitor
/usr/bin/eruption-process-monitor cap_net_admin=ep
```

Gonzo2O28, 2022-06-30:
That is what i am getting:

sudo setcap CAP_NET_ADMIN+ep /usr/bin/eruption-process-monitor
"Failed to set capabilities on file '/usr/bin/eruption-process-monitor': Operation not supported"

X3n0m0rph59, 2022-07-01:
@Gonzo2O28 ah yes, so it seems that this is the cause of the problem!

Would it possible for you to enable support for posix file capabilities on that f2fs filesystem?
I can confirm that f2fs does support capabilities. It just needs the mount option `defaults`, or to be more specific the `suid`option to be specified in `/etc/fstab`.

Gonzo2O28, 2022-07-01:
I am still getting "Operation not supported" after adding "defaults" and rebooted.

X3n0m0rph59, 2022-07-01:
@Gonzo2O28 

What Linux distro are you currently using? The kernel needs to be compiled with the configuration `F2FS_FS_SECURITY` being set.

X3n0m0rph59, 2022-07-01:
I am currently preparing a patch that will allow the `eruption-process-monitor` daemon to continue to run, even if the Linux netlink socket can't be read from. The `eruption-process-monitor` daemon will just not react to process creation/termination events anymore.

This will allow the other sensors like X11 / Wayland window focus notifications to continue to operate.

Gonzo2O28, 2022-07-01:
> @Gonzo2O28
> 
> What Linux distro are you currently using? The kernel needs to be compiled with the configuration `F2FS_FS_SECURITY` being set.

Arch Linux, with linux-clear from the AUR.

X3n0m0rph59, 2022-07-12:
I have published updated packages that contain a fix for this issue. Please re-open the issue if the problem persists!

closed 2022-07-12 by X3n0m0rph59