eruption-project/eruption #148: No error popup

array-in-a-matrix, 2022-07-08:
**Introduction**
Hey, I bought a `Kone Aimo` mouse and was hoping a can help out this awesome project a bit. Unfortunately, I have no rust knowledge (yet).

**Please describe the bug**
The eruption GUI crashes if the configuration has a duplicate entry. The problem is that the GUI does not let the user know why it crashes.

**To Reproduce**
Duplicate any key in the configuration file e.g. `enable_experimental_features`, `driver_maturity_level`. 

**Expected behavior**
A pop up should appear telling the user the config file has a duplicate entry.

**Logs**
<details>
<summary>Eruption GUI (click to expand)</summary><br>

```sh
ERROR eruption_gui > Could not parse configuration file: duplicate key: `enable_experimental_features` for key `global` at line 4 column 1 in ../../etc/eruption/eruption.conf
```

</details>
<details>
<summary>Eruption daemon (click to expand)</summary><br>

```sh
Jul 08 01:38:02 archlinux systemd[1]: Starting Realtime RGB LED Driver for Linux...
Jul 08 01:38:02 archlinux systemd[1]: Started Realtime RGB LED Driver for Linux.
Jul 08 01:38:02 archlinux eruption[39272]: archlinux eruption[39272]: Could not parse configuration file: duplicate key: `enable_experimental_features` for key `global` at line 4 column 1 in etc/eruption/eruption.conf
Jul 08 01:38:02 archlinux systemd[1]: eruption.service: Main process exited, code=exited, status=4/NOPERMISSION
Jul 08 01:38:02 archlinux systemd[1]: eruption.service: Failed with result 'exit-code'.
```

</details>

**Please complete the following information:**
 * Eruption version [e.g. 0.1.20]:
    * Eruption 0.1.24 (0.1.23.r102.g618d4211) (release build) (compiled `eruption-git` from the AUR)
 *  Your device(s) (`sudo lsusb`):
    * Bus 001 Device 007: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K

**Additional context**
This should be pretty easy fix as other error messages are shown with a popup.

X3n0m0rph59, 2022-07-08:
Hi @array-in-a-matrix, thanks for reporting this issue!

This clearly is a bug... I will push a fix for it soon!

array-in-a-matrix, 2022-07-08:
No problem!

X3n0m0rph59, 2022-07-08:
The problem should be fixed now in the `develop` branch! Please feel free to reopen this issue if the problem should persist.

commit 0045d694 (2022-07-08): Improve error handling in `eruption-gui`

Show very early errors too, instead of just terminating the application.
Introduce an error log that accumulates errors for display until the GUI
is ready

Fixes #148
closed 2022-07-08 by X3n0m0rph59