eruption-project/eruption #150: Compiling error on Garuda

Siiire, 2022-07-09:
I can't compile eruption (non git) from the AUR on Garuda Linux.

For more information about this error, try 'rustc --explain E0310'.
error: could not compile 'wayland-backend' due to 13 previous errors
warning: build failed, waiting for other jobs to finish... 

Let me know if you need any additional info. I am not a dev or really used to Github in general, sorry in advance for any noobiness :)

X3n0m0rph59, 2022-07-09:
Hi @Siiire, thanks for reporting this issue!

The build of the `aur/eruption` package (non-git) is broken across all Arch based distros.

X3n0m0rph59, 2022-07-09:
@Siiire I have just pushed a fix for the build issue to the aur!

Building the `aur/eruption` (non-git) package should now work again.

X3n0m0rph59, 2022-07-09:
I will close this issue now, please open up a separate issue regarding the problem with the systemd unit!

closed 2022-07-09 by X3n0m0rph59