eruption-project/eruption #159: Add Support for Roccat Kone XP Air

danir-de, 2022-09-15:
**Introduction**
The new Roccat Kone XP Air mouse got recently released, containing the mouse itself, a usb dongle (for 2.4GHz) and a charging station, that acts as an USB hub and is able to synchronize to the mouse, if the dongle is plugged in into it.

The device itself has four different profiles, supports 1000Hz, debounce, AIMO, lift-off-distance-settings, energy-saving mode, dock-syncing and more.

Support for this device would be awesome! 

Website: https://www.roccat.com/products/kone-xp-air

**Device to add support for**
```
Bus 003 Device 015: ID 1e7d:2cb6 ROCCAT ROCCAT Kone XP Air Dongle
Bus 003 Device 014: ID 1e7d:2cb8 ROCCAT ROCCAT Kone XP Air Docking
Bus 003 Device 013: ID 1e7d:2cb5 ROCCAT USB 2.0 Hub
```

**Please attach your *.pcapng file, if possible**
Currently not possible on my end.

Hatsune-Cthulhu, 2023-11-22:
Just curious if the mouse is supported at all yet?

Hatsune-Cthulhu, 2023-11-23:
I now own this mouse, so if someone can provide instruction on how to get the pcap I'll happily provide it

Andorreta, 2024-02-15:
Hi, i made a pcapng capture with wireshark of the initialization process. Hope it helps. If i can contribute something more, please, let me know :D

[Roccat Kone xp Air init.zip](https://github.com/eruption-project/eruption/files/14299347/Roccat.Kone.xp.Air.init.zip)