eruption-project/eruption #161: Add MVK-Switch improvements

SHU-red, 2022-09-21:
**Introduction**
- Im using a MVK-Switch since a few days to be able to use two pcs but one screen, mouse and keyboard
- [Link to Device](https://www.amazon.de/gp/product/B0B1MP6GPW/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)
- The USB devices (Mouse + Keyboard) are both plugged in to the switch, and the switch is only connected to my PC(s) via USB-C to USB-A cable
- My Roccat vulcan seems to be not very compatible to the mvp switch regarding hotkey to switch (ctrl+ctrl) and keeping/switching the eruption configuration
- I know that i do not get all the details but im pretty sure, that the mvp switch "just" connects and disconnects the usb devices from one pc to another while switching
- for some reason the ROCCAT VULCAN does not "recover" the eruption settings

**Describe The Problem To Be Solved**
- Because of the MVK-switch, the ROCCAT VULCAN always seems to have default configuration
- Eruption configuration never gets applied and does not recover after switching back to a certain pc --> As mentioned above: Switching at the MVK switch should perform the same action as "Plugging out and Plugging in the ROCCAT into another PC"

**Describe the Solution**
- It would already be a great improvement or feature for eruption to immediately reconfigure the keyboard when plugged in, even if it gets plugged in and out multiple times

**My System**
  Kernel: 5.19.9-1-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 12.2.0
    Desktop: awesome v: 4.3-1529-gbde785ee6 vt: 1 dm: SDDM
    Distro: Manjaro Linux base: Arch Linux

And i am switching between a LINUX PC and a WINDOWS PC

If i can provide a log which is helping you, please describe shortly exactly what commands i shall use to be able to provide the log

X3n0m0rph59, 2022-10-01:
Hi @SHU-red, thanks for this excellent bug report!

> I know that i do not get all the details but im pretty sure, that the mvp switch "just" connects and disconnects the usb devices from one pc to another while switching

Yes, I think that this is exactly the way it works...

When Eruption fails to initialize the device after being hotplugged by the KVM switch, what does the following command report?

```shell
sudo systemctl list-units --failed
```

I suspect that the `eruption-hotplug-helper.service` is entering a `failed` state. The service control manager then refrains from starting it again, until a timeout is reached.

The failed state can be cleared by running the following command:

```shell
sudo systemctl reset-failed
```

In case the `eruption-hotplug-helper.service` entered the `failed` state, a hotplug event can be triggered manually using the following command:

```shell
sudo eruption-hotplug-helper hotplug
```

Does the running `eruption` daemon then start to initialize the respective device?

SHU-red, 2022-10-01:
> what does the following command report?
> > sudo systemctl list-units --failed

```
  UNIT                   LOAD   ACTIVE SUB    DESCRIPTION                     
● systemd-binfmt.service loaded failed failed Set Up Additional Binary Formats

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
1 loaded units listed.
```
---
> The failed state can be cleared by running the following command:
> > sudo systemctl reset-failed

```
❯ sudo systemctl reset-failed

~ 
❯ sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.
```
---
> Does the running eruption daemon then start to initialize the respective device?
```
❯ sudo eruption-hotplug-helper hotplug
 INFO  eruption_hotplug_helper > A hotplug event has been triggered, notifying the Eruption daemon...
 INFO  eruption_hotplug_helper > Waiting for the devices to settle...
 INFO  eruption_hotplug_helper > Done, all devices have settled
 INFO  eruption_hotplug_helper > Connecting to the Eruption daemon...
 INFO  eruption_hotplug_helper > Notifying the Eruption daemon about the hotplug event...
 INFO  eruption_hotplug_helper > Disconnected from the Eruption daemon

~ 
❯ sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.
```

Seems that eruption is not in failed state afterwards, but the keyboard does not load my configuration and also has a "default-behavior"

Thank you for taking the time

X3n0m0rph59, 2022-10-01:
@SHU-red So a failed service does not seem to be the cause for Eruption failing to initialize the device. There apparently is some unknown interaction when the KVM switch is involved.

Please post the output of the following command:

```shell
sudo udevadm monitor
```

Most relevant would be the output that is generated while the KVM switch hot-plugs the device. I am interested in the differences between hot-plugging the device via the KVM switch as opposed to plugging it directly into the USB root hub.

X3n0m0rph59, 2022-10-02:
@SHU-red I did some research on this... It seems that this class of problems is rather common when using KVM switches...

Could you please verify that the following issues are ruled out, if you have not already done so:

* Could this be a USB 2 vs. USB 3.x issue? The KVM switch exposes a USB-C port, is it plugged into a USB 3 port on the mainboard (on the Linux PC)? Could you try a different port on another root hub? (Having a USB-C connector does not guarantee that the device really exposes USB 3.x)
* Does the KVM switch perform some kind of keyboard emulation? Maybe this can be turned off completely?

Could you post the output of:

```shell
sudo lsusb --tree -vvv
```

and the output of:

```shell
sudo lsusb -vvv
```

Thank you for your support in getting this working!

SHU-red, 2022-10-02:
```
❯ sudo lsusb --tree -vvv
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb4  /dev/bus/usb/004/001
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 2109:0817 VIA Labs, Inc. 
        /sys/bus/usb/devices/4-2  /dev/bus/usb/004/002
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb3  /dev/bus/usb/003/001
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 1a40:0101 Terminus Technology Inc. Hub
        /sys/bus/usb/devices/3-2  /dev/bus/usb/003/002
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 2109:2817 VIA Labs, Inc. 
            /sys/bus/usb/devices/3-2.2  /dev/bus/usb/003/003
            |__ Port 3: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
                ID 1e7d:3098 ROCCAT 
                /sys/bus/usb/devices/3-2.2.3  /dev/bus/usb/003/005
            |__ Port 3: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                ID 1e7d:3098 ROCCAT 
                /sys/bus/usb/devices/3-2.2.3  /dev/bus/usb/003/005
            |__ Port 3: Dev 5, If 3, Class=Human Interface Device, Driver=usbfs, 12M
                ID 1e7d:3098 ROCCAT 
                /sys/bus/usb/devices/3-2.2.3  /dev/bus/usb/003/005
            |__ Port 3: Dev 5, If 1, Class=Human Interface Device, Driver=usbfs, 12M
                ID 1e7d:3098 ROCCAT 
                /sys/bus/usb/devices/3-2.2.3  /dev/bus/usb/003/005
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 12M
            ID 1a86:8072 QinHeng Electronics 
            /sys/bus/usb/devices/3-2.3  /dev/bus/usb/003/004
            |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                ID 046a:b103 Cherry GmbH 
                /sys/bus/usb/devices/3-2.3.2  /dev/bus/usb/003/006
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb2  /dev/bus/usb/002/001
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    /sys/bus/usb/devices/usb1  /dev/bus/usb/001/001
    |__ Port 10: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
        ID 8087:0a2a Intel Corp. Bluetooth wireless interface
        /sys/bus/usb/devices/1-10  /dev/bus/usb/001/002
    |__ Port 10: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
        ID 8087:0a2a Intel Corp. Bluetooth wireless interface
        /sys/bus/usb/devices/1-10  /dev/bus/usb/001/002
```

```
❯ sudo lsusb -vvv
Bus 004 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x0817 
  bcdDevice            0.50
  iManufacturer           1 VIA Labs, Inc.         
  iProduct                2 USB3.0 Hub             
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
[... 1127 lines cut ...]
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x00 0x00
  PortPwrCtrlMask    0xff 0xff 0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
   Port 9: 0000.0100 power
   Port 10: 0000.0103 power enable connect
   Port 11: 0000.0100 power
   Port 12: 0000.0100 power
   Port 13: 0000.0100 power
   Port 14: 0000.0100 power
   Port 15: 0000.0100 power
   Port 16: 0000.0100 power
Device Status:     0x0001
```
  Self Powered

SHU-red, 2022-10-02:
> Could this be a USB 2 vs. USB 3.x issue? The KVM switch exposes a USB-C port, is it plugged into a USB 3 port on the mainboard (on the Linux PC)? Could you try a different port on another root hub? (Having a USB-C connector does not guarantee that the device really exposes USB 3.x)

Im not very deep into this but the KVM-Switch is connected via USB-C - USB-C cable to one of my red USB 3.1 plugs
I could try to use a blue Usb 3.0 one?

> Does the KVM switch perform some kind of keyboard emulation? Maybe this can be turned off completely? Edit: Apparently emulation can be toggled by pressing Ctrl+Shift+k

Yes, via pressing Shift + Shift you can switch between the PCs
But im having problem with this (e.g. toggling loop etc.) thats why i plugged in the Keyboard to one of the standard USB 3.0 plugs in the device
This way the one usb plug listening to the Keyboard commands is unused

Just tell me what i should do and it would be a pleasure for me to contribute

X3n0m0rph59, 2022-10-03:
@SHU-red Thanks for your support! It is much appreciated!

Could you please post the output of the following command, while hot-plugging the keyboard device using the KVM switch? 

Please switch between the Linux PC and Windows PC while the following command is running on Linux:

```shell
sudo udevadm monitor
```

X3n0m0rph59, 2022-10-03:
@SHU-red Thanks, that was quick! :-)

Looks perfectly normal on first glance...

Could you please post the output of the following commands, while switching between Linux PC and Windows PC multiple times. I would like to get the logs of the error case where the `eruption` daemon fails to initialize the keyboard.

```shell
sudo journalctl -u eruption.service --follow
sudo journalctl -u eruption-hotplug-helper.service --follow
```

SHU-red, 2022-10-03:
> ```shell
> sudo journalctl -u eruption.service --follow
> ```

These messages have already been there directly after starting the upper command
Nothing was added after switching ...

```
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
Oct 03 10:47:53 sebastian-manjaro eruption[686]: sebastian-manjaro eruption[686]: Could not process a keyboard event: receiving on a closed channel
```

> ```
> sudo journalctl -u eruption-hotplug-helper.service --follow
> ```

```
Oct 02 16:18:37 sebastian-manjaro systemd[1]: Starting Eruption device hotplug notification daemon...
Oct 02 16:18:37 sebastian-manjaro systemd[1]: Started Eruption device hotplug notification daemon.
Oct 02 16:18:37 sebastian-manjaro systemd[1]: eruption-hotplug-helper.service: Deactivated successfully.
Oct 03 10:36:34 sebastian-manjaro systemd[1]: Starting Eruption device hotplug notification daemon...
Oct 03 10:36:34 sebastian-manjaro systemd[1]: Started Eruption device hotplug notification daemon.
Oct 03 10:36:35 sebastian-manjaro systemd[1]: eruption-hotplug-helper.service: Deactivated successfully.
Oct 03 10:48:16 sebastian-manjaro systemd[1]: Starting Eruption device hotplug notification daemon...
Oct 03 10:48:16 sebastian-manjaro systemd[1]: Started Eruption device hotplug notification daemon.
Oct 03 10:48:16 sebastian-manjaro systemd[1]: eruption-hotplug-helper.service: Deactivated successfully.
Oct 03 10:49:22 sebastian-manjaro systemd[1]: Starting Eruption device hotplug notification daemon...
Oct 03 10:49:22 sebastian-manjaro systemd[1]: Started Eruption device hotplug notification daemon.
Oct 03 10:49:22 sebastian-manjaro systemd[1]: eruption-hotplug-helper.service: Deactivated successfully.
```

X3n0m0rph59, 2022-10-03:
@SHU-red 

> Im not very deep into this but the KVM-Switch is connected via USB-C - USB-C cable to one of my red USB 3.1 plugs
I could try to use a blue Usb 3.0 one?

Yes, please also try other USB 2.x, 3.0 and 3.1 ports on the Linux PC if possible.

I don't really believe that this is the culprit though... I have a gut feeling that this is a bug with the driver code in `eruption`...

SHU-red, 2022-10-03:
OK for some reason the behavior was different now
As i switched back to the Linux PC, the keyboard stayed completely dark
As i typed a single key, it came up with MY setting and not the default one

```
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: Key up: Index: 62
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:   |0300fb1b 00000000|                   ........         00000000
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:                                                          00000008
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: HID event: 1 args: 27
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: Key up: Index: 14
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: Key down: Index: 89
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 19
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:   |0300fb6b 01000000|                   ...k....         00000000
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:                                                          00000008
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: HID event: 2 args: 107
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:   |0300fb6b 00000000|                   ...k....         00000000
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:                                                          00000008
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: HID event: 1 args: 107
Oct 03 12:10:12 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: Key up: Index: 89
Oct 03 12:10:13 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 16
Oct 03 12:10:14 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 17
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Trying to unplug the failed device
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Main loop terminated gracefully
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Unplugging a failed keyboard device...
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Sending device hot remove notification...
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Entering the main loop now...
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 16
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Keyboard device went away: No such device (os error 19)
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Main loop terminated gracefully
Oct 03 12:10:15 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Entering the main loop now...
Oct 03 12:10:16 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 16
Oct 03 12:10:17 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 10
Oct 03 12:10:18 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:19 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:20 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:21 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:23 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 0
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Eruption SDK client connected
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Hotplug event received, trying to claim newly added devices now...
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Enumerating connected devices...
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Found supported keyboard device: 0x1e7d:0x3098 (3-2.2.3:1.0) - ROCCAT ROCCAT Vulcan AIMO
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Bound driver: ROCCAT Vulcan 100/12x AIMO
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 03 12:10:24 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
[... cut ...]
Oct 03 12:11:19 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]:                                                          00000008
Oct 03 12:11:19 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: HID event: 1 args: 53
Oct 03 12:11:19 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: Macros: Key up: Index: 27
Oct 03 12:11:20 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 19
Oct 03 12:11:21 sebastian-manjaro eruption[1200736]: sebastian-manjaro eruption[1200736]: FPS: 17
```

SHU-red, 2022-10-04:
> Yes, please also try other USB 2.x, 3.0 and 3.1 ports on the Linux PC if possible.

Im not 100% sure because the behavior seems to change sometimes but using other USB-Slots does not solve all problems it seems

X3n0m0rph59, 2022-10-04:
@SHU-red This could be a timing issue then...

Could you please try out the latest commit (646692f) in the `develop` branch... It adds a short delay to the init sequence.

SHU-red, 2022-10-04:
Yes i will try
I did install "eruption-git" from the AUR repository

Could you just give me a quick hint how to switch to the develop path in my case?

X3n0m0rph59, 2022-10-04:
@SHU-red Thank you for your support!

To build Eruption from source please perform the following steps:

* If you have customized `*.profile` files or Lua scripts or `*.state`, please take backups. The following steps will overwrite all customizations of Eruption
* Install build dependencies:
```shell
sudo pacman -Sy libevdev hidapi systemd-libs dbus libpulse lua lua-socket gtksourceview4
sudo pacman -Sy pkgconf cmake protobuf xorg-server-devel libxrandr gtk3
```
* `cd` to a suitable directory like e.g.: `~/projects` or something like that
* Clone the project from GitHub into the current directory, switch to the (unstable) `develop` branch, and finally, start the compilation:
```shell
git clone https://github.com/X3n0m0rph59/eruption.git
git switch develop
cd eruption
make
```
* If the compilation completed successfully: Stop all running instances of Eruption, and then install all the newly built binaries and files. The files from the `eruption-git` package will simply be overwritten:
```shell
make stop; sudo make install
```
* Now start all Eruption daemons
```shell
make start
```

For detailed instructions please refer to [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md)

Please double check that you are running the latest version of Eruption from `develop`:

```
eruption -V
Eruption 0.3.0 (0.2.0.r73.g646692fc) (release build)
```

SHU-red, 2022-10-05:
Executing `make` got the error:

```
❯ make
error: process didn't exit successfully: `rustc -vV` (exit status: 1)
--- stderr
rustc: /home/sebastian/miniconda3/lib/libncursesw.so.6: no version information available (required by /usr/lib/libLLVM-14.so)
rustc: /home/sebastian/miniconda3/lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/libLLVM-14.so)
rustc: /home/sebastian/miniconda3/lib/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by /usr/lib/libLLVM-14.so)
rustc: /home/sebastian/miniconda3/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-14.so)

make: *** [Makefile:30: build] Error 101
```

tried the following line after searching a bit

```
conda install -c conda-forge ncurses
```

i read, that installing `ncurses-6.3-1` should fix the problem

X3n0m0rph59, 2022-10-05:
@SHU-red 

I am not very proficient using anaconda... Could you try to set up your Rust environment using rustup? 

https://rustup.rs/

```shell
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
```

The instructions from [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/develop/docs/INSTALL.md) are most likely to work!

SHU-red, 2022-10-05:
1st im so sorry that i did oversee the rustup thing
2nd im so sorry to present you the next build error below

```
   Compiling sourceview4 v0.3.0
error: linking with `cc` failed: exit status: 1
  |
  = note: "cc" "-m64" "/tmp/rustcsb0nwM/symbols.o" "/home/sebastian/eruption/target/release/deps/eruption_gui-b5fe3657b1245d1d.eruption_gui.2477e8fc-cgu.0.rcgu.o" "-Wl,--as-needed" "-L" "/home/sebastian/eruption/target/release/deps" "-L" "/usr/lib" "-L" "/home/sebastian/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,--start-group" "-Wl,--end-group" "-Wl,-Bstatic" "/home/sebastian/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-ff283b4bf550fa1c.rlib" "-Wl,-Bdynamic" "-lpangocairo-1.0" "-lpango-1.0" "-lgobject-2.0" "-lglib-2.0" "-lharfbuzz" "-lcairo" "-lgtksourceview-4" "-lgtk-3" "-lgdk-3" "-lz" "-lpangocairo-1.0" "-lpango-1.0" "-lharfbuzz" "-latk-1.0" "-lcairo-gobject" "-lcairo" "-lgdk_pixbuf-2.0" "-lgio-2.0" "-lgobject-2.0" "-lglib-2.0" "-ldbus-1" "-lgtk-3" "-lgdk-3" "-lz" "-lpangocairo-1.0" "-lpango-1.0" "-lharfbuzz" "-latk-1.0" "-lcairo-gobject" "-lcairo" "-lgdk_pixbuf-2.0" "-lgio-2.0" "-lgobject-2.0" "-lglib-2.0" "-latk-1.0" "-lgobject-2.0" "-lglib-2.0" "-lgdk-3" "-lz" "-lpangocairo-1.0" "-lpango-1.0" "-lharfbuzz" "-lgdk_pixbuf-2.0" "-lcairo-gobject" "-lcairo" "-lgobject-2.0" "-lglib-2.0" "-lpango-1.0" "-lgobject-2.0" "-lglib-2.0" "-lharfbuzz" "-lgdk_pixbuf-2.0" "-lgobject-2.0" "-lglib-2.0" "-lcairo-gobject" "-lcairo" "-lgobject-2.0" "-lglib-2.0" "-lgobject-2.0" "-lgio-2.0" "-lgobject-2.0" "-lglib-2.0" "-lgobject-2.0" "-lglib-2.0" "-lgobject-2.0" "-lglib-2.0" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-znoexecstack" "-L" "/home/sebastian/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/sebastian/eruption/target/release/deps/eruption_gui-b5fe3657b1245d1d" "-Wl,--gc-sections" "-pie" "-Wl,-zrelro,-znow" "-Wl,-O1" "-nodefaultlibs"
  = note: /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFReadRGBAImageOriented@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFSetField@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFFdOpen@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFIsCODECConfigured@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFGetField@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFClose@LIBTIFF_4.0'
          /usr/bin/ld: /usr/lib/libgdk_pixbuf-2.0.so: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
          collect2: error: ld returned 1 exit status
          
  = help: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `eruption-gui` due to previous error
make: *** [Makefile:30: build] Error 101

```

SHU-red, 2022-10-05:
OK i think it worked

```
❯ eruption -V
Eruption 0.3.0 (0.2.0.r84.g240e824c) (release build)
```

Could be, that my settings have  got lost because i "only" backed up `~/.config/eruption` but maybe also had other files manipulated? Don`t know

Then i re-performed:

```
sudoedit /usr/lib/systemd/system/eruption.service --> Changed to debug
sudo systemctl daemon-reload
sudo systemctl restart eruption.service
sudo journalctl -u eruption.service --follow
```

And this is the generated log after multiple times of switching back and forth (and also switching profiles via Fn-F1-4)

Basically i think it seems to work but im not sure because of the changing nature of the problem
I think i can feel some delay for every keystroke
The keyboard beeps at every keystroke (but i think this is something from the default settings?)
After switching back to my Linux-PC sometimes the Keyboard stayed black until the first keystroke was performed (During these phases, the log showed "Entering the main loop now..")

```
Oct 05 20:06:48 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 1
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 01000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 2 args: 119
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: FN key event registered
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key down: Index: 77
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 0
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 00000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 1 args: 119
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: FN key event registered
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key up: Index: 77
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 1
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 01000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 2 args: 119
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: FN key event registered
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key down: Index: 77
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 0
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 00000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 1 args: 119
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: FN key event registered
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key up: Index: 77
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 1
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 01000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 2 args: 119
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: FN key event registered
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key down: Index: 77
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Notifying audio proxy to play SFX with ID: 0
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb77 00000000|                   ...w....         00000000
Oct 05 20:06:49 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
[... cut ...]
Oct 05 20:07:05 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:   |0300fb17 01000000|                   ........         00000000
Oct 05 20:07:05 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]:                                                          00000008
Oct 05 20:07:05 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: HID event: 2 args: 23
Oct 05 20:07:06 sebastian-manjaro eruption[206347]: sebastian-manjaro eruption[206347]: Macros: Key down: Index: 6
```

X3n0m0rph59, 2022-10-05:
@SHU-red 

> OK i think it worked

OK, great! Glad that you were able to successfully build the project!

> Could be, that my settings have got lost because i "only" backed up ~/.config/eruption but maybe also had other files manipulated? Don`t know

Eruption stores its state (runtime configuration) in multiple places. Namely in the user's home directory at `~/.config/eruption`, the global state of the `eruption` daemon gets saved to `/var/lib/eruption/*.state` and `/var/lib/eruption/profiles/<name of profile>.state`.

> Basically i think it seems to work but im not sure because of the changing nature of the problem

Yes, that sounds plausible... 

> I think i can feel some delay for every keystroke

Apparently the delay is way too long then... I have updated the driver code and moved the delay outside of the polling loop. Additionally I reduced the sleep time as well.

> The keyboard beeps at every keystroke (but i think this is something from the default settings?)

Please try the following command and see if it fixes the problem:

```shell
eruptionctl config soundfx false
```

> After switching back to my Linux-PC sometimes the Keyboard stayed black until the first keystroke was performed (During these phases, the log showed "Entering the main loop now..")

OK, thanks for bringing this to my attention! I will have a look at the code in question...

Thanks again for your patience and continued support! The next steps would be to re-build and install Eruption from the latest commit in the `develop` branch and see if it improves the situation.

In the base directory where the Eruption source code resides, please execute the following commands:

```shell
git switch develop
git pull origin develop
make && make stop; sudo make install; sudo sed -i 's/testing/experimental/' /etc/eruption/eruption.conf; make start
```

This will pull the latest commits from the `develop` branch, compile Eruption, install it, enable `experimental` driver code and finally start the `eruption` daemons.

SHU-red, 2022-10-06:
OK mate,
there should be something important be captured in the log below

I recognized, that when my Windows PC is off, this must be the situation which reproducable causes problems!

Situation:
1. eruption working and KVM is on my Linux PC
2. sudoedit eruption to log-level debug
3. restart eruption service
4. start logging with journalctl
5. switching with the KVM switch to the Windows PC but its OFF!!!!!!!!! --> Keyboard stays black, only the white square buttons shining
6. switching back to the Linux PC --> eruption settings do not get recovered!!!!

To filter-out the annoying "standard-errors" i have, i excluded the word "closed"

Does this help?

```
❯ journalctl -u eruption.service --follow | grep -v "closed"
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: HID event: 2 args: 108
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: Key down: Index: 107
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: Key up: Index: 107
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:   |0300fb6c 00000000|                   ...l....         00000000
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:                                                          00000008
Oct 06 20:45:05 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: HID event: 1 args: 108
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:   |0300fb6b 01000000|                   ...k....         00000000
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:                                                          00000008
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: HID event: 2 args: 107
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: Key down: Index: 89
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:   |0300fb6b 00000000|                   ...k....         00000000
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]:                                                          00000008
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: HID event: 1 args: 107
Oct 06 20:45:06 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Macros: Key up: Index: 89
Oct 06 20:45:11 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Keyboard device went away: No such device (os error 19)
Oct 06 20:45:11 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Main loop terminated gracefully
Oct 06 20:45:11 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Entering the main loop now...
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Eruption SDK client connected
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Hotplug event received, trying to claim newly added devices now...
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Enumerating connected devices...
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found supported keyboard device: 0x1e7d:0x3098 (3-2.2.3:1.0) - ROCCAT ROCCAT Vulcan AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Bound driver: ROCCAT Vulcan 100/12x AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Found unsupported device: 0x46a:0xb103 - <unknown> USB Device
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Device enumeration completed
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Main loop terminated gracefully
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Eruption SDK client disconnected
Oct 06 20:45:17 sebastian-manjaro eruption[782222]: sebastian-manjaro eruption[782222]: Entering the main loop now...
Oct 06 20:45:21 sebastian-manjaro systemd-journald[331]: Suppressed 3224270 messages from eruption.service
```

X3n0m0rph59, 2022-10-07:
@SHU-red 

>> After switching back to my Linux-PC sometimes the Keyboard stayed black until the first keystroke was performed (During these phases, the log showed "Entering the main loop now..")

> OK, thanks for bringing this to my attention! I will have a look at the code in question...

It seems that device hotplug hangs because the `eruption` daemon didn't flag the device as failed previously.

Please test again with the latest commits from the `develop` branch. There is a great chance that it will work this time, with the latest patches applied! 

In the base directory where the Eruption source code resides, please execute the following commands:

```shell
git switch develop
git pull origin develop
make && make stop; sudo make install; sudo sed -i 's/testing/experimental/' /etc/eruption/eruption.conf; make start
```

This will pull the latest commits from the develop branch, compile Eruption, install it, enable experimental driver code and finally start the eruption daemons.

```shell
eruption -V
Eruption 0.3.0 (0.2.0.r94.gf0057f8e) (release build)
```

SHU-red, 2022-10-07:
After installing

```
❯ eruption -V
Eruption 0.3.0 (0.2.0.r95.g8deab11c) (release build)
'``

the keyboard went black and stayed like this even after typing

switching Linux > Win > Linux, it came back and loaded a only red profile with no effects
no problems there, after i changed loglevel and restarted the eruption service, it worked normal

This now is the log with Win-PC ON:
```
eruption on  develop via 🦀 v1.64.0 
❯ journalctl -u eruption.service --follow | grep -v "closed"
Oct 07 07:58:50 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: HID event: 1 args: 117
Oct 07 07:58:50 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key up: Index: 108
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]:   |0300fb6b 01000000|                   ...k....         00000000
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]:                                                          00000008
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: HID event: 2 args: 107
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 89
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]:   |0300fb6b 00000000|                   ...k....         00000000
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]:                                                          00000008
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: HID event: 1 args: 107
Oct 07 07:58:52 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key up: Index: 89
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Error sending LED map to a device: Invalid result
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Trying to unplug the failed device
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Main loop terminated gracefully
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Unplugging a failed keyboard device...
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Sending device hot remove notification...
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Entering the main loop now...
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Keyboard device went away: No such device (os error 19)
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Main loop terminated gracefully
Oct 07 07:58:55 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Entering the main loop now...
Oct 07 07:59:02 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Eruption SDK client connected
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Hotplug event received, trying to claim newly added devices now...
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Enumerating connected devices...
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported keyboard device: 0x1e7d:0x3098 (/dev/hidraw0) - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Bound driver: ROCCAT Vulcan 100/12x AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found supported device: 0x1e7d:0x3098 - ROCCAT ROCCAT Vulcan AIMO
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found unsupported device: 0x46a:0xb103 - <unknown> USB Device
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Found unsupported device: 0x46a:0xb103 - <unknown> USB Device
Oct 07 07:59:03 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Initializing the hotplugged keyboard device...
[... cut ...]
Oct 07 08:02:59 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 6
Oct 07 08:02:59 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 6
Oct 07 08:02:59 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 6
Oct 07 08:02:59 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 6
Oct 07 08:02:59 sebastian-manjaro eruption[11419]: sebastian-manjaro eruption[11419]: Macros: Key down: Index: 6
```

I think this could be it :smile:

SHU-red, 2022-10-11:
One last thing
1. Do you think we got the topic solved?
2. Do i have to do something going back to the correct update-path via pacman at a certain point? And how?

X3n0m0rph59, 2022-10-11:
Hi @SHU-red, thanks for testing the patches and sorting things out!

Just to check whether my understanding is correct, switching between PCs is working now for you most of the time? No more problems with Eruption failing to initialize the keyboard? If that is correct, then the root cause of the problem was indeed a timing issue.

> Do you think we got the topic solved?

Since the problem you encountered seems to be related to a timing issue and the current fix only touches the driver for the `ROCCAT Vulcan 100/120`, this problem may be present in the code of other device drivers as well.

The next step would be for me to fix the driver code of every supported device to adhere more to the USB specs.

> Do i have to do something going back to the correct update-path via pacman at a certain point? And how?

I would recommend you to (re-)install the latest git snapshot package using:

```shell
paru eruption-git
```

SHU-red, 2022-10-11:
> Just to check whether my understanding is correct, switching between PCs is working now for you most of the time? No more problems with Eruption failing to initialize the keyboard? If that is correct, then the root cause of the problem was indeed a timing issue.

I would agree with that

> I would recommend you to (re-)install the latest git snapshot package using:
> 
> ```shell
> paru eruption-git
> ```

Thanks
I've never seen $paru but installed it and worked flawlessly
I have no idea if my version now includes your fixes :sweat_smile: 
But it seems that everything is running smoooooth :smile: 

```
❯ eruption -V
Eruption 0.3.0 (0.2.0.r128.g6ca2eb76) (release build)
```
Thank you for your hard work!

X3n0m0rph59, 2022-10-11:
@SHU-red OK, great! Glad that it is working now!

> I have no idea if my version now includes your fixes :sweat_smile:
But it seems that everything is running smoooooth 

> ❯ eruption -V
Eruption 0.3.0 (0.2.0.r128.g6ca2eb76) (release build)

Yes, so you are currently running `eruption-git` from the `master` branch which includes the patch that fixes the timing issue for your device, but not for others. The latest commit that you are running is 6ca2eb76 which holds the following code:

https://github.com/X3n0m0rph59/eruption/blob/6ca2eb76b9148407179de9a3ac62acc98f83f78d/eruption/src/hwdevices/roccat_vulcan_1xx.rs#L451-L452

> Thank you for your hard work!

You are welcome!

I will leave this issue open for a while, and even going to pin it, so that others can chime in if they are encountering the same problem as well!

WolfBlader, 2022-10-23:
Hi, just to report it also fix the issue, for the integrated KVM of the Philips 439P9H ultrawide monitor with the Vulcan 120.
Running Ubuntu 22.04 and Windows 11.

SHU-red, 2023-06-02:
Ok,
im struggling a bit to post this here because i recently upgraded my setup with
- 4K KVM like [this](https://www.amazon.de/Monitor-Splitter-Ausg%C3%A4ngen-Tastatur-unabh%C3%A4ngiger/dp/B09FTPW36J/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=kvm+4x2&sr=8-6)
- 4K Extender like [this](https://www.amazon.de/dp/B08HRL7WM7?psc=1&ref=ppx_yo2ov_dt_b_product_details)

In general this is a very tough task and also causes difficulties with audo etc.

I did not expect eruption to work with this out of the box and i am absolutely not sure if eruption even has something to do with this but for me it does not work also after restarting the eruption.service

So im carefully asing if i can provide you some logs or execute some tasks to maybe find out that there may be a possible fix for my (and maybe someone others) special case?

By Keyboard is generally working but eruption does not seem to recognize something

During executing sudo systemctl restart eruption.service:
```
╰ $ journalctl -u eruption.service --follow
Jun 02 11:49:55 Fedora.fritz.box systemd[1]: eruption.service: Deactivated successfully.
Jun 02 11:49:55 Fedora.fritz.box systemd[1]: Stopped eruption.service - Realtime RGB LED Driver for Linux.
Jun 02 11:49:55 Fedora.fritz.box systemd[1]: eruption.service: Consumed 4.633s CPU time.
Jun 02 11:49:55 Fedora.fritz.box systemd[1]: Starting eruption.service - Realtime RGB LED Driver for Linux...
Jun 02 11:49:55 Fedora.fritz.box systemd[1]: Started eruption.service - Realtime RGB LED Driver for Linux.
Jun 02 11:49:58 Fedora.fritz.box eruption[8543]: Fedora.fritz.box eruption[8543]: Could not initialize LEDs: Invalid result
Jun 02 11:49:58 Fedora.fritz.box eruption[8543]: Fedora.fritz.box eruption[8543]: Terminating input thread due to a failed device
Jun 02 11:49:58 Fedora.fritz.box eruption[8543]: Fedora.fritz.box eruption[8543]: Skipping uninitialized device, trying to re-initialize it now...
Jun 02 11:49:58 Fedora.fritz.box eruption[8543]: Fedora.fritz.box eruption[8543]: Error opening the keyboard device: Could not open the device file
Jun 02 11:49:58 Fedora.fritz.box eruption[8543]: Fedora.fritz.box eruption[8543]: Could not initialize the device: Device not opened
Jun 02 11:58:47 Fedora.fritz.box systemd[1]: Stopping eruption.service - Realtime RGB LED Driver for Linux...
Jun 02 11:58:48 Fedora.fritz.box systemd[1]: eruption.service: Deactivated successfully.
Jun 02 11:58:48 Fedora.fritz.box systemd[1]: Stopped eruption.service - Realtime RGB LED Driver for Linux.
Jun 02 11:58:48 Fedora.fritz.box systemd[1]: eruption.service: Consumed 6.716s CPU time.
Jun 02 11:58:48 Fedora.fritz.box systemd[1]: Starting eruption.service - Realtime RGB LED Driver for Linux...
Jun 02 11:58:48 Fedora.fritz.box systemd[1]: Started eruption.service - Realtime RGB LED Driver for Linux.
Jun 02 11:58:51 Fedora.fritz.box eruption[18716]: Fedora.fritz.box eruption[18716]: Could not initialize LEDs: Invalid result
Jun 02 11:58:51 Fedora.fritz.box eruption[18716]: Fedora.fritz.box eruption[18716]: Terminating input thread due to a failed device
Jun 02 11:58:51 Fedora.fritz.box eruption[18716]: Fedora.fritz.box eruption[18716]: Skipping uninitialized device, trying to re-initialize it now...
Jun 02 11:58:51 Fedora.fritz.box eruption[18716]: Fedora.fritz.box eruption[18716]: Error opening the keyboard device: Could not open the device file
Jun 02 11:58:51 Fedora.fritz.box eruption[18716]: Fedora.fritz.box eruption[18716]: Could not initialize the device: Device not opened
```

X3n0m0rph59, 2023-06-02:
Hi @SHU-red, thanks for reporting your findings!

This looks suspiciously like the USB initialization problem! I believe this issue can be fixed by increasing the timeouts, which should not be a problem anymore, soon... 

I recently implemented parallel device-probing and parallel initialization (already available in the `unified-canvas` branch) which alleviates the need for device initialization to be as fast as possible. Eruption `0.3.x` still uses serialized probing and initialization which results in the total startup time of the daemon being the sum of the times that each connected device takes to come up. So it is highly desirable for the device-init code to really be as fast as possible. Especially if you have more than one supported device connected to your system.

With parallel probing and initialization, the total startup time of Eruption is the time it takes the slowest device to come up.
Okay, so I will increase the timeouts and push the code out to the `unified-canvas` branch which will ultimately get merged into `develop` some time in the future.

In the meantime: If you feel comfortable with compiling Eruption from source, I can prepare a patch with much larger timeouts, otherwise you would have to wait for the next release.

SHU-red, 2024-02-06:
Sorry i do not want to annoy you!
Is there any chance of progress in these cases?