eruption-project/eruption #169: Improve documentation of Eruption

X3n0m0rph59, 2022-10-04:
## Improve the documentation of Eruption

The following parts of the documentation could need some improvements:

* [USAGE_EXAMPLES.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/howtos/USAGE_EXAMPLES.md)

Contributions are always welcome!

R1yuu, 2022-10-08:
Is this only about the CLI or also the GUI?

X3n0m0rph59, 2022-10-08:
Hi @R1yuu, the Eruption GUI is currently lacking many features. I guess there won't be much to write about at this point in time... But regarding the Eruption GUI, we could use some screenshots for the image gallery on the project's homepage.

I think that the most common use-cases should be documented the best. So the top priority would definitely be to improve the HOWTOs!