eruption-project/eruption #172: Eruption process high CPU utilization after a dock disconnect

enyone, 2022-10-08:
**Introduction**
Eruption process high CPU utilization after a dock disconnect.

**Please describe the bug**
Eruption process high CPU utilization (single core/thread at 100%) after Lenovo USB-C dock disconnect. Keyboard is connected to a dock. Laptop is Lenovo T14 AMD.

**To Reproduce**
Unplug the dock from the laptop.

**Expected behavior**
No CPU utilization when no keyboard present (dock unplugged).

**Logs**

```
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Keyboard device went away: No such device (os error 19)
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
Oct 08 21:59:42 rnetl14 eruption[1603]: rnetl14 eruption[1603]: Could not process a keyboard event: receiving on a closed channel
```

**Please complete the following information:**

 * Eruption 0.2.0
 * Roccat Vulcan 120 Aimo

**Additional context**
A workaround is manually restart Eruption process,

```
$ sudo systemctl restart eruption.service
```

X3n0m0rph59, 2022-10-09:
Hi @enyone, thank you for reporting this issue!

I have pushed out a set of fixes on the `develop` branch. Could you please build Eruption from source and verify if this already fixes the problem?

HOWTO build from source can be found here: [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/develop/docs/INSTALL.md)

Please be sure to build the `develop` branch, by switching to it after cloning the repository!

```shell
git clone https://github.com/X3n0m0rph59/eruption.git
cd eruption
git checkout develop
make
```

enyone, 2022-10-09:
Would it be sufficient as a test to use rpm-build and this spec https://download.copr.fedorainfracloud.org/results/x3n0m0rph59/eruption/fedora-36-x86_64/04836153-eruption/eruption.spec and then applying a patch (created from `releases/v0.2.0` to `develop`) over the src.rpm before building?

I have the build environment in a kvm/qemu virtual environment and would not like to apply build environment to this local machine.

X3n0m0rph59, 2022-10-09:
@enyone I have prepared a .spec file based on `eruption-git.spec` that should be suitable for building the `develop` branch inside your virtual machine. I successfully built `*.srpm` and `*.rpm` package files in a bare minimum Fedora podman container, so the provided .spec file should work.

Please find the files here: [eruption-git.zip](https://github.com/X3n0m0rph59/eruption/files/9741832/eruption-git.zip)