eruption-project/eruption #177: build error: "could not find native static library `evdev`, perhaps an -L flag is missing?"

ensonic, 2022-10-16:
**Introduction**
I am trying to build git head (v0.2.0-137-gcbb50be0), since there are no packages for opensuse.

**Please describe the bug**
Build fails with the above error (see title).

**To Reproduce**
```
make BUILDFLAGS="--release --verbose"
error: could not find native static library `evdev`, perhaps an -L flag is missing?

error: could not compile `evdev-sys` due to previous error

Caused by:
  process didn't exit successfully: `rustc --crate-name evdev_sys --edition=2018 /home/ensonic/.cargo/registry/src/github.com-1ecc6299db9ec823/evdev-sys-0.2.5/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C panic=abort -C linker-plugin-lto -C codegen-units=1 -C overflow-checks=on -C metadata=ffbc97256fceed3d -C extra-filename=-ffbc97256fceed3d --out-dir /home/ensonic/projects/hardware/eruption/target/release/deps -L dependency=/home/ensonic/projects/hardware/eruption/target/release/deps --extern libc=/home/ensonic/projects/hardware/eruption/target/release/deps/liblibc-a3f30de4e29914ee.rmeta --cap-lints allow -L /home/ensonic/projects/hardware/eruption/target/release/build/evdev-sys-e345583c6194707c/out/lib -l static=evdev` (exit status: 1)
```

**Expected behavior**
Can we link this dynamiocally instead? There is not static link library packaged for my distro as it seems:
```
> locate libevdev
/usr/lib64/libevdev.so.2
/usr/lib64/libevdev.so.2.3.0
/usr/share/licenses/libevdev2
/usr/share/licenses/libevdev2/COPYING
> zypper se evdev
Repository-Daten werden geladen...
Installierte Pakete werden gelesen...

S  | Name                   | Summary                                                               | Type
---+------------------------+-----------------------------------------------------------------------+------
i+ | libevdev-devel         | Development files for libevdev library                                | Paket
i+ | libevdev-tools         | Library for handling evdev kernel devices                             | Paket
i+ | libevdev2              | Library for handling evdev kernel devices                             | Paket
   | libevdev2-32bit        | Library for handling evdev kernel devices                             | Paket
   | python38-evdev         | Python bindings to the Linux input handling subsystem                 | Paket
   | python38-libevdev      | Python wrapper around the libevdev C library                          | Paket
   | python39-evdev         | Python bindings to the Linux input handling subsystem                 | Paket
   | python39-libevdev      | Python wrapper around the libevdev C library                          | Paket
   | python310-evdev        | Python bindings to the Linux input handling subsystem                 | Paket
   | python310-libevdev     | Python wrapper around the libevdev C library                          | Paket
i+ | xf86-input-evdev       | Generic Linux input driver for the Xorg X server                      | Paket
i+ | xf86-input-evdev-devel | Generic Linux input driver for the Xorg X server -- Development Files | Paket
```

X3n0m0rph59, 2022-10-16:
Hi @ensonic, thank you for reporting this issue!

Did you follow the HOWTO from [INSTALL.md](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md)? The following dependencies are required to build eruption from source:

```shell
sudo zypper in cmake gcc libdbus-c++-devel libgudev-1_0-devel libX11-devel xorg-x11-devel \
    libhidapi-devel lua54-devel libevdev-devel protobuf-devel gtk3-devel gtksourceview4-devel
```

 [openSUSE based distros](https://github.com/X3n0m0rph59/eruption/blob/master/docs/INSTALL.md#on-opensuse-based-distros)

You need to have `Rust` (rustup recommended) and `git` installed as well!

I just tried to build Eruption in two openSUSE virtual machines! (I used a minimal installation for openSUSE)
On openSUSE Leap it is required to downgrade `Lua54` to `Lua53` to get it to build!

ensonic, 2022-10-16:
Thanks. This + `make clean` and now it built!

closed 2022-10-16 by ensonic