eruption-project/eruption #184: GitHub Sponsors

R1yuu, 2022-11-08:
Hey, did u ever think about joining GitHub Sponsors?
Are you maybe already on the waiting list?, or is it not supported in your country yet?
Would love to support you and your work!

X3n0m0rph59, 2022-11-25:
Hi @R1yuu, thanks for your suggestion!

Yes, I thought about applying for GitHub Sponsors... But nothing has been done yet. I still have to do some research about the legal implications of that.

It would certainly be cool to allow users to donate some money for a specific development task or for a specific device. This would definitely enable me to pick up new hardware, and add support for it more quickly.

X3n0m0rph59, 2023-11-26:
@R1yuu The sponsorship links are now in place on GitHub.
Thanks again for suggesting this and for your donation as well! It is very much appreciated!

closed 2023-11-26 by X3n0m0rph59