eruption-project/eruption #187: Roccat Kain 100 AIMO mouse buttons no longer detected.

Dworv, 2022-11-25:
**Introduction**
After restarting last night, my left and right mouse buttons aren't being detected.

**Please describe the bug**
My mouse movement is still being detected, but not the mouse buttons.

**To Reproduce**
I don't know what I did. It worked normally for months.

**Expected behavior**
I expect all of my mouse buttons to work.

**Logs**
```
Nov 24 07:47:13 epicness systemd[1]: Starting eruption.service - Realtime RGB LED Driver for Linux...
Nov 24 07:47:13 epicness eruption[862]: epicness eruption[862]: ** EXPERIMENTAL FEATURES are ENABLED, this may expose serious bugs! **
Nov 24 07:47:13 epicness eruption[862]: epicness eruption[862]: ** EXPERIMENTAL DRIVERS are ENABLED, this may expose serious bugs! **
Nov 24 07:47:13 epicness systemd[1]: Started eruption.service - Realtime RGB LED Driver for Linux.
Nov 24 07:47:14 epicness eruption[862]: epicness eruption[862]: Step 1: Invalid result
Nov 24 07:47:14 epicness eruption[862]: epicness eruption[862]: Wait 1: Invalid result
Nov 24 07:47:14 epicness eruption[862]: epicness eruption[862]: Wait 2: Invalid result
Nov 24 07:47:14 epicness eruption[862]: epicness eruption[862]: Wait 3: Invalid result
Nov 24 07:47:16 epicness eruption[862]: epicness eruption[862]: More than 82 milliseconds of jitter detected!
Nov 24 07:47:16 epicness eruption[862]: epicness eruption[862]: This means that we dropped at least one frame
Nov 24 07:47:16 epicness eruption[862]: epicness eruption[862]: Loop took: 942 milliseconds, goal: 41
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Error sending LED map to a device: Invalid result
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Trying to unplug the failed device
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: More than 82 milliseconds of jitter detected!
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: This means that we dropped at least one frame
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Loop took: 1020 milliseconds, goal: 41
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Skipping uninitialized device, trying to re-initialize it now...
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Error opening the mouse device: Could not open the device file
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Could not initialize the device: Device not opened
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Error opening the mouse device: Could not open the device file
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: This could be a permission problem, or maybe the device is locked by another process?
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Could not initialize the device: Device not opened
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Could not initialize LEDs: Device not opened
Nov 24 07:47:17 epicness eruption[862]: epicness eruption[862]: Could not grab the device: Device or resource busy (os error 16)
Nov 24 07:47:18 epicness eruption[862]: epicness eruption[862]: Skipping uninitialized device, trying to re-initialize it now...
Nov 24 07:47:18 epicness eruption[862]: epicness eruption[862]: Error opening the mouse device: Could not open the device file
Nov 24 07:47:18 epicness eruption[862]: epicness eruption[862]: Could not initialize the device: Device not opened
Nov 24 07:47:18 epicness eruption[862]: epicness eruption[862]: Skipping uninitialized device, trying to re-initialize it now...
Nov 24 07:47:18 epicness eruption[862]: epicness eruption[862]: Error opening the mouse device: Could not open the device file
```

**Please complete the following information:**

 * Eruption version 0.2.0
 * List your device(s) 
```
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 1e7d:2d00 ROCCAT ROCCAT Kain 100
Bus 005 Device 002: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 1532:0257 Razer USA, Ltd Razer Huntsman Mini
Bus 001 Device 003: ID 09da:166b A4Tech Co., Ltd. USB Device 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

X3n0m0rph59, 2022-11-25:
Hi @Dworv, it seems that the `eruption` daemon fails to initialize the USB device.

Could you please try to power cycle your system/re-plug the mouse device? This most likely will fix the issue.

Dworv, 2022-11-25:
Doing this seems to crash eruption (opening the gui makes this)
[image]

X3n0m0rph59, 2022-11-26:
@Dworv 

Could you post the full debug logs of the Eruption daemon starting up?

Please disable the `eruption.service` unit first:

```sh
sudo systemctl mask eruption.service && sudo systemctl stop eruption.service
```

Then please run `eruption` from a terminal, with debug logging enabled:

```sh
sudo RUST_LOG=debug RUST_BACKTRACE=full eruption
```

Please also post the output of each of the following commands:

```sh
uname -a
ps -elf
sudo lsof /dev/input/*
```

Dworv, 2022-11-28:
Eruption logs:
```
 Eruption is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 Eruption is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with Eruption.  If not, see <http://www.gnu.org/licenses/>.

 Copyright (c) 2019-2022, The Eruption Development Team

 INFO  eruption:1478 > Starting Eruption - Realtime RGB LED Driver for Linux: Version 0.2.0 () (release build)
 WARN  eruption:1529 > ** EXPERIMENTAL FEATURES are ENABLED, this may expose serious bugs! **
 WARN  eruption:1547 > ** EXPERIMENTAL DRIVERS are ENABLED, this may expose serious bugs! **
 INFO  eruption:1552 > Loading saved state...
 INFO  eruption:1571 > Registering plugins...
 INFO  eruption::plugin_manager:52 > Registering plugin: Keyboard - Keyboard related functions
 INFO  eruption::plugin_manager:52 > Registering plugin: Mouse - Mouse related functions
 INFO  eruption::plugin_manager:52 > Registering plugin: Macros - Inject programmable keyboard and mouse events
 INFO  eruption::plugin_manager:52 > Registering plugin: Introspection - Provide runtime status of the Eruption daemon
 INFO  eruption::plugin_manager:52 > Registering plugin: Persistence - A storage and persistence layer for Lua Scripts
 INFO  eruption::plugin_manager:52 > Registering plugin: Profiles - Switch profiles based on system state
 INFO  eruption::plugin_manager:52 > Registering plugin: System - Basic system information and status
 INFO  eruption::plugins::macros:79 > Initializing Linux virtual input devices
 INFO  eruption::plugin_manager:52 > Registering plugin: Sensors - Query system sensor values
 INFO  eruption::plugin_manager:52 > Registering plugin: Linux ULEDs - Linux Userspace LEDs interface
 INFO  eruption::plugins::uleds:161 > The ULEDs subsystem is not available on this kernel
 INFO  eruption::plugin_manager:52 > Registering plugin: SDK Support - Support for the Eruption SDK
 DEBUG eruption::plugins::sdk_support:305 > Unlink of control socket failed: ENOENT: No such file or directory
 INFO  eruption::plugin_manager:52    > Registering plugin: Audio - Audio related functions
 DEBUG eruption::plugins::audio::backends:344 > Unlink of audio socket failed: ENOENT: No such file or directory
 INFO  eruption::plugin_manager:52        > Registering plugin: Animal - Simulation of organic movements (support library)
 INFO  eruption::plugins::persistence:162 > Loading persistent state data from disk...
 INFO  eruption:1579                      > Plugins loaded and initialized successfully
 INFO  eruption:1582                      > Enumerating connected devices...
 DEBUG eruption::hwdevices:962            > Found supported device: 0x1e7d:0x2d00 - ROCCAT ROCCAT Kain 100
 INFO  eruption::hwdevices:1027           > Found supported mouse device: 0x1e7d:0x2d00 (5-2:1.0) - ROCCAT ROCCAT Kain 100
 INFO  eruption::hwdevices::roccat_kain_100:104 > Bound driver: ROCCAT Kain 100 AIMO
 DEBUG eruption::hwdevices:962              > Found supported device: 0x1e7d:0x2d00 - ROCCAT ROCCAT Kain 100
 DEBUG eruption::hwdevices:962              > Found supported device: 0x1e7d:0x2d00 - ROCCAT ROCCAT Kain 100
 DEBUG eruption::hwdevices:962              > Found supported device: 0x1e7d:0x2d00 - ROCCAT ROCCAT Kain 100
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x1532:0x257 - Razer Razer Huntsman Mini
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x1532:0x257 - Razer Razer Huntsman Mini
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x1532:0x257 - Razer Razer Huntsman Mini
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x1532:0x257 - Razer Razer Huntsman Mini
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x9da:0x166b - COMPANY  USB Device 
 INFO  eruption::plugins::audio::backends:381 > Audio proxy connected
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x9da:0x166b - COMPANY  USB Device 
 DEBUG eruption::hwdevices:1119             > Found unsupported device: 0x9da:0x166b - COMPANY  USB Device 
 INFO  eruption:1346                        > Opening mouse device...
 INFO  eruption:1359                        > Initializing mouse device...
 ERROR eruption::hwdevices::roccat_kain_100:452 > Step 1: Invalid result
 ERROR eruption::hwdevices::roccat_kain_100:454 > Wait 1: Invalid result
 ERROR eruption::hwdevices::roccat_kain_100:459 > Wait 2: Invalid result
 ERROR eruption::hwdevices::roccat_kain_100:464 > Wait 3: Invalid result
 INFO  eruption:1366                        > Configuring mouse LEDs...
 INFO  eruption:1372                        > Firmware revision: <unknown>
 INFO  eruption:1625                        > Spawning mouse input thread...
 INFO  eruption:1698                        > Device enumeration completed
 INFO  eruption:1707                        > Performing late initializations...
 INFO  eruption:1710                        > Loading saved device state...
 DEBUG eruption::state:205                  > 0x1e7d:0x2d00: Brightness: 58
[... cut ...]
Xorg      2003 dworv   46u   CHR  13,77      0t0  223 /dev/input/event13
Xorg      2003 dworv   47u   CHR  13,79      0t0  227 /dev/input/event15
Xorg      2003 dworv   48u   CHR  13,80      0t0  236 /dev/input/event16
Xorg      2003 dworv   69u   CHR  13,72      0t0 1016 /dev/input/event8
Xorg      2003 dworv   70u   CHR 13,257      0t0 1498 /dev/input/event257
Xorg      2003 dworv   71u   CHR  13,71      0t0 1042 /dev/input/event7
Xorg      2003 dworv   73u   CHR  13,73      0t0 1494 /dev/input/event9
```