eruption-project/eruption #192: [BUG] Input spam on X11 console screen

alkaris2, 2022-12-23:
**Introduction**
When switching to the X11 Console Screen with <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>F2</kbd> or whatever other F keys for console screen other than the desktop screen, at login screen for whatever reason cannot enter login due to a bug spamming same key sequence.

**Please describe the bug**
When trying to login to X11 Console screen where it asks for username and password, as soon after I have entered my username the key sequence being spammed into the `Password:` field with `^@` repeatedly, and it doesn't stop until I unplug the keyboard. This bug is caused by Eruption for whatever reason unknown. But when it's spamming `^@`  key sequence, it makes it impossible for me to drop to X11 Console screen to login, I am forced to switch out for a different keyboard other than the Vulkan 121 AIMO, because as soon as I plug it back in and try to type anything, it goes back to spamming `^@` into the console again. And Only when Vulkan 121 AIMO is plugged in, which leads to this being a bug with Eruption sending inputs where it shouldn't when nothing is being pressed.

**To Reproduce**
Steps to reproduce the behavior:

1. Drop to X11 console screen with <kbd>Ctrl</kbd>+<kbd>Alt</kbd> and whatever numbered F key
2. Attempt login and be greeted with console being spammed `^@` when password field is prompted.

**Expected behavior**
It should be functioning properly without spamming `^@` into the console screen, and I should be able to login without this happening every time.

**Logs**
```BASH
$ sudo journalctl -u eruption.service -b
[sudo] password for alkaris: 
Dec 23 21:44:41 tohru systemd[1]: Starting Realtime RGB LED Driver for Linux...
Dec 23 21:44:41 tohru systemd[1]: Started Realtime RGB LED Driver for Linux.
Dec 23 21:44:41 tohru eruption[668]: tohru eruption[668]: ** EXPERIMENTAL DRIVERS are ENABLED, this may expose serious bugs! **
Dec 23 21:44:42 tohru eruption[668]: tohru eruption[668]: Failed to query device class: No compatible devices found
Dec 23 21:44:42 tohru eruption[668]: tohru eruption[668]: Failed to query device class: No compatible devices found
Dec 23 21:44:42 tohru eruption[668]: tohru eruption[668]: Failed to query device class: No compatible devices found
Dec 23 21:44:47 tohru eruption[668]: tohru eruption[668]: Step 1: Invalid result
Dec 23 21:45:01 tohru eruption[668]: tohru eruption[668]: Socket error: Audio proxy disconnected
```

**Screenshots (optional)**

[video of behaviour in action](https://photos.app.goo.gl/9NF9q84bijRHXvm66)

**Please complete the following information:**

 * Eruption version: 0.3.0 (0.3.0.r5.gad2e5873) (release build)
 * List your device(s):
 ```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 046d:082d Logitech, Inc. HD Pro Webcam C920
Bus 003 Device 003: ID 046d:0a89 Logitech, Inc. G635 Gaming Headset
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 07ca:1513 AVerMedia Technologies, Inc. Live Gamer Portable 2 Plus
Bus 001 Device 004: ID 1397:0507 BEHRINGER International GmbH UMC202HD 192k
Bus 001 Device 003: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
Bus 001 Device 002: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

m1st0, 2022-12-24:
Confirmed result:
I get a constant stream of null character input from the keyboard after eruption starts on all TTYs.

Tests on Kubuntu:
No improvement from stopping eruption and returning to TTY.
No improvement from restarting eruption and returning to TTY.
No improvement whether keyboard is dark from eruption stopping or keyboard is in a hardware default color wave.

Improvement when keyboard is replugged into USB port.
Confirmed not hardware issue by restarting eruption and null character input returns on switch to any TTYs (i.e CTRL-ALT-F2, F3, ..., F6).

Only workaround is uninstalling eruption 0.3.0 at the moment for TTY screen issue. Also, null character input was seen on shutdown due to switch to TTY.

X3n0m0rph59, 2023-01-05:
Hi @alkaris2 and @m1st0, 

thanks for reporting this issue! I have released Eruption `0.3.2` that should fix the problem.

closed 2023-01-05 by X3n0m0rph59