eruption-project/eruption #200: Initialize Failure Flooding Journal

m1st0, 2023-01-25:
name | about | title | labels | assignees
-- | -- | -- | -- | --
Bug report | initializing devices | Initialize Failure Flooding Journal  |   |  

<strong>Introduction</strong>
I have a branch from [Discussion 155](https://github.com/X3n0m0rph59/eruption/discussions/155) currently updating with the Master branch, 
found at [non-root branch 155](https://github.com/m1st0/eruption/tree/discussion_155_nonroot).

<strong>Please describe the bug</strong>
My journal is flooded with the following.  I believe I have it in the correct 
order because it is non-stop.
```
eruption[1390]: Skipping uninitialized device, trying to re-initialize it now...
eruption[1390]: Error opening the keyboard device: Could not open the device file
eruption[1390]: Could not initialize the device: Device not opened
```
However, the keyboard and gui are functional and everything is working.

<strong>To Reproduce</strong>
Install the Discussion 155 branch, restart your system.
You should see the journal with the messages described.
Messages disappear if I restart Eruption services manually or from returning
system from sleep.

<strong>Expected behavior</strong>
Eruption not complaining since my devices are working.

<strong>Logs</strong>
```
Starting Realtime RGB LED Driver for Linux...
Started Realtime RGB LED Driver for Linux.
Error opening the keyboard device: Could not open the device file
This could be a permission problem, or maybe the device is locked by another process?
Could not initialize the device: Device not opened
Could not initialize LEDs: Device not opened
Skipping uninitialized device, trying to re-initialize it now...
Error opening the keyboard device: Could not open the device file
Could not initialize the device: Device not opened
More than 82 milliseconds of jitter detected!
This means that we dropped at least one frame
Loop took: 156 milliseconds, goal: 41
Skipping uninitialized device, trying to re-initialize it now...
Error opening the keyboard device: Could not open the device file
Could not initialize the device: Device not opened
```
Repeat last 3 log messages.

<strong>Please complete the following information:</strong>

<ul dir="auto">
<li>Eruption 0.3.2 (0.3.0.r18.g84e93fa2) (release build)</li>
<li>Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub</li>
<li>Bus 003 Device 002: ID 2109:0103 VIA Labs, Inc. USB 2.0 BILLBOARD</li>
<li>Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub</li>
<li>Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub</li>
<li>Bus 001 Device 005: ID 1770:ff00 MSI steel series rgb keyboard</li>
<li>Bus 001 Device 004: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO</li>
<li>Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller</li>
<li>Bus 001 Device 002: ID 046d:c332 Logitech, Inc. G502 Proteus Spectrum Optical Mouse</li>
<li>Bus 001 Device 007: ID 5986:0547 Acer, Inc NEC HD WebCam</li>
<li>Bus 001 Device 006: ID 8087:0025 Intel Corp. Wireless-AC 9260 Bluetooth Adapter</li>
<li>Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hu</li>
</ul>

<strong>Additional context</strong>
Again, messages disappear if returning from sleep or restarting eruption 
services. Should I add a delay in eruption.service?

With Discussion 155, I have the following modified in udev rules but it has no
effect on this issue.
```
# Permissions

ACTION=="add|change", SUBSYSTEMS=="input", RUN+="/usr/bin/setfacl -m u:eruption:rw /dev/input/by-id/eruption-virtual-keyboard"
ACTION=="add|change", SUBSYSTEMS=="input", RUN+="/usr/bin/setfacl -m u:eruption:rw /dev/input/by-id/eruption-virtual-mouse"
```

X3n0m0rph59, 2023-04-12:
Duplicate of #208

X3n0m0rph59, 2023-04-16:
Hi @m1st0 thank you for reporting this issue! I have pushed the first round of fixes to the `develop` branch.