eruption-project/eruption #207: Vulcan pro and kone pro not seen on gui

mMilca, 2023-03-18:
I cannot see any of my devices on the gui on debian 11
hwutil let me see the devices if I use sudo, and I can blackout the kone pro but not interaction possible with my keyboard

mMilca, 2023-03-19:
My bad after some more reading on all the available information I managed to make it work but only with root rights, any clue to help me run it with standard user ?

X3n0m0rph59, 2023-04-15:
Hi @mMilca, great that it is working now!

I am currently working on support for non-root Eruption. The basic stuff is working right now, but there are still problems with D-Bus IPC between the session daemons and the eruption core daemon running as non-root user.

I believe that the maybe the next release, or the release after that will ship with non-root support!

I will close this issue now, please open up another issue if you encounter further problems

closed 2023-04-15 by X3n0m0rph59