eruption-project/eruption #215: Bugs

ItzPrism, 2023-04-18:
**Introduction**
 Basically a lot of bugs

**Please describe the bug**
It constantly says "Ambient Effect Disabled" in the middle of my screen. Also Eruption GUI crashes when I have my Kone Pro plugged in. It also says that its a Unknown Generic Mouse.

**To Reproduce**
I have Pop!_OS 22.04 LTS with Gnome 44 without the shell extension. I have a Kone Pro and a GMMK.

Please provide full logs: `sudo journalctl -u eruption.service -b`
No logs shown

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * Eruption version [latest]
 * List your device(s) (`sudo lsusb`)
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0951:1666 Kingston Technology DataTraveler 100 G3/G4/SE9 G2/50
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 014: ID 222a:011f ILI Technology Corp. Multi-Touch-V5300
Bus 001 Device 012: ID 0408:50e2 Quanta Computer, Inc. HP IR Camera
Bus 001 Device 007: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 0d62:06eb Darfon Electronics Corp. HP Premium Lifestyle Wireless Keyboard/Mouse
Bus 001 Device 005: ID 0bda:b00b Realtek Semiconductor Corp. Realtek Bluetooth 4.2 Adapter
Bus 001 Device 004: ID 1058:25a1 Western Digital Technologies, Inc. Elements / My Passport
Bus 001 Device 003: ID 0c45:652f Microdia Backlit Gaming Keyboard
Bus 001 Device 002: ID 1e7d:2c88 ROCCAT ROCCAT Kone Pro
Bus 001 Device 013: ID 0408:51a1 Quanta Computer, Inc. HP 2.0MP High Definition Webcam
Bus 001 Device 008: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**Additional context**
Add any other context about the problem here.

magicalraccoon, 2023-04-29:
As far as the "Ambient Effect Disabled" issue, I was able to reproduce this on my end. (pop_os 22.04)

Additionally, the Settings tab shows a flip between the Effects Proxy daemon states. Manually turning the Effects Proxy off works for removing the constant desktop popup. This doesn't seem to persist though reboots, so the longterm workaround may be to turn off the daemon completely-- I don't use the RGB features on my mouse (kone xp), so it's not something I need.

magicalraccoon, 2023-05-02:
Update: Possibly not the best course of action-- My `/var/log/syslog` is ballooning with errors. I woke up to it taking up 800gb 😨

It continues to do so:
```
May  2 08:54:55 pop-os pop-os eruption[971503]: An error occurred while closing the device
May  2 08:54:55 pop-os pop-os eruption[971503]: Could not process a mouse event: receiving on a closed channel
May  2 08:54:55 pop-os pop-os eruption[971503]: An error occurred while closing the device
May  2 08:54:55 pop-os pop-os eruption[971503]: Could not process a mouse event: receiving on a closed channel
May  2 08:54:55 pop-os pop-os eruption[971503]: An error occurred while closing the device
May  2 08:54:55 pop-os pop-os eruption[971503]: Could not process a mouse event: receiving on a closed channel

```

magicalraccoon, 2023-05-02:
I was able to surpress the flashing "Ambient Effect Disabled" message by using the [gnome profile switcher extension](https://extensions.gnome.org/extension/2621/eruption-profile-switcher/) to uncheck "Show notifications on settings change".  
[image]

This will, of course, not alert you to any settings changes, but it also hides the popup without fully disabling the fx proxy.