eruption-project/eruption #216: Eruption 0.3.3 fails to start automatically on Ubuntu 23.04

RoggeHaj, 2023-05-25:
**Introduction**
After updating to eruption-0.3.3 (installed from PPA) on my Ubuntu 23.04 machine, LEDs are no longer working and the notification "Ambient Effect Disabled" pop up over and over. Restarting the eruption service resolves the problem.

**Please describe the bug**
As described in the introduction

**To Reproduce**
Steps to reproduce the behavior:
```shell
# When boot complete and logged in, LEDs not lit, notification says "Ambient Effect Disabled" over and over
DDIR=$(mktemp -d)

sudo dmesg > ${DDIR}/dmesg.txt
sudo journalctl -u eruption.service  -b > ${DDIR}/eruption.service.txt
sudo journalctl -u eruption-hotplug-helper.service -b > /${DDIR}/eruption-hotplug-helper.service.txt
sudo service eruption restart
# LEDs light up, notifications stop, working as expected
sudo journalctl -u eruption.service  -b > ${DDIR}/eruption.service_after_restart.txt
sudo journalctl -u eruption-hotplug-helper.service -b > /${DDIR}/eruption-hotplug-helper.service_after_restart.txt
```

**Expected behavior**
LEDs light up when machine is booted.
No notifications stating "Ambient Effect Disabled"

**Logs**
[dmesg.txt](https://github.com/X3n0m0rph59/eruption/files/11563063/dmesg.txt)
[eruption.service.txt](https://github.com/X3n0m0rph59/eruption/files/11563066/eruption.service.txt)
[eruption.service_after_restart.txt](https://github.com/X3n0m0rph59/eruption/files/11563067/eruption.service_after_restart.txt)
[eruption-hotplug-helper.service.txt](https://github.com/X3n0m0rph59/eruption/files/11563072/eruption-hotplug-helper.service.txt)
[eruption-hotplug-helper.service_after_restart.txt](https://github.com/X3n0m0rph59/eruption/files/11563078/eruption-hotplug-helper.service_after_restart.txt)

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * eruption/lunar,now 0.3.3 amd64 [installed]
 * [lsusb.txt](https://github.com/X3n0m0rph59/eruption/files/11563096/lsusb.txt)

X3n0m0rph59, 2023-05-25:
@RoggeHaj Thanks for reporting this issue! I can reproduce this in a Ubuntu 23.04 VM.

Eruption hangs under the following conditions:

- When it was started without detecting any devices
- E.g.: On bootup of the system when devices are not settled yet
-  `Slot 1` being the active one, all other slots seem to work

When the aforementioned conditions are met, subsequent device hotplug events will fail to completely bring up the plugged device. If the `eruption` daemon is able to detect any device during its startup-phase (e.g. after `systemctl restart eruption.service` on a fully booted system, when all devices are settled) the problem seems to not occur.

Please Note: In my Ubuntu VM it suffices to switch between slots to bring up the device to a usable state; e.g.:

```
eruptionctl switch slot 2
```

and then

```
eruptionctl switch slot 1
```

RoggeHaj, 2023-05-26:
Thanks, I will test a bit more during the day, both rebooting with another slot active and also trying to switch slots using `eruptionctl` when the problem occurs.

RoggeHaj, 2023-06-07:
Finally got around to test eruption-0.3.4 and that seems to have solved my problems. Great work!

closed 2023-06-07 by RoggeHaj