eruption-project/eruption #223: Vulcan Pro - wrong keys get illuminated, others don't even though they should

JonLit, 2023-07-17:
**Introduction**
Introductory text.

**Please describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Please provide full logs: `sudo journalctl -u eruption.service -b`

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**

 * Eruption version [e.g. 0.3.3]
 * List your device(s) (`sudo lsusb`)

**Additional context**
Add any other context about the problem here.

FNHG, 2024-08-15:
I'm having the same issue. I would like to help, if I can.

FNHG, 2024-08-15:
I need to add, that I'm using the german keyboard layout as well.