eruption-project/eruption #224: Vulcan Pro - Correct Keys get Illuminated, GUI shows wrong Keys

JonLit, 2023-07-17:
#223 but reverse

**Please describe the bug**
I noticed it with the Gaming: Generic Profile, the GUI shows F3, Y, Win and Alt Illuminated, but the Keyboard correctly shows W, A, S and D Illuminated.

**To Reproduce**
- Choose Gaming: Generic Effect with Vulcan Pro connected

**Expected behavior**
GUI shows correct Keys Illuminated

**Logs**
```
Jul 17 19:06:59 xmgneo-arch systemd[1]: Starting Realtime RGB LED Driver for Linux...
Jul 17 19:06:59 xmgneo-arch systemd[1]: Started Realtime RGB LED Driver for Linux.
Jul 17 19:06:59 xmgneo-arch eruption[5810]: xmgneo-arch eruption[5810]: ** EXPERIMENTAL FEATURES are ENABLED, this may expose serious bugs! **
Jul 17 19:06:59 xmgneo-arch eruption[5810]: xmgneo-arch eruption[5810]: ** EXPERIMENTAL DRIVERS are ENABLED, this may expose serious bugs! **
Jul 17 19:06:59 xmgneo-arch eruption[5810]: xmgneo-arch eruption[5810]: Wait 1: Invalid result
Jul 17 19:06:59 xmgneo-arch eruption[5810]: xmgneo-arch eruption[5810]: Wait 2: Invalid result
Jul 17 19:06:59 xmgneo-arch eruption[5810]: xmgneo-arch eruption[5810]: Wait 3: Invalid result
```

**Screenshots (optional)**
[image]
(again, sorry for the Dust and Dirt, will be cleaned soon™)

**Please complete the following information:**

 * Eruption 0.3.3 () (release build)
 * `sudo lsusb`
```
Bus 004 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. Card Reader
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 1e7d:2c8b ROCCAT ROCCAT Kone XP
Bus 003 Device 005: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 004: ID 04f2:b71a Chicony Electronics Co., Ltd Integrated IR Camera
Bus 003 Device 003: ID 048d:6004 Integrated Technology Express, Inc. ITE Device(8291)
Bus 003 Device 008: ID 1e7d:30f7 ROCCAT ROCCAT Vulcan Pro
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**Additional context**
see #223

FNHG, 2024-08-15:
I have the same issue as in #223, but using the specific "Gaming" preset you are using, the colors are correct on the keyboard but incorrect in the GUI, too.

KaptainAhoibrause, 2024-08-19:
I also experience this on ROCCAT Magma Keyboard. Also on German layout. When I press spacebar, Eruption illuminates the H-Key.