eruption-project/eruption #228: Add Support for Roccat Vulkan II

dr460nf1r3, 2023-10-28:
**Introduction**
Hello! 😊 I acquired the first mechanical keyboard in my life, a Roccat Vulkan II. I would love to see it supported by this wonderful project :) 

**Device to add support for**
Roccat Vulkan II
~~~
Bus 003 Device 003: ID 1e7d:2f4e ROCCAT [unknown]
~~~

**Please attach your *.pcapng file, if possible**
Below, you will find a zip with two files in it - one of them captured the initialization sequence for AIMO mode only, the other one captured the initialization sequence and additionally every single key clicked on the keyboard.
[wireshark.zip](https://github.com/eruption-project/eruption/files/13194931/wireshark.zip)

Thanks for your efforts!

X3n0m0rph59, 2023-11-10:
Hi @dr460nf1r3, thanks for your support! It is very much appreciated!

I have just found the time to take a look at the provided `*.pcapng` files... They are looking good! I should be able to add support for the `ROCCAT Vulcan II` using the packet traces.
I will get a hold on a `ROCCAT Vulcan II Max` very soon! (In the hopes that they are built reasonably similar)

> A sidenote: would you be interested in a NixOS build of this software? I could try packaging it.

Yes! That would really be great!

dr460nf1r3, 2023-11-10:
> I have just found the time to take a look at the provided `*.pcapng` files... They are looking good! I should be able to add support for the `ROCCAT Vulcan II` using the packet traces.

That sounds great! If you need someone to test anything you come up with in this regard, I'm right here 😊 

> > 
> 
> Yes! That would really be great!

I'm definitely giving it a try the next time I have some free time on my hands, then! Will need it with Vulkan II support arriving anyways :yum:

AidanRB, 2023-12-07:
+1 for NixOS packaging!

dr460nf1r3, 2023-12-08:
Still on my To-Do list! I'm currently quite busy but that will change in the future :)

gimmemabrewski, 2024-02-22:
I just got the vulkan 2 max [1e7d:2ee2] and would love to have this working with your software. My 120 AIMO works better with Eruption than Swarm! lol If there's anything I can do to help, let me know. Thanks!