eruption-project/eruption #230: Eruption profile switcher shell extension not compatible with Gnome 45

RoggeHaj, 2023-11-04:
**Introduction**
After upgrading to Ubuntu 23.10, including Gnome 45 desktop, the Gnome shell extension for Eruption Profile Switcher no longer works. Navigating to [Gnome Shell extensions page](https://extensions.gnome.org/local/) gives an error message as depicted in the 
[image].

**Please describe the bug**
The icon for the Eruption Profile Switcher is no longer displayed in shell top bar, even though installed. As described in the introduction, the Gnome Shell extensions page indicates that the installed version of Eruption is not compatible with the installed Gnome version.

**To Reproduce**
Steps to reproduce the behavior:

1. Log into Gnome
2. Take note of missing icon in Gnome Shell top bar
3. Take note of error message when navigating to [Gnome Shell extensions page](https://extensions.gnome.org/local/)

**Expected behavior**

1. Log into Gnome
2. Eruption Profile Switcher icon present in Gnome Shell top bar 

**Logs**
No applicable logs for the problem in the journal.

**Screenshots (optional)**
[image].

**Please complete the following information:**

 * Eruption version: eruption-git/mantic,now 0.3.5-4 amd64 [installed]
 * List your device(s): Bus 005 Device 005: ID 1e7d:311a ROCCAT ROCCAT Vulcan Pro TKL

**Additional context**
None

X3n0m0rph59, 2023-11-04:
@RoggeHaj Thanks for reporting this issue.

I have just submitted the latest revision of the `Eruption Profile Switcher` GNOME shell extension suitable for the GNOME Desktop >= 45. It is now pending code review and will be available soon!

RoggeHaj, 2023-11-13:
I've installed the new version of the Shell extension and it works as expected. Thanks for the quick response and excellent work.

Closing the issue.

closed 2023-11-13 by RoggeHaj