eruption-project/eruption #235: Issues with Media keys

alkaris2, 2023-12-29:
**Introduction**
I recently switched over to Manjaro KDE Plasma, and I've come to find this driver doesn't seem to respond to media control inputs just like when I  was using Manjaro Cinnamon. The Roccat Vulcan 121 with it's media keys and dial knob for volume control and LED brightness adjustment doesn't appear to work, at least the volume part anyway, switching to FX mode I can change the  the LED brightness, but when it's on Volume control it doesn't appear to be do anything except for light up half the keyboard to show volume level. 

**Please describe the bug**
Not too sure if this  is KDE bug or driver bug. But media keys and  knob dial doesn't function, media  keys for Stop and Pause/Play only seems to work when a video or music is being played, but it'll only let you use the volume controls when something is in use, but you can't change volume with the knob dial even though it shows the volume level when you turn it.

**To Reproduce**
Just using Manjaro KDE Plasma

**Expected behavior**
Was just hoping and expecting the media keys to continue to function after installing Eruption. The system does  seem to register inputs for media control, but no idea how to map them to the media controls.

**Logs**
[eruption-service.log](https://github.com/eruption-project/eruption/files/13790534/eruption-service.log)

**Please complete the following information:**

 * Eruption version: 0.1.9
 * List your device(s) (`sudo lsusb`)

``` 
Bus 001 Device 006: ID 1e7d:2e2c ROCCAT ROCCAT Kone Aimo 16K
Bus 001 Device 007: ID 1e7d:3098 ROCCAT ROCCAT Vulcan AIMO
```

**Additional context**
n/a

cooldudeseven7Dev, 2023-12-29:
Have you tried to reinstall eruption?

alkaris2, 2023-12-29:
> Have you tried to reinstall eruption?

it's literally the same result every time.

cooldudeseven7Dev, 2023-12-29:
> > Have you tried to reinstall eruption?
> 
> it's literally the same result every time. So no.

# NOTE: 
## does eruption gui open?
Thank you for the speedy reply. It seems like we are in the same time zone! :smile:
It seems like Manjaro KDE has different drivers than Manjaro Cinnamon. They can have different drivers. Heres what I suggest you do:

-- Install the following packages:
```
sudo pacman -Sy libevdev hidapi systemd-libs dbus libpulse lua lua-socket gtksourceview4
sudo pacman -Sy gcc pkgconf make cmake protobuf xorg-server-devel libxrandr gtk3
```
Also ensure that you have git and rust. MSRV 1.64, see install.MD for more details

Thank you!

herrcaptain, 2024-06-07:
I ran into a similar issue and found this thread looking for an answer. I thought my issue was the same as OP here as I'm also using KDE Plasma on an Arch-based system (Garuda) and the exact same keyboard.

It turned out that my keyboard had outdated firmware. Running 'sudo journalctl -u eruption.service' I found this message:

Jun 06 18:58:32 elysium eruption[253579]: elysium eruption[253579]: Outdated firmware version: 1.31, should be: >= 1.36

As per a comment by @X3n0m0rph59 on a similar issue from a few years ago ( found here: https://github.com/eruption-project/eruption/issues/79#issuecomment-904275257 ) I used a Windows computer to install the Swarm software from here (check the list of supported products for each to see which you need for your keyboard):

Swarm I: https://support.turtlebeach.com/s/downloads?language=en_US#/76/Roccat/ROCCAT-Swarm
Swarm II: https://support.turtlebeach.com/s/downloads?language=en_US#/319/Swarm-II

I accidentally downloaded Swarm II first and it auto-detected that it was the wrong version and automatically downloaded Swarm I.

From there you'll be able to update your firmware. Then, as soon as I plugged the keyboard back into my Linux system everything worked as expected.

This doesn't appear to be the OPs issue, as their linked eruption-service.log file doesn't seem to indicate outdated firmware. I'm just posting this comment in case anyone else finds this issue looking for a fix and wants to try this.

For further context, in case it helps anyone: I had installed the eruption package via Paru and Garuda automatically routed the install through their chaotic-AUR repo, which I believe just pulls from the standard AUR. Previous to discovering my firmware issue, I also applied the suggested fix by @cooldudeseven7Dev in the previous comment here and made sure to run the following commands to enable all the services in systemctl (from the install documentation):

systemctl --user enable --now eruption-fx-proxy.service
systemctl --user enable --now eruption-audio-proxy.service
systemctl --user enable --now eruption-process-monitor.service

sudo systemctl enable --now eruption.service

I hope this helps someone!