eruption-project/eruption #243: Installation fails on Ubuntu 24.04 LTS

panyann, 2024-05-29:
**Introduction**
The repository 'https://ppa.launchpadcontent.net/x3n0m0rph59/eruption/ubuntu noble Release' does not have a Release file.

**Please describe the bug**
Can't install the program.

**To Reproduce**
Steps to reproduce the behavior:
- follow the steps in readme.md for Ubuntu installation https://github.com/eruption-project/eruption?tab=readme-ov-file#ubuntu-or-pop_os

**Expected behavior**
Reposity probably should  have a "release file"?

**Logs**
```
yan@yan-ubuntu:~$ sudo apt update
Hit:1 http://pl.archive.ubuntu.com/ubuntu noble InRelease
Hit:2 http://pl.archive.ubuntu.com/ubuntu noble-updates InRelease                         
Hit:3 http://pl.archive.ubuntu.com/ubuntu noble-backports InRelease                       
Hit:4 https://repo.steampowered.com/steam stable InRelease                                
Ign:5 https://repo.vivaldi.com/stable/deb stable InRelease                                
Ign:6 https://ppa.launchpadcontent.net/x3n0m0rph59/eruption/ubuntu noble InRelease        
Hit:7 https://shop.softmaker.com/repo/apt stable InRelease                                
Hit:8 https://repo.vivaldi.com/stable/deb stable Release            
Hit:9 http://security.ubuntu.com/ubuntu noble-security InRelease    
Err:10 https://ppa.launchpadcontent.net/x3n0m0rph59/eruption/ubuntu noble Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/x3n0m0rph59/eruption/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

vake, 2024-06-06:
git clone https://github.com/eruption-project/eruption.git

cd eruption

sudo apt install pkg-config libudev-dev libatk1.0-dev libgdk-pixbuf2.0-dev libcairo2-dev  libgtk-3-dev libusb-1.0-0-dev libevdev-dev  liblua5.4-dev  protobuf-compiler libgtksourceview-4-dev libsystemd-dev

make

sudo make install 

make start

panyann, 2024-07-03:
@vake It worked, could be added to the readme...

Btw Eruption managed to change the RGB for my Kone AIMO, but it also removed all my buttons settings, so not worth it...

walker63, 2024-10-26:
> git clone https://github.com/eruption-project/eruption.git
>
> cd eruption
>
> sudo apt install pkg-config libudev-dev libatk1.0-dev libgdk-pixbuf2.0-dev libcairo2-dev libgtk-3-dev libusb-1.0-0-dev [...]

This does not work on a brand new and updated Install of Ubuntu 24.04 for me. This software has been working for me with a POPOS 22.04 install.