eruption-project/eruption #249: Can't build from source

kris99p, 2024-09-12:
**Introduction**
An error occurs during compilation.

**Please describe the bug**

kris@marvin:~
$ git clone https://github.com/eruption-project/eruption.git
Cloning into 'eruption'...
remote: Enumerating objects: 44486, done.
remote: Counting objects: 100% (5439/5439), done.
remote: Compressing objects: 100% (1262/1262), done.
remote: Total 44486 (delta 4208), reused 5242 (delta 4087), pack-reused 39047 (from 1)
Receiving objects: 100% (44486/44486), 95.25 MiB | 22.15 MiB/s, done.
Resolving deltas: 100% (28375/28375), done.
kris@marvin:~
$ cd eruption/
kris@marvin:~/eruption
$ make
    Updating git repository `https://github.com/ruabmbua/hidapi-rs.git`
error: failed to get `hidapi` as a dependency of package `eruption v0.3.6 (/home/kris/eruption/eruption)`

Caused by:
  failed to load source for dependency `hidapi`

Caused by:
  Unable to update https://github.com/ruabmbua/hidapi-rs.git?branch=master#b69e5db0

Caused by:
  object not found - no match for id (b69e5db0b7267901da65910d563e9d142e491948); class=Odb (9); code=NotFound (-3)
make: *** [Makefile:31: build] Error 101
kris@marvin:~/eruption

**To Reproduce**
git clone and than make

**Expected behavior**
the program should compile

AzhamProdLive, 2024-12-04:
+1; facing the same problem here

AzhamProdLive, 2025-03-01:
For those having the issue, use the patch available in the PR #253, this does fix the issue.